johanmeijer/grott #367: Layout   :  T060135XSPH - not processed

osos, 2023-07-14:
I see a lot of these records in the log, what are they, and should we not be able to understand them:

(SPH3600 with SDM630 and Shine Wifi-S)

```
grott  | 2023-07-14T15:11:10.217975198Z - Grott automatic protocol detection
grott  | 2023-07-14T15:11:10.217982402Z - Grott data record length 1093
grott  | 2023-07-14T15:11:10.217990761Z - layout   :  T060135XSPH
grott  | 2023-07-14T15:11:10.217997961Z - no matching record layout found, try generic
grott  | 2023-07-14T15:11:10.218005240Z - Record layout used: T060135XSPH
grott  | 2023-07-14T15:11:10.226240680Z - Grott Growatt data decrypted
grott  | 2023-07-14T15:11:10.227923932Z - Growatt plain data:
grott  | 2023-07-14T15:11:10.227945592Z 	01340006043d01354a50433843473930335300000000000000000000000000000000000000004f4
grott  | 2023-07-14T15:11:10.227951247Z 	d4354424d33303735000000000000000000000000000000000000000017070e110a3604753075ac
grott  | 2023-07-14T15:11:10.227955291Z 	0dad4f4d4354424d33303735000000000000000000000000000000000000000000008ca000008ca
grott  | 2023-07-14T15:11:10.227959181Z 	0000054600000546000008ca000007530000101d000010000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.227962949Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.227966724Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.227970477Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.227989128Z 	00000000000010023000000170007000e0011000a00040000000000010000000000000000000000
grott  | 2023-07-14T15:11:10.227993776Z 	0000000000000000000000000075ad7629000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.227997597Z 	0000000000000000000000000000000000000000000000000000000000000640000000000640000
grott  | 2023-07-14T15:11:10.228001789Z 	0000000000000000003c00002710000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228007690Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228012929Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228017970Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228023552Z 	000000000000000000000000000000000000000000000000000000000000762a76a600000000000
grott  | 2023-07-14T15:11:10.228028521Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228032407Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228037035Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228042478Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228046547Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228050353Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228054101Z 	00000000000000076a7772300000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228059201Z 	0000000000000000000000000000000000000000000000000000000000000064000a000a0000000
grott  | 2023-07-14T15:11:10.228063456Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228068740Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228074521Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228084483Z 	0000000000000000000000000000000000000000000000000000000000000000000000000000000
grott  | 2023-07-14T15:11:10.228090702Z 	000000000000000000000000000000000000000000000000027cb
grott  | 2023-07-14T15:11:10.228096617Z - Grott data ack record or data record not defined no processing done

```

johanmeijer, 2023-07-18:
Ok interesting I have not seen this record before. Let's see if we can found out what it is. Do you see the record often or only a few times a day? Do you think you miss information?

osos, 2023-07-19:
It is a brand new Wifi-S module with firmware 3.0.0.2 (I bricked the old one while trying to update it from the server.growatt.com interface)