johanmeijer/grott #369: Data Transfer Interval on Shine LAN Box

andymaycoates, 2023-07-17:
The Shine Lan Box / Shinelink has a network setting called Data Transfer Interval which is set to 5 minutes by default. I understand this is the frequency at which the logged data is sent to the specified server. I understand that Growatt do not take kindly to folks hitting their APIs too often, and can lock accounts. So I wondered with Grott implemented, is there a way to set the interval on the Shine Lan Box to, say, 30 seconds, but have Grott only push that data to the Growatt server every 5 minutes?

johanmeijer, 2023-07-18:
No at this moment that is not possible. You can set the interval time via the growatt website but I do not think you will be able to achieve a  interval less then 2 minutes (the growatt server will not keep up with lower intervals.

(it is implemented for pvoutput so probably is also possible for growatt. We can put it on the wish list.

andymaycoates, 2023-07-18:
Thanks for the quick reply. Interestingly, changing the Shine Lan Box interval to 1 minute seems to make no difference. Data in MQTT and on Growatt Server both showing 5 minute intervals.

rany2, 2023-07-18:
@andymaycoates You need to use grottserver to achieve sub-5 minute data collection. It works fine for me with 0.1 seconds as interval.

andymaycoates, 2023-07-19:
@rany2 thanks!

I just got a Docker container up and running on my RPI and it’s working great with my SPH5000 and HA.

I’ll look at grottserver next as I’m keen to explore direct API access to my inverter.

Although, I’m in no rush so I might wait until a Docker container is available with grottserver as standard.