johanmeijer/grott #370: Inverter Type

BarmanMaz, 2023-07-18:
Hi,
I have these inverter SPH4~10k TL3 BH, are they SPH or TL3, what i have to set up as invtype?

Thanks

johanmeijer, 2023-07-18:
Do you have grott already running and are you able to shareware log data (the growatt original data) then I can look at the format and determine which will be the best.

My first guess would be SPH. But let's look at it to be sure.

BarmanMaz, 2023-07-21:
Hi Here the plain datas 

- Growatt original Data:
		 \x00\x01\x00\x06\x01\x14\x22\x1b\x03\x2b\x2b\x47\x25\x36\x40\x77\x41\x3c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x34\x76\x73\x61\x4d\x6e\x45\x77\x8b\x47\x40\x77\x41\x5f\x59\x56\x42\x58
		 \x76\x42\x5e\x42\x59\x5a\x41\x75\x5e\x5f\x59\x51\x58\x45\x7f\x47\x41\x44\x4d
		 \x44\x5a\x77\x5e\x42\x4e\x57\x5a\x44\x6b\x43\x56\x4f\x4f\x41\x58\x74\x41\x5f
		 \x59\x58\x58\x4d\x71\x5c\x5e\x5b\x53\x42\x4d\x69\x47\x43\x44\x52\x44\x5a\x7e
		 \x5e\x5f\x59\x51\x58\x44\x69\x44\x43\x47\x4f\x44\x58\x75\x40\x58\x59\x53\x58
		 \x46\x75\x44\x41\x43\x4d\x46\x46\x73\x5c\x5b\x5b\x50\x5a\x40\x6b\x43\x41\x42
		 \x4d\x45\x5a\x7f\x5e\x5e\x4f\x50\x5a\x45\x6b\x46\x59\x4e\x4f\x4c\x58\x73\x41
		 \x5c\x59\x55\x58\x44\x69\x41\x43\x43\x58\x5a\x4d\x6b\x40\x5c\x42\x53\x4d\x5a
		 \x71\x5e\x5e\x47\x54\x44\x45\x69\x42\x43\x43\x50\x45\x4d\x69\x4a\x43\x41\x51
		 \x47\x4c\x69\x44\x43\x44\x54\x41\x45\x73\x5c\x59\x5b\x55\x45\x43\x73\x5c\x57
		 \x5b\x52\x44\x47\x73\x5c\x5d\x5b\x53\x4d\x40\x7e\x5c\x5b\x5b\x51\x5a\x44\x6b
		 \x42\x41\x47\x4d\x44\x5a\x77\x5e\x5b\x43\x50\x4c\x4d\x69\x40\x57\x5b\x52\x4d
		 \x45\x69\x47\x43\x44\x58\x44\x5a\x70\x5e\x5c\x4e\x51\x5a\x4d\x6b\xb5\x4c
	 - Grott automatic protocol detection
	 - Grott data record length 284
	 - layout   :  T06221b
	 - no matching record layout found, try generic
	 - Record layout used :  T06221b
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 000100060114221b4459443044423430335300000000000000000000000000000000000000000
		 0431707150a1c2a00ea33343033302e37362c31303135382e35322c302e302c3138352e332c30
		 2e302c2d39362e302c3139382e352c3333302e392c39362e312c3236392e352c3333302e392c3
		 02e302c302e362c302e302c3232372e322c3232362e342c3232342e342c312e342c312e352c31
		 2e382c3138312e312c3436392e382c3433332e342c302e332c34392e392c32333532392e362c3
		 1303530312e302c343131392e382c363033382e362c33353531342e362c343137342e382c3330
		 33342e322c323934392e342c302e302c302e302c302e302c34343138392e32382c3339312e352
		 c3339302e372c3339302e392cc723

johanmeijer, 2023-07-26:
This seem to be a record from a SDM630. Do you have one of these in your configuration?

BarmanMaz, 2023-07-27:
Yes, i have shinemaster as logger

johanmeijer, 2023-07-28:
Do you also have a original datarecord from the inverter (T06xxxx04 layout)?

johanmeijer, 2023-08-13:
Do you still  need this? Please send me an original  data record.