johanmeijer/grott #372: Grott keeps restarting;

GroenZoom, 2023-07-21:


GroenZoom, 2023-07-21:
Sorry, beiing unfamiliar with this medium I started another issue by accident.

rany2, 2023-07-21:
Please share your verbose logs, "Grott keeps restarting" without any additional information no one will be able to help you.

GroenZoom, 2023-07-22:
First of all many thanks for your work, contribution and support. 
I have found out that grottrpi iso grott container image keeps running.
The Shinelink network has been changed from server.growatt.com to my RPI3b+ HA host IP.
The original Growatt integration in HA is still working so data seems to arrive there.
Attached the output of Putty with verbose comments, is this what you meant with log?
My goal is to snif the data in order to reduce the interval to 1/min as in Shinelink.
It looks like Grott is working but I cannot get data out of the grott integration.
The latter integration is installaed and reports no devices or entities
MQTT server add-on is up and running for a number of RfRaw controlled curtains and covers.
I have no clou as howto check if data is available or datagrams arrive in HA.
Any suggestions as how to proceed towards from here are wecome.

Thanks in advance for any hint with kind regards
Martin J.
[ledidobe_grottrpi Putty output .txt](https://github.com/johanmeijer/grott/files/12136022/ledidobe_grottrpi.Putty.output.txt)

GroenZoom, 2023-07-22:
It appears grott is working. See attached Putty output after some time.
Apperently something wrong with MQTT. I will keep trying to get this up and running.
Ragards
[ledidobe_grottrpi Putty output part 2 .txt](https://github.com/johanmeijer/grott/files/12136097/ledidobe_grottrpi.Putty.output.part.2.txt)

rany2, 2023-07-23:
> MQTT send failed: Connection Refused: not authorised.

In practice,  MQTT failure should not result in grott restarting. However you have not configured MQTT properly, it appears your server requires authorization and you are not providing it.

GroenZoom, 2023-07-24:
Thanks for your advice. I have tried with the Grott HA-extension and that worked right from the start without any changes to my MQTT server settings. 
HomeAssistint 2023.7.3
Raspberry 3B+ Bullseye
Local via http:/ 192.168.xx.xx
Remote via https:/ xxxxx.duckdns.org
MQTT performs well on curtains, screens etc.
Thanks for the support, this may be closed.
Regards Martin

closed 2023-07-26 by johanmeijer