johanmeijer/grott #373: Help with Inverter MID 17KTL3-X

Snille, 2023-07-23:
Hi, I just started up gott... It seams to be working, just that it cant figure out what inverter I use...
And some numbers does not match... 
Here is a pictures from todays dashboard:
[image]
[image]

I still get data in to growatt servers... All good there.

I get this respons from PVOutput:  "Bad request 400: Energy value [20716352200] too high for system size [17010]"

I get numbers in to HA, but they seam to be a bit wrong... Let me know if you want a picture of that as well... :)

And here is a part of the log... Let me know if I can add anything else to help:
```
Jul 23 18:59:23 Grott01 grott[113]: #011 - Growatt original Data:
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x00\x22\x00\x06\x02\x41\x01\x04\x00\x22\x28\x47\x22\x33\x43\x76\x26\x5e\x77
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x2a\x21\x27\x3c\x37\x0c\x45\x5f\x45\x58\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x74\x3b\x72\x6e\x77\x61\x7b\x52\x49\xd2\x6f\x72\x61\x74\x73\x17\x63\x49
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x77\x62\x74\x74\x42\x57\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x79
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x8d\x67\xfd\x4e\xf2\x6f\x79\x61\x74\x79\x0b\x7b\x03\x77\x6f\x74\x74\x4a\x42
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x66\xf7\x61\x79\x74\x47\x7e\x36\x67\xe3\x64\x2c\x57\x25\x6f\x77\x60\x07\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x47\xc7\xa9\x77\x05\xb6\x2c\x47\x72\x6f\x84\x61\x74\x06\xa8\x72\x6f\x77\xe6
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x74\x05\xd1\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\xc2\xf3\x75
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x95\x46\xb6\x6e\xde\x61\x74\x74\x47\x7e\xf8\x7b\xfa\x3a\x54\x47\x72\x6f\x75
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \xf9\x64\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x6f\x77\x63\x74\x48\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x3a\x72\x96\x77\x61\x74\x74\x47\x72\x6f\x77\x61
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x60\x61\x73\x74\x51\x72\x78\x77\x66
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x63\x47\x72\x6f\x77\x61\x74\x8b\xbd\x72\x6c\x77\x67\x8b\x83\x47\x72\x6f
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \xdd\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x53\xe8\x47\x72\x6f\x77
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x61\x75\x35\xe0\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
Jul 23 18:59:23 Grott01 grott[113]: #011#011 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x93\x2b
Jul 23 18:59:23 Grott01 grott[113]: #011 - Grott automatic protocol detection
Jul 23 18:59:23 Grott01 grott[113]: #011 - Grott data record length 585
Jul 23 18:59:23 Grott01 grott[113]: #011 - layout   :  T060104X
Jul 23 18:59:23 Grott01 grott[113]: #011 - no matching record layout found, try generic
Jul 23 18:59:23 Grott01 grott[113]: #011 - Record layout used :  T06NNNNX
Jul 23 18:59:23 Grott01 grott[113]: #011 - Growatt data decrypted V2
Jul 23 18:59:23 Grott01 grott[113]: #011 - Grott Growatt data decrypted
Jul 23 18:59:23 Grott01 grott[113]: #011 - Growatt plain data:
Jul 23 18:59:23 Grott01 grott[113]: #011#011 00220006024101044750473043473731543100000000000000000000000000000000000000004
Jul 23 18:59:23 Grott01 grott[113]: #011#011 5564648434b373032390000000000000000000000000000000000000000000000000000020000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 007c000100000f260ea0000500000750112600030000052500000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 000000000000000000000000000000000000000000000000000000000000000000000000eec13
Jul 23 18:59:23 Grott01 grott[113]: #011#011 890980000e00000d4c096c000e00000d300980000d00000c59108210581057000001730000b5c
Jul 23 18:59:23 Grott01 grott[113]: #011#011 60064c258000000f3000072ef00000087000042a3000000000000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 0000000000000000000000000000000000000000000000000000000000000000b59201e101c40
Jul 23 18:59:23 Grott01 grott[113]: #011#011 1a9000000000c970c9b4e20000000029810000000000000000000000000000000000000000200
Jul 23 18:59:23 Grott01 grott[113]: #011#011 3c0000000000000000000000000000000000000000007d00f9000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 00000000000000000000000000000000000000000000000000000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: message repeated 2 times: [ #011#011 00000000000000000000000000000000000000000000000000000000000000000000000000000]
Jul 23 18:59:23 Grott01 grott[113]: #011#011 000000001700070016001700070017000000000000fffa00030006fff7000000aa00000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 00000000000000000000000000000000000000000000000000000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 000000000000279c00000000000141a7000000000000000000000000000000000000000000000
Jul 23 18:59:23 Grott01 grott[113]: #011#011 00000000000e144
Jul 23 18:59:23 Grott01 grott[113]: #011 - Possible Inverter serial EVFHCK7029
[... 45 lines cut ...]
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2":
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 250352521,
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 "totworktime": 851968, "pvenergytoday": 207163522, "pvenergytotal":
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 274206807, "epvtotal": 371, "epv1today": 46534, "epv1total": 6603352,
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 "epv2today": 243, "epv2total": 29423, "pvtemperature": 481,
Jul 23 18:59:23 Grott01 grott[113]: #011#011#011 "pvipmtemperature": 0}}
Jul 23 18:59:23 Grott01 grott[113]: #011 - Grott MQTT topic used : energy/growatt
Jul 23 18:59:23 Grott01 grott[113]: #011 - MQTT connection refused by target
Jul 23 18:59:23 Grott01 grott[113]: #011 - Grott send data to PVOutput systemid:  XXXXXX for inverter:  EVFHCK7029
Jul 23 18:59:23 Grott01 grott[113]: #011#011 -  {'X-Pvoutput-Apikey': 'VeRySeCrEtKeyHere!', 'X-Pvoutput-SystemId': '100326'}
Jul 23 18:59:23 Grott01 grott[113]: #011#011 -  {'d': '20230723', 't': '18:59', 'v2': 0.0, 'v6': 0.0, 'v1': 20716352200}
Jul 23 18:59:24 Grott01 grott[113]: #011 - Grott PVOutput response:
Jul 23 18:59:24 Grott01 grott[113]: #011#011 -  Bad request 400: Energy value [20716352200] too high for system size [17010]
Jul 23 18:59:24 Grott01 grott[113]: #011 - Grott Send data to Influx disabled
Jul 23 18:59:24 Grott01 grott[113]: #011 - Grott extension processing started :  grott_ha
[... cut ...]
Jul 23 18:59:24 Grott01 grott[113]: #011#011  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.26', 5279), raddr=('10.0.0.217', 1029)>
Jul 23 18:59:24 Grott01 grott[113]: #011 - Data less then minimum record length, data not processed
Jul 23 18:59:25 Grott01 grott[113]: #011 - Growatt packet received:
Jul 23 18:59:25 Grott01 grott[113]: #011#011  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('10.0.0.26', 50844), raddr=('47.91.67.66', 5279)>
Jul 23 18:59:25 Grott01 grott[113]: #011 - Data less then minimum record length, data not processed
```

Thank you!

Snille, 2023-07-26:
Thank you! It worked! :) Where was this information? I must have missed it somehow... :)

closed 2023-07-26 by Snille

johanmeijer, 2023-07-26:
I think I have documented this very well Sorry for that.