johanmeijer/grott #375: Different update frequency's...

Snille, 2023-07-27:
I'm trying to get "every" update in to HA but only update PVOutput every 5 minutes... 
Is there a way to do this? I can se that data is coming in about every other second (in the syslog), but it's not sent to HA.
Both are sent every 5th minute... 
Is there a way to separate the update intervalls?

Thank you.

Snille, 2023-07-28:
Yes, that I have figured out. However, it also only reports every 5:th minute to HA (Home Assistant) when the pvuplimit is set to 5...
I would like it to update HA as soon as it get's data from the Inverter, but keep pvuplimit set to 5. :)

johanmeijer, 2023-08-13:
To achieve this you have to change the update frequency of the datalogger (register 4). This can be done via the Growatt website (I think also in the Phone App) or  Grottserver (see #98 ).

Snille, 2023-08-14:
Hi again, I'm probably missing something, but here is what I see... :)
In the "Growatt ShineLan-X Settings" I have set the update interval to every minute (lowest I can), default is 5 minutes.
[image]
Then when I check the syslogfile (attached) 
[Syslog-file.txt](https://github.com/johanmeijer/grott/files/12332348/Syslog-file.txt)
And search for "Grott extension processing started :  grott_ha" it apperas every 5 minutes (same as the Grott PVOutput).
But all other times (when: "PVOut: Update refused for SERIAL due to time limitation") the ha extension is not processed either.
However I can see that the datalogger do send more often, but it's not "processed"... 
The result in HA then looks like this...
https://github.com/johanmeijer/grott/assets/1011699/22ea9b78-72fb-4f5d-a70b-209ffe104b02

What am I missing? :)

johanmeijer, 2023-08-14:
Setting the frequency in the shinelink does not works I think. You have do this via the Growatt website (set datalogger register 4) or using Grottserver.

Snille, 2023-08-14:
Hi again, it is set to every minute, here is a screenshot from the actual Growatt server...
[image]
Is there another place as well? I don't know how to change the "datalogger register to 4"...  When I try to "read" the value 4 from the regristry, I get is this:
[image]
Am I in the wrong place? :)

GeraldPape, 2023-09-27:
Did you succeed?

johanmeijer, 2023-09-27:
I am not sure what is going wrong. But the key to save has the format: growattyyyymmdd (with the actual date of today). 
But if specify that wrong you get a key error and not a set parameter error I think.

typxxi, 2023-09-28:
> Hi again, it is set to every minute, here is a screenshot from the actual Growatt server...
> [image]
> Is there another place as well? I don't know how to change the "datalogger register to 4"...  When I try to "read" the value 4 from the registry, I get is this:
> [image]
> Am I in the wrong place? :)

In that screenshot you can clearly see that you did not follow the instructions regarding the password.

growattyyyxmmdd

means exactly that while you were only using

20230814

Be aware  that you can easily kill the stick and inverter as seen before by not following the instructions

If you type register 41 instead of 4 and write a wrong value into it the stick or inverter might be gone. The last guy had written it into the register of the inverter cause he had messed it up. He had to send it in and it was a 288€ bill for the inverter and took about 2 weeks till he had figured out what to do and 10 day repair + shipping, in total 25 days no solar production which meant another 75€.

Be carefull and think twice if you need figures every single minute.
This is not a feature and growatt can overwrite it remotely back to 5 minutes. Every minute means 5 times the server load and it will end in a mess cause it is free and growatt might increase that to 10 minutes to reduce their server load..

Snille, 2023-09-28:
Oh, I had completely forgot about this. I kind of gave up in the end... 
Anyway, as you wrote I had the wrong password, but I figured out later on... So this is where I gave up:

I can set the "interval" (register 4) to 1. I can "read" the register and see that it is set to 1. It is the same value that is possible to set on the Datalogger (local web interface (my first picture), because it also gets updated when updating on the growatt web and vise versa). So, that is not an issue...
Also, I do see the in the grott log, data coming in about every minute (at least more often then every 5 minutes) So, the update interwall is working from the datalogger is my conclusion.
However, HA still only gets an update every 5 minutes (same as PVOutput). 

That's as far as I've manage to figure out. :)

johanmeijer, 2023-10-01:
FO HA I can not help you. The MQTT message are sent as often as the inverter/datalogger is sending data. In your case 1 per minute. 

PVOutput interval is always at 5 minutes. That is a PVOutput limit. It does not matter how often you load the data to PVOutput.

closed 2023-10-02 by Snille

typxxi, 2023-10-03:
Did you uninstall previous growatt integration in HA ?
If you can watch the update time in the dataloggers screen (web) updated every minute then the data will be transmitted and the graphs direction and level might be update too every minute (only if that changes due to cloud) , but without adding another datapoint till the 5 minutes are over. But that is the whole Growatt cloud world (web and shinephone app). 
And there is nothing more to achieve than 1 minute.