johanmeijer/grott #378: Inverter serial NWCRBJK003 - Missing Fields

danmiles86, 2023-08-01:
Sorry if I'm just being a noob here but I'm missing many fields and I'm guessing its because its not understaind my inverter serial model?

From log:
	 - Possible Inverter serial NWCRBJK003
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX

Model as per server.growatt.com
         Device Type：SPH3~6k TL BL

Is there an easy modification I can do to make it use a more appriopriate template?
TIA

johanmeijer, 2023-08-02:
Try it with invtype=sph in the [general] section of the grott.ini.

twisty, 2024-05-20:
I have the same inverter. Try using `invtype=SPH` (note case).

johanmeijer, 2024-05-21:
It will work with both upper en lowercase (tested in 2.8.3).

danmiles86, 2024-05-22:
@johanmeijer can I ask what the env command would be for this? ginvtype = "sph"?

Been working great since August but trying to redeploy in docker and having a few challenges.

johanmeijer, 2024-06-18:
It should be ginvtype = "sph" (or "SPH") indeed.