johanmeijer/grott #379: Incorrect date

farsonic, 2023-08-01:
I've just tried moving from a dedicated Raspberry PI for grott to a docker container on a server. I'm getting the following appear in my logs, where is this date coming from? 

- Growatt new layout processing
- decrypt : True
- offset : 6
- record layout : T06NNNNX

- Grott data record date/time processing started
- date-time: 2000-06-16T09:32:16

Systems "look" to have the correct date and the docker container has the correct date. Some of the records being sent are correct though.

johanmeijer, 2023-08-02:
Normally the data is coming from the datalogger (is part of the data record being sent and grott is decoding)  and is set (sent from) the growatt server.

farsonic, 2023-08-02:
ok, I'm going to revert to the Raspberry PI and have a look at the dates being presented in the logs there.