johanmeijer/grott #380: Growatt MOD 10kXTL3 wrong Values

Kjxll15, 2023-08-01:
Hi,

I have a Growatt MOD 10kTL3 with an ARK storage. I set up the Grott Edge addon in Homeassistant and the Grott integration. In the log file of the addon, I can see that data from two devices is always recognized alternately. There is the smart meter which is connected to the inverter and then the inverter itself. The data from the smart meter is correctly encrypted and passed on to homeassistant. However, the inverter data is incorrect. I have already tried to insert the adapted record layout from issue [81](https://github.com/johanmeijer/grott/issues/81), which was created for the MOD 9000TL3, into the used one, but the result is the same. What am I doing wrong?

The log of the layout recognition:
```
- Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:

```
The start log:

``` 

Grott settings:
_Generic:
	version:             	 2.7.8
	verbose:             	 True
	trace:               	 False
	config file:         	 /config/grott/grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 default
	invtypemap:          	 {'QDLXXXXXX': 'tl3'}
	include_all:         	 True
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 26
	offset:              	 6
	inverterid:          	 QDLXXXXXX
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 172.30.33.0
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 False
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 addons
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 XX.XX.XX.XXX
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 True
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 1
	pvsystemid:          	 systemid1
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 True
	extname:            	 grott_ha
	extvar:             	 {'ha_mqtt_host': 'core-mosquitto', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'addons', 'ha_mqtt_password': 'Deiji3Iech4shieVaesho3Airie5yiiquahghaiQueuShaf0ees2voRai3seeCei', 'ha_mqtt_retain': True}
Grott proxy mode started
	 **********************************************************************************
	 - Grott - libscrc not installed, no CRC checking only record validation on length!
	 **********************************************************************************
Hostname : 8ce4abae-grott-current
IP :  172.30.33.10 , port :  5279

Kjxll15, 2023-08-01:
This is the encrypted data:

```
 - date-time:  2023-08-01T11:55:00
	 - Grott values retrieved:
		 -  datalogserial        :  XGDXXXXX
		 -  pvserial             :  QDLXXXXXX
		 -  recortype1           :  3000
		 -  recortype2           :  3124
		 -  pvstatus             :  1
		 -  pvpowerin            :  1651.9
		 -  pv1voltage           :  237.3
		 -  pv1current           :  1.6
		 -  pv1watt              :  379.6
		 -  pv2voltage           :  454.4
		 -  pv2current           :  2.8
		 -  pv2watt              :  1272.3
		 -  pv3voltage           :  0.0
		 -  pv3current           :  0.0
		 -  pv3watt              :  0.0
		 -  pvpowerout           :  58982.4
		 -  pvfrequentie         :  21.2
		 -  pvgridvoltage        :  408.0
		 -  pvgridcurrent        :  405.2
		 -  pvgridpower          :  26869760.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  2346188.8
		 -  pvgridvoltage3       :  380.1
		 -  pvgridcurrent3       :  2.5
		 -  pvgridpower3         :  338473779.2
		 -  Vac_RS               :  5.1
		 -  Vac_ST               :  0.0
		 -  Vac_TR               :  227.9
		 -  pvenergytoday        :  224.1
		 -  pvenergytotal        :  1.2
		 -  totworktime          :  0.1
		 -  epv1today            :  4.3
		 -  epv1total            :  168.4
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  epvtotal             :  644.5
		 -  pvtemperature        :  34.8
		 -  pvipmtemperature     :  51.3
		 -  pvboottemperature    :  33.5
		 -  temp4                :  0.0
		 -  uwBatVolt_DSP        :  38.5
		 -  pbusvolt             :  3224
		 -  nbusvolt             :  3222
```

johanmeijer, 2023-08-02:
Can you sent the original growatt data from the log (the data with /x in it). hen I can test it. 

Can you also sent the log information about whit the information on the layout being used?

Kjxll15, 2023-08-02:
```
		 \x72\x3f\x24\x63\x8b\x36\x46\x0c\x66\xb3\x68\xb0\x74\x47\x72\x0b\x2f\xa1\x3d
		 \xb4\x47\x72\x6f\x77\x48\x5a\x74\x47\x72\x6f\x7b\x90\x78\x9a\x15\x8a\x6f\x77
		 \x61\x74\x74\x47\x73\x4b\x02\x99\x74\x74\x44\x9a\x3a\x5f\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x21\xe9
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 05fe00060241010458474436434c5247484300000000000000000000000000000000000000005
		 1444c31434b4230484b0000000000000000000000000000000000000000170802170625020bb8
		 0c3400010000000000ba000000000000016300000000000000000000000000000000000000000
		 0000000113a000000470000116113850915000800000744092600080000075109290009000008
		 3e0fca0fbe0fea000000000000004a000010f0001d8314000000a700000a4d00000a460000002
		 50000028500000083000007c100000000000000000000000600000074000000290000038c0000
		 007b000006d60000000000000000000000a8000000005ba80000000000000000190b012201cf0
		 10c0000015b0c840c854e200004000186a0000000000000000000000000000000000000000000
		 00003c00000000000100000000000010f00000009d0c350cb10000002c000002ed00000036000
		 003450000000000000000000009f5000000750000066700000000000000000000000000000000
		 000000000000000000000000000000000000000027100000190b0000000002010000000052cb0
		 015005019090c5e000c000c011c00000000118a00000000000002ed00000345000000030cab00
		 1f0019000000e900d20001000f0000000000010000002f002f000000000000000000000000000
		 000000000000000020000000000505302ff42017e09c409c40000006458c049c000000000292e
		 000000000cf10cee52f8000000000000012475f8000003e855280000000000000000000000000
		 000000000005386
	 - Possible Inverter serial QDLXXXXXX
	 - Matched inverter serial to inverter type tl3
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNXTL3
	 - Grott data record date/time processing started
	 - date-time:  2023-08-02T23:06:37
	 - Grott values retrieved:
		 -  datalogserial        :  XGD6XXXXX
		 -  pvserial             :  QDLXXXXX
		 -  recortype1           :  3000
		 -  recortype2           :  3124
		 -  pvstatus             :  1
		 -  pvpowerin            :  0.0
		 -  pv1voltage           :  18.6
		 -  pv1current           :  0.0
		 -  pv1watt              :  0.0
		 -  pv2voltage           :  35.5
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pv3voltage           :  0.0
		 -  pv3current           :  0.0
		 -  pv3watt              :  0.0
		 -  pvpowerout           :  58982.4
		 -  pvfrequentie         :  21.1
		 -  pvgridvoltage        :  404.2
		 -  pvgridcurrent        :  403.0
		 -  pvgridpower          :  26699366.4
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  484966.4
		 -  pvgridvoltage3       :  433.6
		 -  pvgridcurrent3       :  2.9
		 -  pvgridpower3         :  219912601.6
		 -  Vac_RS               :  16.7
		 -  Vac_ST               :  0.0
		 -  Vac_TR               :  263.7
		 -  pvenergytoday        :  263.0
		 -  pvenergytotal        :  3.7
		 -  totworktime          :  0.1
		 -  epv1today            :  13.1
		 -  epv1total            :  198.5
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  epvtotal             :  641.1
		 -  pvtemperature        :  29.0
		 -  pvipmtemperature     :  46.3
		 -  pvboottemperature    :  26.8
		 -  temp4                :  0.0
		 -  uwBatVolt_DSP        :  34.7
		 -  pbusvolt             :  3204
		 -  nbusvolt             :  3205
	 - MQTT jsonmsg: 
			 {"device": "QDL1XXXXX", "time": "2023-08-02T23:06:37", "buffered": "no",
			 "values": {"datalogserial": "XGD6XXXXXX", "pvserial": "QDLXXXXXXX",
			 "recortype1": 3000, "recortype2": 3124, "pvstatus": 1, "pvpowerin": 0,
			 "pv1voltage": 186, "pv1current": 0, "pv1watt": 0, "pv2voltage": 355,
			 "pv2current": 0, "pv2watt": 0, "pv3voltage": 0, "pv3current": 0, "pv3watt":
			 0, "pvpowerout": 589824, "pvfrequentie": 2110, "pvgridvoltage": 4042,
			 "pvgridcurrent": 4030, "pvgridpower": 266993664, "pvgridvoltage2": 0,
			 "pvgridcurrent2": 0, "pvgridpower2": 4849664, "pvgridvoltage3": 4336,
			 "pvgridcurrent3": 29, "pvgridpower3": 2199126016, "Vac_RS": 167, "Vac_ST":
			 0, "Vac_TR": 2637, "pvenergytoday": 2630, "pvenergytotal": 37,
			 "totworktime": 645, "epv1today": 131, "epv1total": 1985, "epv2today": 0,
			 "epv2total": 0, "epvtotal": 6411, "pvtemperature": 290, "pvipmtemperature":
			 463, "pvboottemperature": 268, "temp4": 0, "uwBatVolt_DSP": 347, "pbusvolt":
			 3204, "nbusvolt": 3205}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - PVOut: Update refused for QDL1CKB0HK due to time limitation
```

Kjxll15, 2023-08-11:
Have you had time to take a look?

johanmeijer, 2023-08-11:
No not yet sorry. Can you sent a copy of the grott log with the original data (the part with the /x data above but then completely). That makes it more easy for me to debug.

johanmeijer, 2023-08-13:
@Kjxll15 

I did not realized that the definition file for the MOD is already in the Grott 2.8.2. version. It seems to be the same as the MIN layout. 

If you specify invtype=min in the grott.ini [Generic] section you probably see usable data.

Kjxll15, 2023-08-13:
Okay, I have done. It's now an other output but doesn't work as expected either. Here's the log:

```
Growatt packet received: 		 <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.10', 5279), raddr=('192.168.1.64', 5889)> 	 - Data less then minimum record length, data not processed 	 - Growatt packet received: 		 <socket.socket fd=4, family=2, type=1, proto=0, laddr=('172.30.33.10', 33154), raddr=('47.91.67.66', 5279)> 	 - Growatt original Data: 		 \x04\xf7\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x22\x38\x26\x00\x3a\x2c\x77 		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72 		 \x3e\x33\x2d\x45\x37\x0c\x30\x5f\x3f\x2a\x74\x74\x47\x72\x6f\x77\x61\x74\x74 		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7c\x79\x55\x6d\x76\x75\x6a 		 \xcc\x78\x73\x72\x6e\x77\x61\x43\x6b\x4e\x75\x6f\x78\x61\x74\x79\xcd\x62\xcd 		 \x77\x78\x74\x74\x6e\xe7\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47 		 \x72\x6f\x77\x61\x74\x0f\x44\x72\x6f\x77\xf1\x74\x74\x3b\xdc\x7c\xf2\x68\x66 		 \x74\x6a\x72\x6f\x5f\xb0\x7d\x52\x47\x5f\x6f\x77\x48\x5f\x7d\x60\x72\x43\x77 		 \x61\x5c\x31\x48\xba\x60\xca\x6e\x90\x74\x47\x77\x8a\x77\x61\x74\x74\x47\x72 		 \xef\x9f\x61\x4d\x10\xc2\x72\x6f\x76\x7d\x74\x74\x5f\x08\x6f\x77\x79\x09\x74 		 \x47\x72\x28\x77\x61\x71\xa0\x47\x72\x6f\x98\x61\x74\x66\xee\x72\x6f\x77\x61 		 \x74\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x72\xdf\x77\x61\x74\x29\x47\x72\x67 		 \x24\x61\x74\x74\xfb\x72\x6f\x67\x23\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47 		 \x73\x59\x77\x61\x74\x74\x74\x02\x67\x7d\x60\x22\x73\xcb\x72\x6f\x6e\x61\x75 		 \xf7\x45\x79\x6e\x11\x61\x74\x75\xc9\x7e\xca\x7b\x3b\x3a\x54\x47\x6d\x6f\x76 		 \xe7\xd4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72 		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74 		 \x3c\x71\x6f\x77\x60\x63\x78\x72\x7e\xde\x77\x61\x74\x58\x47\x72\x6a\x22\x61 		 \x74\x74\x0b\x72\x6f\x72\x8c\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x65\x82 		 \x77\x61\x74\xcf\x47\x72\x60\xef\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47 		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74 		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x76\x75\x61\x74\x74\x4c\x70\x6e\x77 		 \x61\x76\xcb\x15\x27\x6f\x26\x61\x15\x6d\x47\x7e\x35\x77\x4b\x74\x5e\x46\x42 		 \x6f\x77\x61\x74\x37\x61\x72\x6f\x77\x61\x74\x74\x42\x27\x6f\x77\x64\x99\x74 		 \x47\x72\x6d\x7b\xc7\x74\x71\x47\x6b\x6f\x77\x61\x81\x74\x9a\x72\x6d\x77\x6c 		 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x68\x72\x40\x77\x61\x74\x74\x47\x72\x6f 		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61\x74\x74\x47 		 \x72\x0e\x25\xe1\x88\xa2\x46\xf1\x66\xb3\x68\xb0\x74\x47\x72\x0b\x2f\xa1\x3d 		 \xb4\x47\x72\x6f\x77\x48\x78\x74\x47\x72\x6f\x7b\x80\x78\xa3\x15\xf2\x6f\x77 		 \x61\x74\x74\x47\x73\x4b\x02\x99\x74\x74\x44\x9a\x3a\x5f\x61\x74\x74\x47\x72 		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xcd\xbd 	 - Grott automatic protocol detection 	 - Grott data record length 585 	 - layout : T060104XMIN 	 - no matching record layout found, try generic 	 - no matching record layout found, standard processing performed 	 - Record layout used : none 	 - Growatt data decrypted V2 	 - Grott Growatt data decrypted 	 - Growatt plain data: 		 04f700060241010458474436434c5247484300000000000000000000000000000000000000005 		 1444c31434b4230484b000000000000000000000000000000000000000017080d121f19020bb8 		 0c3400010000371f0907000f00000d8a10a200190000299500000000000000000000000000000 		 00000007b030000009000007cae13850912002d000028d10926002d0000292b0927002c000028 		 450fc80fbd0fe4000005e500000000000080e8003964850000011c0000187a0000187d0000004 		 7000005d4000000ef000012a9000000000000000000000003000000b00000005d000008530000 		 00bc00001042000000000000000000000136000000003370080a0156078c000019000183020b0 		 1660000018e0ca50c5a4e20001f000186a0000000000000000000000000000000000000000000 		 00003c0000000000010000000000007b03000001170c350cb10000002c000005550000004c000 		 005ed0000000000000000000017ed000000bb00000f9800000000000000000000000000000000 		 00000000000000000000000000000000000000002710000019020000000b0201000002bf52550 		 051006119000c5a002a002a01300000000043260000000000000555000005ed000000020ca600 		 050019000000f500dd0002000d0000000000010000002f002f000000000000000000000000000 		 000000000000000020000000000615280fcd6018309c409c40000006458c049c000000000290c 		 000000000ce10cd75280000000000000012475f8000003e855280000000000000000000000000 		 00000000000bfd2 	 - Grott data ack record or data record not defined no processing done

```

Kjxll15, 2023-08-13:
Okay, I have done. Now it has an other output, but it doesn't work as expected eiher. Here is the log: 

```
 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('xxx.xx.33.10', 5279), raddr=('192.168.1.64', 5889)>
	 - Data less then minimum record length, data not processed
	 - Growatt packet received:
		  <socket.socket fd=4, family=2, type=1, proto=0, laddr=('xxx.xx.33.10', 33154), raddr=('xx.xx.67.66', 5279)>
	 - Growatt original Data:
		 \x05\x12\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x22\x38\x26\x00\x3a\x2c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3e\x33\x2d\x45\x37\x0c\x30\x5f\x3f\x2a\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7c\x79\x55\x5b\x5c\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x62\xcf\x4f\x8f\x6f\x72\x61\x74\x70\x39\x62\xfa
		 \x77\x6a\x74\x74\x55\x4f\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x68\x46\x72\x6f\x77\x19\x74\x74\x67\x34\x7c\xf2\x68\x7e
		 \x74\x4a\x72\x6f\x7c\xa1\x7d\x68\x47\x7e\x6f\x77\x6b\x9a\x7d\x59\x72\x62\x77
		 \x61\x7f\xae\x48\xca\x60\xd9\x6e\xa2\x74\x47\x72\x6f\x77\x61\x75\x29\x47\x72
		 \x75\xd3\x61\x4d\x1d\xa1\x72\x6f\x76\x41\x74\x74\x5f\x0c\x6f\x77\x79\x0a\x74
		 \x47\x72\x28\x77\x61\x71\xa0\x47\x72\x6f\x87\x61\x74\x66\xed\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x72\xdf\x77\x61\x74\x29\x47\x72\x67
		 \x24\x61\x74\x74\x87\x72\x6f\x67\x27\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x73\x58\x77\x61\x74\x74\x74\x02\x68\x3e\x62\xfc\x7d\x44\x72\x6f\x6f\x8f\x75
		 \x04\x45\x7c\x6e\x20\x61\x74\x75\xca\x7e\xe8\x7b\x06\x3a\x54\x47\x7a\x6f\x76
		 \xe7\xd4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
		 \x5d\xd6\x6f\x77\x60\x68\x78\x72\x7e\xde\x77\x61\x74\x5b\x47\x72\x6a\x2f\x61
		 \x74\x74\x0b\x72\x6f\x72\x8c\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x65\x9d
		 \x77\x61\x74\xcb\x47\x72\x60\xeb\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\x98\x61\x74\x74\x47\x70\x6e\x77
		 \x61\x74\x74\x14\x7b\x6f\x71\x61\x2a\x6c\xaa\x7e\x3e\x77\x64\x74\x71\x46\x4c
		 \x6f\x77\x61\x74\x71\xed\x72\x6f\x77\x61\x74\x74\x42\x2a\x6f\x77\x64\x99\x74
		 \x47\x72\x6c\x7b\xfd\x74\x66\x47\x52\x6f\x77\x61\x82\x74\x9a\x72\x6d\x77\x6c
		 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x68\x72\x40\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x61\x74\x74\x47
		 \x72\x31\x24\x55\x8b\xb0\x46\xf5\x66\xb3\x68\xb0\x74\x47\x72\x0b\x2f\xa1\x3d
		 \xb4\x47\x72\x6f\x77\x48\x78\x74\x47\x72\x6f\x7b\x98\x78\x82\x14\x52\x6f\x77
		 \x61\x74\x74\x47\x73\x4b\x02\x99\x74\x74\x44\x9a\x3a\x5f\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x5a\x97
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104XMIN
	 - no matching record layout found, try generic
	 - no matching record layout found, standard processing performed
	 - Record layout used :  none
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 051200060241010458474436434c5247484300000000000000000000000000000000000000005
		 1444c31434b4230484b000000000000000000000000000000000000000017080d122933020bb8
		 0c340001000016bb08fd00050000047e1095000b0000123d00000000000000000000000000000
		 00000001c0100000078000020461385090a000d00000bc0091c000c00000aee091e000d00000b
		 da0fb80fae0fd6000000000000015d00001aa4003969e6000001200000187e0000187e0000004
		 7000005d4000000f0000012aa000000000000000000000003000000b00000005d000008530000
		 00c000001046000000000000000000000137000000003370074903880903000018ee0170020e0
		 1570000018d0c870c674e200008000186a0000000000000000000000000000000000000000000
		 00003c0000000000010000000000001aa40000011c0c350cb10000002f000005580000004c000
		 005ed0000000000000000000017f2000000bf00000f9c00000000000000000000000000000000
		 00000000000000000000000000000000000000002710000018ef0000000002010000000053090
		 006005e18ed0c5100050005013e0000000005aa0000000000000558000005ed000000030c9c00
		 120020000000f600dd0002000d0000000000010000002f002f000000000000000000000000000
		 0000000000000000200000000005e5334ffc4018709c409c40000006458c049c000000000290c
		 000000000cf90cf65320000000000000012475f8000003e855280000000000000000000000000
		 0000000000028f8
	 - Grott data ack record or data record not defined no processing done
```

johanmeijer, 2023-08-13:
Which github branch or docker image are you using?

Kjxll15, 2023-08-14:
I'm using the Gritty Edge Homeassistant Addon. This is my installed version: edge-5821354480

johanmeijer, 2023-08-14:
Yes that uses the docker image that does not have the min layout in it. Maybe @egguy can help here?

geve09, 2023-08-22:
On your data I used record layout :  T06NNNNXMOD (experimental adapted T06NNNNX). 
x underscore with a number I did not find a field name for.
The result seem logic if I look at VAC-R/S/T. 
Maybe you can check some of the fields.

		 -  x_Ordernumber        :  1298
		 -  x_MessageTypePart1   :  0006
		 -  x_MessageLength      :  577
		 -  x_MessageTypepart2   :  0104
		 -  datalogserial        :  X.....
		 -  x_026_spare          :  0000000000000000000000000000000000000000
		 -  pvserial             :  QD.....
		 -  x_096_spare          :  0000000000000000000000000000000000000000
		 -  x_148                :  02 (2)
		 -  recortype1           :  0bb8 (3000)
		 -  recortype2           :  0c34 (3124)
		 -  pvstatus             :  1
		 -  pvpowerin            :  581.9
		 -  pv1voltage           :  230.1
		 -  pv1current           :  0.5
		 -  pv1watt              :  115.0
		 -  pv2voltage           :  424.5
		 -  pv2current           :  1.1
		 -  pv2watt              :  466.9
		 -  x_202_spare          :  00000000000000000000000000000000
		 -  pvpower_234          :  716.9
		 -  x_242_volt?          :  120
		 -  pvpowerout           :  826.2
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage1       :  231.4
		 -  pvgridcurrent1       :  1.3
		 -  pvgridpower1         :  300.8
		 -  pvgridvoltage2       :  233.2
		 -  pvgridcurrent2       :  1.2
		 -  pvgridpower2         :  279.8
		 -  pvgridvoltage3       :  233.4
		 -  pvgridcurrent3       :  1.3
		 -  pvgridpower3         :  303.4
		 -  Vac_RS               :  402.4
		 -  Vac_ST               :  401.4
		 -  Vac_TR               :  405.4
		 -  x_322_spare          :  000000000000015d0000
		 -  x_342_?              :  1aa4 (6820)
		 -  totworktime          :  522.6
		 -  pvenergytoday        :  28.8
		 -  pvenergytotal        :  627.0
		 -  epvtotal             :  627.0
		 -  epv1today            :  7.1
		 -  epv1total            :  149.2
		 -  epv2today            :  24.0
		 -  epv2total            :  477.8
		 -  x_410_spare          :  000000000000000000000003000000b00000005d00000853
		 -  x_458_?              :  000000c0 (192)
		 -  x_466_?              :  00001046 (4166)
		 -  x_474_spare          :  0000000000000000
		 -  x_490_?              :  00000137 (311)
		 -  x_498_?              :  0000 (0)
		 -  x_502_?              :  00003370 (13168)
		 -  x_510_?              :  0749 (1865)
		 -  x_514_?              :  0388 (904)
		 -  x_518_?              :  0903 (2307)
		 -  x_522_?              :  000018ee (6382)
		 -  pvtemperature        :  36.8
		 -  pvboottemperature    :  52.6
		 -  temp4                :  34.3
		 -  uwBatVolt_DSP        :  0.0
		 -  pvipmtemperature     :  39.7
		 -  pbusvolt             :  320.7
		 -  nbusvolt             :  317.5
		 -  x_558_?              :  4e20 (20000)
		 -  x_562_?              :  0008 (8)
		 -  x_566_?              :  000186a0 (100000)
		 -  x_574_spare          :  0000000000000000000000000000000000000000
		 -  x_614_?              :  0000003c (60)
		 -  x_622_spare          :  00000000000100000000
		 -  x_642_?              :  00001aa4 (6820)
		 -  x_650_?              :  0000011c (284)
		 -  x_658_?_0c350cb1     :  0c350cb1 (204803249)
		 -  x_666_spare          :  0000002f000005580000004c000005ed0000000000000000
		 -  x_714_?              :  000017f2 (6130)
		 -  x_722_?              :  000000bf (191)
		 -  x_730_?              :  00000f9c (3996)
		 -  x_738_spare          :  00000000000000000000000000000000000000000000000000000000000000000000
		 -  x_806_?              :  00002710 (10000)
		 -  x_814_?              :  000018ef (6383)
		 -  x_822_spare          :  0000000002010000000053090006005e18ed0c5100050005013e0000000005aa0000000000000558000005ed000000030c9c
		 -  x_922_spare          :  00120020000000f600dd0002000d0000000000010000002f002f000000000000000000000000000000000000000000020000
		 -  x_1022_spare         :  0000005e5334ffc4018709c409c40000006458c049c000000000290c000000000cf90cf65320000000000000012475f8
		 -  x_1118_?             :  000003e8 (1000)
		 -  x_1126_?             :  5528 (21800)
		 -  x_1130_?             :  0000 (0)
		 -  x_1134_spare         :  00000000000000000000000000000000
		 -  x_1166_?_response    :  28f8 (10488)

Kjxll15, 2023-08-23:
Hi, i would like to help but i don't know exactly how to assign these values, cause the data is only displayed in the Growatt Portal on average for 5 minutes at a time and i have not found a way to download the transmitted data as a .csv, for example . Do you have any tips on how I can help you better?

Kjxll15, 2023-09-10:
Hi, Anything new?

isakrubin, 2023-09-11:
Got the same model and would be happy to help testing out

SvenBaron, 2023-09-11:
Here Original Data from a MOD10KTL3-XH(BP) inverter .... Missing all Battery Values and cant get them decoded..
Help would be really appreciated !

- Growatt original Data:
		 \x00\xa6\x00\x06\x03\x3f\x01\x04\x1f\x35\x2b\x41\x22\x38\x35\x70\x20\x3a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2c\x2d\x2c\x44\x30\x04\x3e\x5f\x4f\x56\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7d\x7f\x40\x53\x6a\x74\x6a
		 \xcc\x78\x73\x72\x6e\x77\x60\x76\x6a\x5f\x9f\x6f\x4d\x61\x74\xe4\xd6\x61\xfb
		 \x77\x5b\x74\x74\x36\xff\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x75\x74\xe5\x72\x6f\x74\x5e\x74\x75\x44\xfd\x7c\xf0\x68\xe2
		 \x74\x1e\x72\x6f\x22\x31\x7d\xfe\x47\x2f\x6f\x77\x39\xc2\x7d\xd3\x72\x38\x77
		 \x61\x27\x20\x57\xe6\x7f\xf9\x71\xe0\x74\x47\x72\x6f\x77\x61\x81\x45\x47\x72
		 \x64\x06\x61\x63\xbe\x97\x72\x6f\x77\x91\x74\x74\x54\x7b\x6f\x77\x72\xd6\x74
		 \x47\x72\x28\x77\x61\x7d\xb6\x47\x72\x6f\xba\x61\x74\x7d\xa7\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6c\x77\x61\x74\xc9\x47\x72\x61
		 \x23\x61\x74\x74\x6d\x72\x6f\x74\x8e\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x73\x7b\x77\x61\x74\x74\xb8\x88\x6e\x51\x60\x59\x75\x06\x72\x6e\x69\x6e\x76
		 \x69\x45\x2c\x6e\x90\x61\x74\x75\xa5\x7d\x75\x79\x94\x3a\x54\x47\x33\x6f\x76
		 \xe7\xd4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
		 \x4c\x03\x6f\x77\x61\x93\x78\x72\x7e\xde\x77\x61\x74\x69\x47\x72\x6e\x28\x61
		 \x74\x74\x0d\x72\x6f\x75\xa0\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x60\x2b
		 \x77\x61\x74\x5d\x47\x72\x6c\x98\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x72\xdb\x61\x74\x74\x4c\x70\x6e\x77
		 \x61\x76\xcb\xe4\x2e\x6f\x76\x61\x10\x6a\x55\x7d\x70\x77\x60\x74\x74\x46\x11
		 \x6e\xbc\x61\x74\x74\x47\x72\x6f\x76\xad\x74\x74\x46\x2d\x6f\x77\x63\xb5\x74
		 \x47\x72\x6d\x79\x93\x74\x77\x47\x6b\x6e\x54\x60\x4e\x75\x49\x72\x6c\x77\x70
		 \x74\x10\x47\x16\x6f\x75\x61\x74\x74\x80\x72\xa7\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73\x6f\x73\x61\x74\x74\x47
		 \x72\x0b\xd4\x3d\x74\x7e\x47\x72\x6f\x77\x4b\x8c\x74\x47\x72\x0b\x61\x51\x66
		 \x54\x47\x72\x6f\x77\x63\x2c\x74\x47\x72\x6e\x7a\x5d\x79\x66\x53\x88\x6f\x77
		 \x61\x74\x74\x47\x73\x75\x77\x61\x74\x74\x44\x9a\x3e\x37\x3c\x2a\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x7e\xdd\x7a\x4f\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x16\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x44\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x2c\x2c\x1f\x2a\x6f\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a
		 \x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\xaf\xc2

johanmeijer, 2023-09-11:
Can you try it with invtype=min in the grott.ini. the layout of min and mod type inverters seems to be the same.

SvenBaron, 2023-09-11:
With MIN Setting

 - Growatt original Data:
		 \x00\x26\x00\x06\x03\x3f\x01\x03\x1f\x35\x2b\x41\x22\x38\x35\x70\x20\x3a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x2c\x2d\x2c\x44\x30\x04\x3e\x5f\x4f\x56\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7d\x7f\x48\x72\x75\x74\x61
		 \x74\x74\x3b\x72\x6e\x77\xdc\x8b\x8b\x47\x16\x6f\x77\x46\x64\x74\x46\xf4\xcf
		 \x71\x21\x30\x3a\x76\x5c\x5f\x77\x3b\x36\x10\x25\x72\x63\x77\x63\x74\x74\x41
		 \x32\x6f\x4b\x61\x48\x74\x1d\x72\x35\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x75\x47\x16\x6f\x77\x61\x74\x54\x67\x52\x3f\x21
		 \x41\x3d\x1a\x31\x17\x1d\x03\x04\x06\x54\x67\x72\x6f\x62\x79\x7c\x77\x40\x95
		 \x6f\x7e\x61\x7f\x74\x48\x72\x6f\x77\x7b\x74\x75\x4b\x01\x7c\x04\x73\xfa\x60
		 \x59\x75\x6f\x64\x12\x66\xfa\x53\x6c\x68\x77\x72\x07\x66\xc9\x66\x71\x7a\x5b
		 \x65\x6a\x55\xef\x7c\xe5\x61\xec\x74\x43\x72\x7f\x77\x62\x74\x77\x47\x71\x6f
		 \x74\x61\x77\x74\x57\x72\x6c\x77\x62\x74\x77\x56\x6c\x6f\x77\x25\x3a\x15\x26
		 \x40\x5d\x42\x50\x44\x44\x76\x40\x6e\x46\x61\x74\x74\x47\x61\xf3\x77\x53\x64
		 \x73\x57\xd4\x60\x6f\x6f\x0d\x74\x53\x72\x6a\x67\x36\x7b\xe4\x61\x92\x49\x97
		 \x47\xa1\x53\x2e\x55\x06\x50\x47\x74\x7e\x47\x72\x6e\x93\x61\x8b\x3a\x67\x72
		 \x90\x39\x41\x74\x8b\x09\x52\x6f\x88\x2f\x54\x73\x46\x72\x6f\x78\x60\x74\x10
		 \x47\x73\x6c\x9f\x61\x74\x7f\xff\x7e\x5b\x77\x61\x37\x2e\x0a\x42\x2b\x34\x2d
		 \x44\x4c\x70\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6e\x83\x61\x74\x75\xd7\x72\x6e\x77\x61\x5c\x48\x47
		 \x73\x6d\x2f\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x51\xda\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x10\x47\x78\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x23\x72\x0b\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x2c\x2f\x2c\x44\x44\x77
		 \x42\x5f\x45\x52\x45\x45\x77\x43\x5f\x45\x61\x74\x2e\x02\x11\x0e\x77\x61\x22
		 \x30\x26\x13\x6f\x7a\x70\xca\x74\x4a\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x67
		 \x61\x74\x74\x47\x72\xf9\x77\x60\x75\x71\x47\x78\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x54\x75\x47\x16\x78\x7e\x6a\x7b\x74\x5e\x7e\xdd\x7a\x4f\x74\x74
		 \x1f\x2a\x37\x2f\x61\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39
		 \x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x27\x8c
	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060103XMIN
	 - no matching record layout found, try generic
	 - Record layout used :  T060103XMIN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00260006033f010358474436434c4137525500000000000000000000000000000000000000004
		 35a4d3044434c303837000000000000000000000000000000000000000017090b0f001a030000
		 007c000100bdffff006400002710000186a00640444e312e30005a426462000c0002000006400
		 03c003c005a005a00000000000000000000000000000000000100640000000020202050562049
		 6e766572746572202000001518080307e70009000b000f0000001a00010c731373128e141e070
		 01373128e141e07001373128e141e0d3a111e129d139200980004001000030003000300030003
		 0010000300030003111e0000444e61613232353130303132013100000000139c0032100710a60
		 f180e790014000510570f9026e026e026d5276927692726000a000001e400ff4e2000ff4e2000
		 ff4e2000ff4e20070100000f010064000103e800000bb80c340000435a4d3044434c303837000
		 0000000000000000000000000000000000000000001f40000019000010000283c000102580000
		 000000000000000016a8000000000000000000000064000a00000000000000000000000000000
		 00000000064006400000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000043584d3030303030323331313031303200005a456361000056446161000d11be
		 000d0000000000000000001000000000009600010105000a00000000000000000000200100641
		 7090b0f00190cb20d2e0000585858580058585858585858585858585858585858585858585858
		 58585858585858585800000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000050ed
- Grott data ack record or data record not defined no processing done
-

SvenBaron, 2023-09-11:
On Generic some Values are ok ... : 
But still missing Battery Values etc ....

	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00c10006033f010458474436434c4137525500000000000000000000000000000000000000004
		 35a4d3044434c303837000000000000000000000000000000000000000017090b0f1601030bb8
		 0c3400010000d4821954004a0000bb6d10b9000f0000191500000000000000000000000000000
		 0000000d298000003150000d159138a097c00490000453c0969004e0000496609750047000043
		 25106210551064000000000000c1fb0000109d0017fe460000016000001379000014110000009
		 700000a12000000ec000009ff000000000000000000000000000000030000012200000eb90000
		 0033000003f800000000000000000000018300000000fffa0071008f01a800011dc801f702510
		 1cd000001d80ef70ed14e200034000186a0000000000000000000000000000000000000000000
		 00003c000000000001000000000000109d000001550c350cb10000001d0000015f0000004b000
		 002c20000000000000000000012b200000032000003f800000000000000000000000000000000
		 0000000000000000000000000000000000000000271000001d640000000b0101000002bfa30c0
		 00100641dcb0efb00010000016001c700000000000001fe0000015f000002c2000000020ecf00
		 0300190114012b010200030011006400640002000000c700c8000000000000000000000000000
		 00000000000010004000000000064a30c000a000000002af80000006416301220000000000258
		 000000010d350d1014f0000000000000011a0000000003e851405d5e000000000000000000000
		 000000000000cb20d2e0000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000640000000000000000000000000000000000000003000
		 00000000000000000000000000000000000000000000000000000000000005858585800585858
		 58585858585858585858585858585858585858585858585858585858000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000000000000000000ff45
	 - Possible Inverter serial CZM0DCL087
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott data record date/time processing started
	 - date-time:  2023-09-11T15:22:01
	 - Grott values retrieved:
		 -  datalogserial        :  XGD6CLA7RU
		 -  pvserial             :  CZM0DCL087
		 -  pvstatus             :  1
		 -  pvpowerin            :  5440.2
		 -  pv1voltage           :  648.4
		 -  pv1current           :  7.4
		 -  pv1watt              :  4798.1
		 -  pv2voltage           :  428.1
		 -  pv2current           :  1.5
		 -  pv2watt              :  642.1
		 -  pvpowerout           :  5359.3
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  242.8
		 -  pvgridcurrent        :  7.3
		 -  pvgridpower          :  1772.4
		 -  pvgridvoltage2       :  240.9
		 -  pvgridcurrent2       :  7.8
		 -  pvgridpower2         :  1879.0
		 -  pvgridvoltage3       :  242.1
		 -  pvgridcurrent3       :  7.1
		 -  pvgridpower3         :  1718.9
		 -  totworktime          :  218.4
		 -  pvenergytoday        :  35.2
		 -  pvenergytotal        :  498.5
		 -  epvtotal             :  513.7
		 -  epv1today            :  15.1
		 -  epv1total            :  257.8
		 -  epv2today            :  23.6
		 -  epv2total            :  255.9
		 -  pvtemperature        :  50.3
		 -  pvipmtemperature     :  47.2
	 - MQTT jsonmsg: 
			 {"device": "CZM0DCL087", "time": "2023-09-11T15:22:01", "buffered": "no",
			 "values": {"datalogserial": "XGD6CLA7RU", "pvserial": "CZM0DCL087",
			 "pvstatus": 1, "pvpowerin": 54402, "pv1voltage": 6484, "pv1current": 74,
			 "pv1watt": 47981, "pv2voltage": 4281, "pv2current": 15, "pv2watt": 6421,
			 "pvpowerout": 53593, "pvfrequentie": 5002, "pvgridvoltage": 2428,
			 "pvgridcurrent": 73, "pvgridpower": 17724, "pvgridvoltage2": 2409,
			 "pvgridcurrent2": 78, "pvgridpower2": 18790, "pvgridvoltage3": 2421,
			 "pvgridcurrent3": 71, "pvgridpower3": 17189, "totworktime": 1572422,
			 "pvenergytoday": 352, "pvenergytotal": 4985, "epvtotal": 5137, "epv1today":
			 151, "epv1total": 2578, "epv2today": 236, "epv2total": 2559,
			 "pvtemperature": 503, "pvipmtemperature": 472}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	 - Grott extension processing ended :  0

Kjxll15, 2023-10-02:
Hi, anything new to this issue? I would like to help figuring out the layout type.

SvenBaron, 2023-10-02:
No , sadly no news to it . Grott still not decodes all values . Missing important ones like SoC and Total Solar ( daily etc .. )With best regardsSvenVon meinem iPhone gesendet

egguy, 2024-01-17:
@Kjxll15 I have updated the addons, you can use the new repo: https://github.com/egguy/ha-addons

The beta branch is the 2.8.2 (the 2.8.3 is on the https://github.com/egguy/ha-addons-edge, but should arrive soon in the main repo)