johanmeijer/grott #388: Incorrect data / mappings for Shine-Wifi-F data logger?

fhteagle, 2023-08-08:
Hello and thank you for this great piece of software. I have grottserver running in a docker container, and the Growatt data logger device successfully registers to it on my LAN, and data is pushed to local MQTT broker and Influxdb 2 container. However, I believe the data is being misinterpreted and many fields values are in error. 

```
- Grott values retrieved:
2023-08-07T23:20:21.001298631Z 		 -  datalogserial        :  ***datalogger_sn***
2023-08-07T23:20:21.001306512Z 		 -  pvserial             :  ***inverter_sn***
2023-08-07T23:20:21.001314754Z 		 -  recortype1           :  0
2023-08-07T23:20:21.001322409Z 		 -  recortype2           :  44
2023-08-07T23:20:21.001329905Z 		 -  pvstatus             :  12
2023-08-07T23:20:21.001354826Z 		 -  pvpowerin            :  3525836.8
2023-08-07T23:20:21.001363290Z 		 -  pv1voltage           :  0.0      *** voltage was 0 all day, even with PV production occurring ***
2023-08-07T23:20:21.001371081Z 		 -  pv1current           :  5.0    *** incorrect, solar panels are shaded and making 0A at this point ***
2023-08-07T23:20:21.001379494Z 		 -  pv1watt              :  0.0      *** wattage was 0 all day, even with PV production occurring ***
2023-08-07T23:20:21.001390017Z 		 -  pv2voltage           :  0.1
2023-08-07T23:20:21.001397827Z 		 -  pv2current           :  0.0
2023-08-07T23:20:21.001405615Z 		 -  pv2watt              :  0.0
2023-08-07T23:20:21.001413278Z 		 -  pvpowerout           :  33.0
2023-08-07T23:20:21.001422517Z 		 -  pvfrequentie         :  0.0
2023-08-07T23:20:21.001431306Z 		 -  pvgridvoltage        :  0.0
2023-08-07T23:20:21.001464882Z 		 -  pvgridcurrent        :  0.0
2023-08-07T23:20:21.001490910Z 		 -  pvgridpower          :  525.9    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001520162Z 		 -  pvgridvoltage2       :  10.0    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001545570Z 		 -  pvgridcurrent2       :  207.9    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001577619Z 		 -  pvgridpower2         :  0.0
2023-08-07T23:20:21.001608572Z 		 -  pvgridvoltage3       :  120.0    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001622044Z 		 -  pvgridcurrent3       :  600.0    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001630307Z 		 -  pvgridpower3         :  29.6    *** incorrect, unit is not connected to grid input ***
2023-08-07T23:20:21.001646228Z 		 -  pvenergytoday        :  1867776.2
2023-08-07T23:20:21.001673675Z 		 -  pvenergytotal        :  0.0
2023-08-07T23:20:21.001692373Z 		 -  totworktime          :  0.0
2023-08-07T23:20:21.001718702Z 		 -  pvtemperature        :  26.6
2023-08-07T23:20:21.001763021Z 		 -  isof                 :  275
2023-08-07T23:20:21.001771871Z 		 -  gfcif                :  2
2023-08-07T23:20:21.001794198Z 		 -  dcif                 :  11
2023-08-07T23:20:21.001823912Z 		 -  vpvfault             :  0
2023-08-07T23:20:21.001840738Z 		 -  vacfault             :  0
2023-08-07T23:20:21.001893157Z 		 -  facfault             :  0
2023-08-07T23:20:21.001928306Z 		 -  tmpfault             :  0
2023-08-07T23:20:21.001941621Z 		 -  faultcode            :  0
2023-08-07T23:20:21.001950315Z 		 -  pvipmtemperature     :  0.0
2023-08-07T23:20:21.001960303Z 		 -  pbusvolt             :  0.0      *** should be battery voltage????***
2023-08-07T23:20:21.002003213Z 		 -  nbusvolt             :  0.0
2023-08-07T23:20:21.002025701Z 		 -  epv1today            :  1.7
2023-08-07T23:20:21.002047961Z 		 -  epv1total            :  30.3
2023-08-07T23:20:21.002134162Z 		 -  epv2today            :  0.0
2023-08-07T23:20:21.002222296Z 		 -  epv2total            :  0.0
2023-08-07T23:20:21.002243400Z 		 -  epvtotal             :  0.0
2023-08-07T23:20:21.002252738Z 		 -  rac                  :  0
2023-08-07T23:20:21.002261026Z 		 -  eractoday            :  25
2023-08-07T23:20:21.002269342Z 		 -  eractotal            :  159
```

Here is the serial sticker of the data logger device. It is running firmware 1.7.6 if I recall correctly, from circa 2020 but I have made no attempt to update it at all.

[image]

A sanitized log from the past hour or so is available here (items with *** have been removed for security, but are valid values in the original log):
https://pastebin.com/fpix1M0w

Let me know what else you need for troubleshooting or comparison. I can compare the display of the unit to the 5 minute update ticks tomorrow to get any needed data points.

Thanks

cyberdiamond, 2023-08-08:
You need to set the inverter type to match your inverter in the grott.ini config file.
I think you need spf as from memory wifi-f is for spf?
eg:

invtype = spf

But for docker I think its ginvtype

fhteagle, 2023-08-08:
I guess I must have missed that option in the documentation...

I added 

```
- ginvtype=spf 
```

into my docker-compose environment variable stanza. That did seem to add/change quite a few fields in what is being recorded to influxdb. Several new ones seem to have more reasonable values. I will watch it carefully during tomorrow's solar production and make sure things match.

Thanks for the quick reply @cyberdiamond

fhteagle, 2023-08-08:
Okay ginvtype=spf is definitely an improvement. MQTT reports seem reasonable based on what I am doing with the inverter and PV. 

However, what is recorded to influxdb is way off. Does not match MQTT values, nor the inverter display. Examples: 

1. Influxdb batWatt is always a negative number and continuously increases, regardless of battery charging or discharging (meanwhile MQTT BatWatt was positive and reasonable for the load I put on the inverter, did not catch the MQTT report while charging....)
2. influxdb bat_Volt is always near ~5240, but inverter display was showing 50.6 volts and MQTT bat_Volt was 5055

Is there something else I need to change to get the right inverter type for data mapping to influxdb2?

johanmeijer, 2023-08-08:
Hmm that is strange. Grott uses the same data (fields) to create the MQTT and  Influxdb JSON message's. 

Can you see both JSON messages in the Grott log and compare them (or upload them)?

fhteagle, 2023-08-09:
I think I have it sorted. Turns out it was a 3 way time mismatch that was creating influxdb values problem. 

I originally had set up the ShinePhone app as UTC -6, which was correct for Mountain Daylight Time. But this made the inverter unit stubbornly revert to being 1 hour ahead of MDT, despite what I set in the AP configuration web GUI. So when grottserver was set to TZ=America/Denver, timestamps were off by one hour into the future. So when I was looking at influxdb data explorer, it would not show me the future values at all, until the timestamp became later than now.

Set everything to be UTC all the way through and values are matching finally.

johanmeijer, 2023-08-09:
Yeah that is a little bit the risk of doing reverse engineering. The SPF layout is created with a document from Growatt: (https://watts247.com/manuals/gw/GrowattModBusProtocol.pdf) and comparison of the real output of the Inverter. 

The fields:

            "epv1tot"           
            "epvToday"          
            "pvenergytoday"   

Are pointing to the same field in the record:

That also applies to:

            "epv2tod"           
            "epvTotal"              
            "pvenergytotal"    

Probably you have specified includeall=True so you see all fields. Some should be hidden when you not specify this. 

If you do not want this we can create a "stripped" layout without these fields.

fhteagle, 2023-08-10:
@johanmeijer -

Yeah no worries. I plan on putting another string onto the second slave inverter tomorrow and seeing if/how it changes the data at all.

But I think I have what I need to make HomeAssistant start making intelligent decisions about when to activate my OpenEVSE that is on the solar trailer. If you want to work to try to improve the field mapping for the SPF / Wifi-F combo, I would be happy to help feed data your way. If not, could probably call it good enough as is.

Thanks.