johanmeijer/grott #390: How to add second inverter

chesterflaps, 2023-08-08:
I have one inverter sorted but cannot for the life of me work out how to add the second one.

All the documentation makes no sense to me and where it refers to things like 'if you want separate topics for each inverter you can add to the [MQTT] section of the grott.ini:

mqttinverterintopic="True"

I have no files anywhere called grott.ini so not sure what to do.

See attached image because it does make reference to PV2 but only the voltage and it clearly isn't working as it says zero.
[image]

johanmeijer, 2023-08-08:
If you have 2 the same inverters you do not have to do anything (except add mqttinverterintopic="True") if you want to sent the info from the inverters to 2 different topics.

If you have to different types you can specify the invtypemap= parameter:

      invtypemap = {'<serial>': 'default', '<serial>': 'sph', '<serial>': 'default', '<serial>': 'sph'}

chesterflaps, 2023-08-08:
They are both exactly the same. Where do I add mqttinverterintopic="True"

thanks

johanmeijer, 2023-08-08:
You can add that in the [MQTT] section of grott.ini

Be aware if you might have to specify your inverter type in the [Generic] part of grott.ini:

     invtype = nnn (nnn = sph, spf, tl3).

chesterflaps, 2023-08-08:
I'm using Home Assistant btw

chesterflaps, 2023-08-08:
Does anybody know where I can find grott.ini?

chesterflaps, 2023-08-09:
> You can add that in the [MQTT] section of grott.ini
> 
> Be aware if you might have to specify your inverter type in the [Generic] part of grott.ini:
> 
> ```
>  invtype = nnn (nnn = sph, spf, tl3). 
> ```

Ok so I have managed to find the grott.ini and have added that piece of code but I still do not get details of both inverters. Please see images below.

[image]
[image]

chesterflaps, 2023-08-09:
[image]