johanmeijer/grott #393: Crashes when influx not correct

vingerha, 2023-08-11:
When influx is setup but is no longer valid (e.g. I deleted the influx db) then the whole setup fails, nothing is pushed to MQTT nor forwarded to growat server. Suggest to 'catch' this error

johanmeijer, 2023-08-11:
Yes that has been a discussion before. If grott completely stops in case  of a failure the data is not sent to growatt server and stored in the datalogger to be sent later. 

Maybe I can create a parameter that the you can choose to ignore errors and continue processing.

vingerha, 2023-08-12:
Something like that would indeed be nice, a 'simple' try/except?

johanmeijer, 2023-08-13:
Question does Grott chrashes or only stops sending records?

vingerha, 2023-08-13:
Indeed, I forgot to specify. The container kept running but showed erross in the log related to the influxdb. Nothing went to MQTT or Growattserver