johanmeijer/grott #394: grott missing data... Record type T060136X?

splynch, 2023-08-12:
I have a new growatt system and tried installing grott. I get some data, but am missing things like production in kwh, or anything about the battery.

I have a lot of messages in my logs about message type T060136X:
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.7', 59230), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x02\x92\x00\x06\x02\x41\x01\x36\x1f\x35\x2b\x41\x25\x36\x3c\x74\x31\x3a\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x24\x24\x2c\x44\x37\x0b\x39\x5f\x44\x3b\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7c\x7f\x51\x6b\x79\x75\x18
		 \x6c\x0d\xd3\x72\x6a\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x4e\x2e\x6f\x6a\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x5c\x2a\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x5c\x1d\x6f\x77
		 \x62\x81\x74\x47\x61\xe1\x7e\xaf\x74\x74\x47\x72\x6f\x6b\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x75\xe6\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x0d\xd2\x08\x7e\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x8b\xb8\x8c\x63\x77\x61\x74\x7f\x47\x72\x6e
		 \xa1\x61\x74\x74\x4d\x72\x6f\x75\x02\x74\x74\x3d\x2a\x6f\x77\x1b\x2c\x76\x4d
		 \x8d\x90\x88\x9a\x74\x63\x47\x16\x6f\x77\x61\x17\x74\x47\x72\x78\x77\x78\x74
		 \x8e\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x77\x24
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060136X
	 - no matching record layout found, try generic
	 - Record layout used :  T060136X
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 02920006024101365847443644424833435500000000000000000000000000000000000000004
		 b534d30434c4b30335a000000000000000000000000000000000000000017080b161916027918
		 79940005000000000000000000000000000000000000095c001d0000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000001b580000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000001b6f000003f50000138e09ce00000000001c000000000000000001
		 92000000000000000000000000000000000000000079957a11000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000fffffe0c0000000b000001d60000000a000
		 0026300007a5800007a58020afffffffb001700640000006300000017001900fa000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000054b
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.7', 5279), raddr=('192.168.105.204', 62454)>
	 - Data less then minimum record length, data not processed
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.7', 59230), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.30.33.7', 5279), raddr=('192.168.105.204', 62454)>
	 - Data less then minimum record length, data not processed

And when i do get a successful parse:

         - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T060104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00a90006033f01045847443644424833435500000000000000000000000000000000000000004
		 b534d30434c4b30335a000000000000000000000000000000000000000017080b16241c030000
		 007c00050000196408ec001c0000196d00000000000000000001420009a20000000300fa00000
		 000000000000000000000000000000000000000000000000000000000000000000000001a1b13
		 8d09c8001a00001a3f0000000000000000000000000000000000000000000000000025000005f
		 2003eeae90000002e000006990000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000000000000000000000000006b1019f01720
		 1990000020e10b810a60000000000000000000000000000000000000000000000000000000300
		 00006800000000000000010000000000000000000003e80464000500000000000000000000000
		 0000000000000032000000000020d001900000000000000000000000000000000000000000000
		 0000000000000000000000001c20000000000000000000001c200000001900050c42177000000
		 0220000081b00000001000000160000000a000002630000000c000001d70000004e00000f4200
		 0000000000000009a500000000000000000000000000000000000000000000000003e80000006
		 30000000000191482ff92001917700018006300030002000a0064163000000000000000030000
		 000000005447544700000cd20cd000010e10132403df006618000000000000006300000000000
		 00000000000046504e100030000006800000000040e0000000000000000000000000000002a00
		 0006b10000002a000006ad00001c9300001c930000002f0001000002060000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000010cd20cd0000100100002000100fa00fa0001000100000000000000000000000000
		 03001000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000008925
	 - Possible Inverter serial KSM0CLK03Z
	 - Inverter serial not recognised - using inverter type default
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T06NNNNX
	 - Grott data record date/time processing started
	 - date-time:  2023-08-11T22:36:28
	 - Grott values retrieved:
		 -  datalogserial        :  XGD6DBH3CU
		 -  pvserial             :  KSM0CLK03Z
		 -  pvstatus             :  5
		 -  pvpowerin            :  650.0
		 -  pv1voltage           :  228.4
		 -  pv1current           :  2.8
		 -  pv1watt              :  650.9
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  0.0
		 -  pvfrequentie         :  0.0
		 -  pvgridvoltage        :  0.0
		 -  pvgridcurrent        :  0.0
		 -  pvgridpower          :  0.0
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  43798209.3
		 -  totworktime          :  0.0
		 -  pvenergytoday        :  0.0
		 -  pvenergytotal        :  0.0
		 -  epvtotal             :  3.7
		 -  epv1today            :  152.2
		 -  epv1total            :  412336.9
		 -  epv2today            :  4.6
		 -  epv2total            :  168.9
		 -  pvtemperature        :  41.5
		 -  pvipmtemperature     :  52.6
           - MQTT jsonmsg:

The inverter is a SPH3000TL BL-UP hybrid inverter. 

[image]

Has anyone seen these messages before? I am happy to help try and parse them, but not entirely sure how to start.

johanmeijer, 2023-08-12:
I am not sure what the T060136X records are. They are new for me. 

Let's first look at the known data records (T060104X) while that is the usual data record. Can you please add invtype=sph to the grott.ini (or do you use docker? Then you have to specify ginvtype="sph"). That will give more useful information I think. 

For the "36"record we have to detect why they are there. How does your configuration look like? I might be a battery data record but I am not sure.

splynch, 2023-08-15:
So, comparing what i can see in the growatt server dashboard to the MQTT messages...
The figures I think I need to get are:
Total solar generation: epv1total [X/10]kwh
Total amount battery charged: eharge1_tot [x/10]kwh
total amount discharged from battery: edischarge1_tot [x/10]kwh

And good to know, state of charge on battery: SOC [x]%

I do not know why EPV1total and EPVtotal are different, but EPV1Total matches what i see in the shine app.

Those show up in the MQTT messages, but dont appear to get parsed for home assistant.  I know very little about MQTT... it usually just works. Do I need to right some rules (topics?) so that home assistant knows how to parse those updates? or should Grott be sending them?

andreaswillems, 2023-10-04:
> I am not sure what the T060136X records are. They are new for me.
>
> Let's first look at the known data records (T060104X) while that is the usual data record. Can you please add invtype=sph to the [...]

Hi @johanmeijer , thank you very much for the work on this. Is there something one can do to help enable parsing of T060136X records?

johanmeijer, 2023-10-04:
The question do we miss information that justifies to put effort in debugging the "36" type data records.

For now I assume these are announcement records (but from which device?). Do you have next to the inverter also a samrt meter (e.g. sdm630) or other device connected?
What type of datalogger do you have (shinewifi/lan, shinelan box, ....).

andreaswillems, 2023-10-04:
I have a Smart Meter installed (based on SDM 630). The datalogger is a Shine WiFi-X stick.

Are there any resources to look at in order to decipher the (unknown) records?

gavintweedie, 2023-10-12:
I get something similar, my inverter is SPH5000 TL BL UP. Note the SPH3k to SPH6k I believer are same hardware just software limited for output. I also have a SDM630 attached.

It appears that some values populate fine, for example Phase1 voltage (inverter is single phase, smart meter is 3 phase) but the other two phase voltages do not in home assistant, they do show in the grott log output however. I'm going to dig this weekend into it and see if I can work out if it's Grott not sending it or the homeassistant plugin not collecting/using it.