johanmeijer/grott #395: Blank serial number (rare)

amund7, 2023-08-15:
Hello,

I have been running GROTT + Home Assistant Grott for around 3 months now, and am very happy with it. However, there is one rare glitch happening maybe once every few weeks - MQTT reports 'u0000\u0000\u0000' etc as the serial number. This means the Home Assistant plugin registers a new device with a blank name, and chaos ensues. It always works by just restarting Grott.

I am guessing the serial number is read on startup, and that it occationally fails, or the shine-wifi module actually reports only zeroes. But maybe it could retry if all nulls?

This is how it looked this morning from MQTT Explorer, when I realized my production meter was not moving.

[image]

johanmeijer, 2023-08-15:
Interesting. Are you able to look at the grott log at that time and see what the original data is that is being sent? 

If it is a generic issue (more people have this problem) we can of course set a filter for these types of records.

amund7, 2023-08-15:
When exactly will the serial be sent? At Grott startup, at inverter boot, or constantly? Then I can look for the log details.

johanmeijer, 2023-08-15:
With every record. I (grott) processes it at the moment the data record is beeing sent (when you see the mqtt message in the grott log). Just before this message the original data is being displayed (with /x format). That is the data that will help.

amund7, 2023-08-15:
What is /x format, a command line switch?

amund7, 2023-08-17:
Happened again today, but I don't see much interesting in the log? 

`      - Growatt packet received:
                  <socket.socket fd=980, family=2, type=1, proto=0, laddr=('192.168.0.125', 5279), raddr=('192.168.0.109', 5127)>

         - Growatt packet received:
                  <socket.socket fd=1868, family=2, type=1, proto=0, laddr=('192.168.0.125', 61423), raddr=('47.91.67.66', 5279)>

         - Growatt packet received:
                  <socket.socket fd=980, family=2, type=1, proto=0, laddr=('192.168.0.125', 5279), raddr=('192.168.0.109', 5127)>

         - Growatt packet received:
                  <socket.socket fd=1868, family=2, type=1, proto=0, laddr=('192.168.0.125', 61423), raddr=('47.91.67.66', 5279)>
         - Growatt data decrypted V2
         - Grott MQTT topic used : energy/growatt

         - Growatt packet received:
                  <socket.socket fd=980, family=2, type=1, proto=0, laddr=('192.168.0.125', 5279), raddr=('192.168.0.109', 5127)>

         - Growatt packet received:
                  <socket.socket fd=1868, family=2, type=1, proto=0, laddr=('192.168.0.125', 61423), raddr=('47.91.67.66', 5279)>
         - Growatt data decrypted V2
         - Grott MQTT topic used : energy/growatt

         - Growatt packet received:
                  <socket.socket fd=980, family=2, type=1, proto=0, laddr=('192.168.0.125', 5279), raddr=('192.168.0.109', 5127)>

         - Growatt packet received:
                  <socket.socket fd=1868, family=2, type=1, proto=0, laddr=('192.168.0.125', 61423), raddr=('47.91.67.66', 5279)>
         - Growatt data decrypted V2
         - Grott MQTT topic used : energy/growatt`

johanmeijer, 2023-08-25:
Do you run grott with the -v option?

So if you use the command line:

python3 grott.py -v