johanmeijer/grott #396: Growatt portal doesn't show settings for the inverter any more.

burnsjake, 2023-08-19:
I'm using an SPF6000T DVM off-grid inverter. 

When I put grott in the middle, the growatt portal now shows only inverter settings with no battery settings. I can't adjust anything on the inverter now.

How do I adjust grott to pass through data without growatt thinking my device is something different?

johanmeijer, 2023-08-19:
Grott is not changing or filtering anything. It would be very strange that it affects the behaviour of the growatt website. 

There are more spf users and I have not heard about  this behaviour before. 

If it is really the case can you sent me the grott log maybe we see something strange.

rany2, 2023-08-19:
grottproxy only intercepts the data but does not modify it. Does this occur with grottsniffer as well? Maybe they are checking which server the datalogger is using and disabling it if not server.growatt.com; but I doubt it.

johanmeijer, 2023-08-20:
Do you know which server the data is sent. How is this defined in your datalogger (server.growatt.com, ip address).

Maybe in your regio you have to use another server address (where are you located).

burnsjake, 2023-08-20:
I run 'nohup python grott.py -v' - the contents of nohup.out are attached as grottlog.txt.
I logged into the growatt portal and changed the data logger setting "Set Ip(Caution Ip Transfer With)" to '10.0.0.29'. 
Prior to that, I used server-us.growatt.com in the "Set domain" field.

I also attached my grott.ini file.

[grott.ini.txt](https://github.com/johanmeijer/grott/files/12389337/grott.ini.txt)

[grottlog.txt](https://github.com/johanmeijer/grott/files/12389344/grottlog.txt)

burnsjake, 2023-08-21:
This seems sort of fixed. I had to talk to growatt support, they did something on the backend in the portal and restored the settings.

closed 2023-08-21 by burnsjake

johanmeijer, 2023-08-21:
Ok strange. I am pretty sure Grott does not change any records. So it must be an error at Growatt site. 

But if works now, you are probably happy ;)