johanmeijer/grott #399: Set IP not present on Growatt Website

patrick131184, 2023-08-27:
Hi,
i just tried to set the IP Adresse as described here (https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server).

My Settings Page from my Datalogger is showing up like this and i have no Option to change the IP Adresse.
What am i doing wrong?

[image]
[image]
[image]
[image]

bvweerd, 2023-08-28:
That's because you can find this option in the settings of the datalogger, not in the settings of the inverter. Click "all device" in the Growatt interface:
[image]

then click datalogger settings:
[image]

patrick131184, 2023-08-28:
Ah, got it. Thank you!