johanmeijer/grott #401: UnboundLocalError: local variable 'crc' referenced before assignment

HoekieN, 2023-08-29:
i am using this to start my docker container:
docker run --name grott    --restart unless-stopped    -p 5279:5279    -v /home/****/grott/grott.ini:/app/grott.ini  -v /home/****/grott/grott_ha_spf.py:/app/grott_ha_spf.py  -d ledidobe/grott:2.8.2

But it always crashes with the following error:

Grott settings:
_Generic:
	version:             	 2.8.2
	verbose:             	 True
	trace:               	 False
	config file:         	 grott.ini
	minrecl:             	 100
	decrypt:             	 True
	compat:              	 False
	invtype:             	 default
	invtypemap:          	 {}
	include_all:         	 False
	blockcmd:            	 False
	noipf:               	 False
	time:                	 auto
	sendbuf:             	 True
	timezone:            	 local
	valueoffset:         	 6
	offset:              	 6
	inverterid:          	 automatic
	mode:                	 proxy
	grottip              	 default
	grottport            	 5279
_MQTT:
	nomqtt               	 False
	mqttip:              	 192.168.150.200
	mqttport:            	 1883
	mqtttopic:           	 energy/growatt
	mqttmtopic:          	 False
	mqttmtopicname:      	 energy/meter
	mqttinverterintopic: 	 True
	mqtttretain:         	 False
	mqtttauth:           	 True
	mqttuser:            	 mqtt-user
	mqttpsw:             	 **secret**
_Growatt server:
	growattip:           	 47.91.67.66
	growattport:         	 5279
_PVOutput:
	pvoutput:            	 False
	pvdisv1:             	 False
	pvtemp:              	 False
	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
	pvapikey:            	 yourapikey
	pvinverters:         	 2
	pvsystemid:          	 {1: '69900', 2: '562264'}
	pvinvertid:          	 {1: 'inverter1', 2: 'inverter2'}
_Influxdb:
	influx:             	 False
	influx2:            	 False
	database:           	 grottdb
	ip:                 	 localhost
	port:               	 8086
	user:               	 grott
	password:           	 **secret**
	organization:       	 grottorg
	bucket:             	 grottdb
	token:              	 **secret**
_Extension:
	extension:          	 True
	extname:            	 grott_ha_spf
	extvar:             	 {'ha_mqtt_host': '192.168.150.200', 'ha_mqtt_port': '1883', 'ha_mqtt_user': 'mqtt-user', 'ha_mqtt_password': '****'}

Grott proxy mode started
Hostname : 4c424cc58f7f
IP :  172.17.0.5 , port :  5279 
	 - ('192.168.150.148', 5465) has connected
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.5', 43278), raddr=('47.91.67.66', 5279)>
	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.5', 43278), raddr=('47.91.67.66', 5279)>
Traceback (most recent call last):
  File "/app/grott.py", line 35, in <module>
    proxy.main(conf)
  File "/app/grottproxy.py", line 144, in main
    self.on_recv(conf)
  File "/app/grottproxy.py", line 189, in on_recv
    validatecc = validate_record(vdata)
  File "/app/grottproxy.py", line 67, in validate_record
    if protocol != "02" and crc != crc_calc:     
UnboundLocalError: local variable 'crc' referenced before assignment

i have tried 2.7.8 also enable and disable the extention.
Anyone an idea how i can fix this.
i have 2 inverters pointing to grott

HoekieN, 2023-09-14:
Anyone??

johanmeijer, 2023-09-21:
Grott (docker) never worked for you?

HoekieN, 2023-09-22:
it works fine with:
version: '3.3'
services:
    grott:
        image: ledidobe/grott
        container_name: grott
        restart: unless-stopped
        ports:
            - "5279:5279"
        volumes:

But when i want to use the latest version and only change this line to 2.8.2:
image: ledidobe/grott:2.8.2

then it dont work anymore, i got a second converter now and want to use the new version for the change that HA made for the  new names.

johanmeijer, 2023-09-24:
Oke. What types of inverter do you have? Can you sent me the log at startup with the standard grott image (till something has been sent to MQTT)?

Then I will try to replay the communication and see what causes the error. 

Do you run with a intel x86 based server or a raspberry pi?

HoekieN, 2023-09-27:
When i tried to get the logs i played a bit with the options again and removed:
            - gpvinverterid1
            - gpvsystemid1=69900
            - gpvinverterid2
            - gpvsystemid1=562264
After that the docker container started with version 2.8.2
Thanks so much for your time.

HoekieN, 2023-09-30:
Sorry nvm, it worked for a few hours and now it keeps crasing again as soon as i start it.

HoekieN, 2023-09-30:
[grott.log](https://github.com/johanmeijer/grott/files/12775608/grott.log)
i get no message in MQTT is it crash before that

johanmeijer, 2023-10-01:
@HoekieN I can not recreated the problem :(

What type of inverter(s) do you have and how are you running the grott container (RPI, X86, NAS etc)?

rany2, 2023-10-01:
@johanmeijer Could it be non-inverter data (not protocol 5/6) but instead data from smart meter? I don't have a smart meter to know how the data layout works but with validate_record the assumption is that if protocol != 2 then it is has to protocol 5 or 6 which is not guaranteed.

HoekieN, 2023-10-01:
> @HoekieN I can not recreated the problem :(
> 
> What type of inverter(s) do you have and how are you running the grott container (RPI, X86, NAS etc)?

Ah sorry i forgot that, i have an 3600MTL-S that is not connected yet but want to and i have the Growatt 1000-S connected through the shine lan box, OS of docker:  Ubuntu 22.04.3 LTS (GNU/Linux 5.15.0-84-generic x86_64)

johanmeijer, 2023-10-01:
Oke I have a 1000-s and 1500-s one connected with a shinelink and the other with a shinelink (and rf stick).

I have not seen this problem before. You do not have a SDM smart meter as suggested by @rany2 in use?

HoekieN, 2023-10-01:
I have a SDM smart meter yes, but that is not sending any data to my network, the P1 is connected to usb to an other server (PI)

HoekieN, 2023-12-11:
Can be closed

johanmeijer, 2023-12-17:
@HoekieN Did you fixed it?

HoekieN, 2023-12-20:
Hi @johanmeijer sadly not, in the end i switched to https://github.com/egguy/addon-grott-beta and run it as an add-on on my HA instead of my own docker machine and that works pretty great.

closed 2023-12-20 by HoekieN