johanmeijer/grott #405: Which register is Load First Discharge Stopped Soc ?

osos, 2023-09-09:
Hi,

I can't seem to find the register to set the Load First Discharge Stopped SOC:

[image]

Anyone know?

cyberdiamond, 2023-09-09:
Yes that is one that is not in any official documentation, but I have found it at register 1109

cyberdiamond, 2023-09-09:
P.S. I have a SPH so it is possible that it could be different for a different model?

osos, 2023-09-09:
If I read the register 1109 I do get "10", however if try to write e.g. "30" to it, it remains "10". I have a SPH-3600

cyberdiamond, 2023-09-09:
Yes I noticed that as well, if I change it via server.growatt.com and read it from grottserver it is changed.

It may be one of the registers where you need to change more than 1 register at once to get it to change.

cyberdiamond, 2023-09-09:
Well it was found from checking the communication from the server.growatt.com website that it was changing a single register 608 when setting the Load First Discharge Stopped SOC %

The only mention I can find of 608 is One key to set battery mode function with settings of:
0: self-use
1: battery priority
2: Grid priority
255：Disable
And a note: Currently used for US models

I am in Australia so I did not think anything of it when I could not read from it originally.

I tried again to read from register 608 but could not.
I did test writing to register 608 and it did work, but then tried reading from it again which still did not work?

To make it even more interesting, reading from register 1109 shows the new value I wrote to register 608?

So yes very strange! 
But it works?

There must be some link between them?

Maybe Growatt have not finished, which maybe why it's not documented at all yet?

cyberdiamond, 2023-09-09:
Also note we may have different models (mine is SPH5000 TL BL) and firmware as I have updated mine.