johanmeijer/grott #414: Missing data Growatt Shine usb in Homeassistant

vunnies, 2023-09-18:
Hi, 

I have a question regarding my 3 dataloggers Wifi shine usb.

I have Grott installed and my sticks configured. It seems all 3 are working but I miss some data like the total power (as shown in the pictures) Strange thing is the logger garage 1 and 2 have the same inverter. A quick scan with mqtt explorer shows that the data is fetched correctly but does not show up as an entity.

How could I solve this

[image]
[image]
[image]

vunnies, 2023-09-25:
No one?

DanielAtkinson, 2023-09-27:
Have you enabled data buffering?

johanmeijer, 2023-09-27:
Do you use the grott_ha.py extension ? Maybe @egguy can help here?

thoompje, 2023-10-03:
Looks a bit the same as for my issue which i raised.

https://github.com/johanmeijer/grott/issues/429

@vunnies already found a solution to get all the sensors in MQTT?

egguy, 2023-10-03:
I will try to take a look. If you are using the pure MQTT export, I think I've found the bug. My weeks are a bit busy at the moment. I'm trying to update the extension with a better release process.