johanmeijer/grott #421: HA - grott edge add-on: Grott InfluxDB server IP address env invalid

GeraldPape, 2023-09-27:
So rewceiving this

```
Pulled 'gifdbname=grott' from the environment
**Grott InfluxDB server IP address env invalid**
Grott Influxdb initialisation error

```

configured this: (also tried gifip: localhost ) but this is the same config as for HA itself - besides the db name and user.
```
ginflux: true
ginflux2: true
gifdbname: grott
gifip: a0d7b954-influxdb
gifport: 8086
gifuser: username
gigpassword: password
```

johanmeijer, 2023-09-27:
gifip: a0d7b954-influxdb looks a little bit weird to me for an IP address. I really expect a 192.168.yy.xxx (or 172....) as ip address of the influxdb host server. 

Localhost will not work I think while localhost is the IP address of the grott docker container.

GeraldPape, 2023-10-13:
> gifip: a0d7b954-influxdb looks a little bit weird to me for an IP address. I really expect a 192.168.yy.xxx (or 172....) as ip address.

ssh into homeassistant and found the ip address. seems the grott docker was unable to dns the name. the db was created but I do not get any values into it. will open another issue when I have more time to document the problem.

closed 2023-10-13 by GeraldPape