johanmeijer/grott #427: No Meter Data collected

lepo80, 2023-10-01:
I started to use the grott into Docker installation, it works perfectly in proxy mode and write all inverter information into my MQTT server properly. but the meter data is not avaiable into mqtt; it seems that the meter data is not collected properly (e.g. power export to grid data is missing). The meter installed into my flat is ChiNT DDSU666. you should find below the Grott log.
do you how how it is possible to collect and provide this data to MQTT, please?

thx

grott log:
		 - Growatt original Data:
		 \x00\xb5\x00\x06\x03\x3f\x51\x04\x09\x33\x2c\x36\x22\x38\x33\x75\x35\x3f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3c\x3c\x2d\x44\x37\x0f\x43\x5f\x32\x36\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x74\x6a
		 \xcc\x78\x73\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x7c\xf0\x68\x29
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x2f\x6f\x77\x61\x74\x74\x47\x61\xe3\x77\x61\x74\x74\x47\x72
		 \x7c\xfb\x61\xe0\x6e\xc5\x72\x6f\x77\xd4\x74\x74\x5c\xae\x6f\x77\x7c\x97\x74
		 \x47\x72\x04\x77\x61\x64\x09\x47\x72\x6f\x20\x61\x74\x79\x21\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x77\x74\x47\x71\x25\x77\x61\x74\xd4\x47\x72\x64
		 \xe8\x61\x74\x74\x62\x72\x6f\x62\xef\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\xad\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x75
		 \x35\x46\x33\x6f\x77\x61\x74\x75\x50\x73\x8d\x77\x61\x3a\x54\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x85\x72
		 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\xb6\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x6f\x8a
		 \x77\x61\x74\x56\x47\x72\x7d\x32\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x61\x1c\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x70\x74\x46\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x71\x61\x41\x74\x71\x73
		 \x1e\x79\x71\x74\x75\x47\x7b\x6f\x77\x61\x74\x74\x47\x7e\xdd\x7a\x4f\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x64
		 \x6a\x2c\x2c\x1f\x2a\x6f\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a
		 \x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x1a\xa6
	 - Grott automatic protocol detection
	 - Grott data record length 839
	 - layout   :  T065104X
	 - no matching record layout found, try generic
	 - Record layout used :  T06NNNNX
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00b50006033f51044e414341434c4732475000000000000000000000000000000000000000005
		 34b4c304348313045570000000000000000000000000000000000000000000000000000030bb8
		 0c340000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000000000000000000001387095d000000000000000000000000000000000000000000
		 00095d000000000000138c000000000000138c00941a82000000b500001bdc00001de30000006
		 b0000107d0000005700000d660000000000000000000000030000034a000000a000000b9f0000
		 00250000158e0000000000000000000000c200000000fffa00000000000000000000014101410
		 0000000011701e200004e20000000000000000000000000000000000000000000c20000000000
		 00003c0000000000000000000000000000000000c20c350cb1000000000000000000000000000
		 00000000000000000000000001de5000000220000124500000000000000000000000000000000
[... 71 lines cut ...]
	 - Grott automatic protocol detection
	 - Grott data record length 238
	 - layout   :  T065120
	 - no matching record layout found, try generic
	 - Record layout used :  T065120
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00a3000600e651204e414341434c4732475000000000000000000000000000000000000000005
		 34b4c30434831304557000000000000000000000000000000000000000000000000000000a000
		 00008c0000096f0000000000000000000000170000000000000000000011d0000000000000000
		 0000012220000000000000000fffffd120000000000000000000003db00000000000000000000
		 11d000001222fffffd12000003db000001f400000000000000000000000000000d3500000c550
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 0000000000c439
	 - Grott data ack record or data record not defined no processing done
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.5', 5279), raddr=('192.168.1.16', 1030)>
	 - Data less then minimum record length, data not processed
	 - Growatt packet received:
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.5', 58604), raddr=('47.91.67.66', 5279)>
	 - Data less then minimum record length, data not processed
	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.5', 5279), raddr=('192.168.1.16', 1030)>
	 - Data less then minimum record length, data not processed

lepo80, 2023-10-04:
[SOLVED] - you should find below what I implemented to consider the new Growatt inverter and the specific meter models. unfortunatly I don't have any battery installed in my solar system, then I wasn't able to verify their data.

--------

# this is the configuration to be add into grottconf.py to read Growatt 3600TL-HX (T065104XTL3) and ChiNT DDSU666 meter (T065120). I used this configuration into this Docker version: ledidobe/grott:2.8.2
#all relevant information are collected now, but, for my needs, i put some comments to avoid to collect not relevant data, fell free to remove these comments basing on your preference.
#due to I don't have any battery installed, I wasn't able to verify all battery values.

        self.recorddict15 = {"T065104XTL3": {
            "decrypt"           : {"value" :"True"},
            "pvserial"          : {"value" :76, "length" : 10, "type" : "text", "divide" : 10},
            "date"              : {"value" :136, "divide" : 10}, 
            "pvstatus"          : {"value" :158, "length" : 2, "type" : "num"},  
            #"pvpowerin"         : {"value" :162, "length" : 4, "type" : "num", "divide" : 10},        
            #"pv1voltage"        : {"value" :170, "length" : 2, "type" : "num", "divide" : 10},        
            #"pv1current"        : {"value" :174, "length" : 2, "type" : "num", "divide" : 10},                        
            #"pv1watt"           : {"value" :178, "length" : 4, "type" : "num", "divide" : 10},                      
            #"pv2voltage"        : {"value" :186, "length" : 2, "type" : "num", "divide" : 10},                
            #"pv2current"        : {"value" :190, "length" : 2, "type" : "num", "divide" : 10},                
            #"pv2watt"           : {"value" :194, "length" : 4, "type" : "num", "divide" : 10}, 
            #"pvpowerout"        : {"value" :250, "length" : 4, "type" : "num", "divide" : 10},                
            #"pvfrequentie"      : {"value" :258, "length" : 2, "type" : "num", "divide" : 100},                
            #"pvgridvoltage"     : {"value" :262, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridcurrent"     : {"value" :266, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridpower"       : {"value" :270, "length" : 4, "type" : "num", "divide" : 10},                
            #"pvgridvoltage2"    : {"value" :278, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridcurrent2"    : {"value" :282, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridpower2"      : {"value" :286, "length" : 4, "type" : "num", "divide" : 10},                
            #"pvgridvoltage3"    : {"value" :294, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridcurrent3"    : {"value" :298, "length" : 2, "type" : "num", "divide" : 10},                
            #"pvgridpower3"      : {"value" :302, "length" : 4, "type" : "num", "divide" : 10},  
            #"totworktime"       : {"value" :346, "length" : 4, "type" : "num", "divide" : 7200},
            "pvenergytoday"     : {"value" :354, "length" : 4, "type" : "num", "divide" : 10},                  
            "pvenergytotal"     : {"value" :362, "length" : 4, "type" : "num", "divide" : 10},
            "epvtotal"          : {"value" :370, "length" : 4, "type" : "num", "divide" : 10},
            "epv1today"         : {"value" :378, "length" : 4, "type" : "num", "divide" : 10},                
            #"epv1total"         : {"value" :386, "length" : 4, "type" : "num", "divide" : 10},                
            "epv2today"         : {"value" :394, "length" : 4, "type" : "num", "divide" : 10},                
            #"epv2total"         : {"value" :402, "length" : 4, "type" : "num", "divide" : 10},                           
            "pvtemperature"     : {"value" :530, "length" : 2, "type" : "num", "divide" : 10},                 
            "etouser_tod"       : {"value" :426, "length" : 4, "type" : "num", "divide" : 10},
            #"etouser_tot"       : {"value" :434, "length" : 4, "type" : "num", "divide" : 10},
            "etogrid_tod"       : {"value" :442, "length" : 4, "type" : "num", "divide" : 10},
            "etogrid_tot"       : {"value" :450, "length" : 4, "type" : "num", "divide" : 10},
            #"elocalload_tod"    : {"value" :458, "length" : 4, "type" : "num", "divide" : 10},
            #"elocalload_tot"    : {"value" :466, "length" : 4, "type" : "num", "divide" : 10},
            #"pbusvolt"          : {"value" :550, "length" : 2, "type" : "num", "divide" : 1},
            #"nbusvolt"          : {"value" :554, "length" : 2, "type" : "num", "divide" : 1},
            #"ISO"               : {"value" :490, "length" : 4, "type" : "num", "divide" : 10},
            #"Temperature2"     : {"value" :526, "length" : 4, "type" : "num", "divide" : 10},
            #"Temperature5"     : {"value" :542, "length" : 4, "type" : "num", "divide" : 10},
            "pvipmtemperature"  : {"value" :534, "length" : 2, "type" : "num", "divide" : 10}
        } }
 
        self.recorddict16 = {"T065120": {
            "decrypt"           : {"value" :"True"},
            "datalogserial"     : {"value" :16, "length" : 10, "type" : "text","incl" : "yes"},
            "pvserial"          : {"value" :76, "length" : 10, "type" : "text"},
[... cut ...]
            "eacharge_total"    : {"value" :398, "length" : 4, "type" : "num", "divide" : 10},  
            "edischrtoday"      : {"value" :406, "length" : 4, "type" : "num", "divide" : 10},  
            "edischrtotal"      : {"value" :414, "length" : 4, "type" : "num", "divide" : 10}
        }}
        
        
# also, it is needed to add all lines below into the right position        

        self.recorddict.update(self.recorddict15)  #T065104XTL3
        self.recorddict.update(self.recorddict16)  #T065120

cantupaz, 2023-10-07:
@lepo80 Thanks for entering your solution here. My power meter sends the T065120 format of messages and your solution helped me to get the readings I needed.

@johanmeijer Would you consider integrating this into grott? I'd be happy to help test/validate if needed.