johanmeijer/grott #428: date-time - issue with time being sent to MQTT

J4CE2, 2023-10-02:
Hi, 

I've noticed the time seems to be getting out of sync or delayed? 

The data otherwise seems correct coming across to MQTT.

Some examples below. 
Afer 1pm, it actually permanently stays worse and reports a 7 hr time difference.

Timezones are set correctly to Perth/West Australia on both Grott host, MQTT and HA hosts.

Bit wacky behaviour? 

ShineWifi-X, 3.1.0.5 firmware
Inverter, Growatt MOD 5000TL3-X

**Date/Times shown in HA**
[image]

[image]

**Log File on Grott Server**
[image]

`Oct 02 09:07:00 rpi-monitor grott[29144]:          - Grott data record date/time processing started`
`Oct 02 09:07:00 rpi-monitor grott[29144]:          - date-time:  2023-10-02T08:34:20`
`Oct 02 09:07:00 rpi-monitor grott[29144]:          - Grott values retrieved:`

Oct 02 09:07:00 rpi-monitor grott[29144]:                          {"device": "SFH3BL602W", "time": "2023-10-02T08:34:20", "buffered": "no",
Oct 02 09:07:00 rpi-monitor grott[29144]:                          "values": {"datalogserial": "XGD5BL21HQ", "pvserial": "SFH3BL602W",
Oct 02 09:07:00 rpi-monitor grott[29144]:                          "pvstatus": 1, "pvpowerin": 27930, "pv1voltage": 2998, "pv1current": 47,
Oct 02 09:07:00 rpi-monitor grott[29144]:                          "pv1watt": 14090, "pv2voltage": 1812, "pv2current": 70, "pv2watt": 12684,
Oct 02 09:07:00 rpi-monitor grott[29144]:                          "pvpowerout": 27512, "pvfrequentie": 4995, "pvgridvoltage": 2378,
Oct 02 09:07:00 rpi-monitor grott[29144]:                          "pvgridcurrent": 38, "pvgridpower": 9036, "pvgridvoltage2": 2442,

J4CE2, 2023-10-03:
From looking at the code, I suspect an offset is being applied. Not really sure how it's calculating that. However if I change time in the grott.ini file from auto to server, I suspect this will fix the problem.

Testing later today.

J4CE2, 2023-10-04:
Confirmed, using time=server resolves the issue.

# Time = auto/server parameter enable/disable date/time retrieval from data record (server), default is
# auto: grott decides which time is used (data record if valid otherwise Server)
# If time = server Grott server time is alwas used
time = server

Not sure what "auto" is doing to get mixed up.

johanmeijer, 2023-10-04:
Grott normal time processing (auto) is using the date/time in the datarecord sent by the inverter/datalogger. If there is no or invalid (not recognisable) date/time found it will use the server time.

So if you get out of sync with the time it is caused by the datalogger. Normally the time of the datalogger is set once a day when the inverter starts. In initialisation phase the growatt server sent a date/time update. We noticed that some dataloggers (I am not sure which one you are using) have problems with keeping up the time (firmware error/hw problem ?). 

The alternative is to update the time more regularly , via growatt website, growatt api or stop/start datalogger. Or using grottserver and send time updates via its api.

An other alternative is, and you already found that one, to set time=server in grott. But this will disable the functionality of sending historical(buffered) records,while it make no sense to send historical record with an actual time.

J4CE2, 2023-10-04:
Thanks for the information.
FYI I am using a ShineWifi-X, 3.1.0.5 firmware data logger.

J4CE2, 2023-10-04:
Resolved by config setting

closed 2023-10-04 by J4CE2