johanmeijer/grott #429: Sensors getting suddenly unavailable

thoompje, 2023-10-03:
Can anyone help me? Suddenly a lot of sensors getting unavailable

[image]

`	 - Grott data record date/time processing started
	 - date-time:  2023-10-03T12:26:39
	 - Grott values retrieved:
		 -  datalogserial        :  XXXXXXXX
		 -  pvserial             :  XXXXXXXX
		 -  pvstatus             :  1
		 -  pvpowerin            :  347.2
		 -  pv1voltage           :  352.6
		 -  pv1current           :  0.9
		 -  pv1watt              :  347.2
		 -  pv2voltage           :  0.0
		 -  pv2current           :  0.0
		 -  pv2watt              :  0.0
		 -  pvpowerout           :  340.3
		 -  pvfrequentie         :  50.0
		 -  pvgridvoltage        :  235.9
		 -  pvgridcurrent        :  1.3
		 -  pvgridpower          :  341.3
		 -  pvgridvoltage2       :  0.0
		 -  pvgridcurrent2       :  0.0
		 -  pvgridpower2         :  0.0
		 -  pvgridvoltage3       :  0.0
		 -  pvgridcurrent3       :  0.0
		 -  pvgridpower3         :  0.0
		 -  totworktime          :  7532.1
		 -  pvenergytoday        :  1.2
		 -  pvenergytotal        :  6484.1
		 -  epvtotal             :  6561.8
		 -  epv1today            :  1.2
		 -  epv1total            :  6561.8
		 -  epv2today            :  0.0
		 -  epv2total            :  0.0
		 -  pvtemperature        :  38.1
		 -  pvipmtemperature     :  46.2
	 - MQTT jsonmsg: 
			 {"device": "XXXXXXXX", "time": "2023-10-03T12:26:39", "buffered": "no",
			 "values": {"datalogserial": "XXXXXXXX", "pvserial": "XXXXXXXX",
			 "pvstatus": 1, "pvpowerin": 3472, "pv1voltage": 3526, "pv1current": 9,
			 "pv1watt": 3472, "pv2voltage": 0, "pv2current": 0, "pv2watt": 0,
			 "pvpowerout": 3403, "pvfrequentie": 5001, "pvgridvoltage": 2359,
			 "pvgridcurrent": 13, "pvgridpower": 3413, "pvgridvoltage2": 0,
			 "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
			 "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 54231432,
			 "pvenergytoday": 12, "pvenergytotal": 64841, "epvtotal": 65618, "epv1today":
			 12, "epv1total": 65618, "epv2today": 0, "epv2total": 0, "pvtemperature":
			 381, "pvipmtemperature": 462}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT send failed: [Errno 99] Address not available
	 - Grott Send data to PVOutput disabled 
	 - Grott Send data to Influx disabled 
	 - Grott extension processing started :  grott_ha
	 - Grott extension processing ended :  0`

thoompje, 2023-10-03:
This message: 'MQTT send failed: [Errno 99] Address not available' is weird because data is getting updated the only thing is that i miss some sensors...

thoompje, 2023-10-03:
> I'm not expert but I don't think it is publishing things to MQTT based on what that error message is saying.

But its weird that the other sensors are updating and being published in MQTT, maybe @johanmeijer knows more about this how it works?

thoompje, 2023-10-03:
It looks like that he now takes the first 9 values and not doing anything with the rest.

thoompje, 2023-10-03:
Alright All, I think I found the problem/challenge. I was deep diving into the problem with MQTT explorer as well and  find out that the connection were cut off after a bit of data was send.

So i stopped Zigbee2mqtt Edge and zigbee2mqttassistant-dev, after that i restarted the Grott - HA extension addon and suddenly all the sensors came in. So it looks like Mosquitto broker is to busy on my PI4 (with SSD). Maybe something for the addon to edit to send data in batches?

closed 2023-10-03 by thoompje

muppet3000, 2023-10-03:
@thoompje  - Just as an option - If you install the https://github.com/muppet3000/homeassistant-grott integration and disable the grott_ha one you're using, you'll get a lot more of the data available in HA.
Just an option.

thoompje, 2023-10-04:
@muppet3000 Thanks had this one installed before with the gorwatt connection, was working great! for now this is working and i'll keep an eye on it! Thanks all.

Vinisz, 2023-10-04:
> @thoompje - Just as an option - If you install the https://github.com/muppet3000/homeassistant-grott integration and disable the grott_ha one you're using, you'll get a lot more of the data available in HA. Just an option.

[image]

muppet3000, 2023-10-04:
> > @thoompje - Just as an option - If you install the https://github.com/muppet3000/homeassistant-grott integration and disable the grott_ha one you're using, you'll get a lot more of the data available in HA. Just an option.
> 
> [image]

It's not an add-on, it's an integration for displaying the information that's published to MQTT by grott.

thoompje, 2023-10-04:
> It's not an add-on, it's an integration for displaying the information that's published to MQTT by grott.

Question, I don't get any data in the intergration? [image]

With the + as well with the devicenumber/serial number. What can be wrong? So less to check or configure.

muppet3000, 2023-10-04:
Have you configured MQTT in Home assistant correctly?

Have you told Grott to publish to MQTT?

What do the Grott logs show?

thoompje, 2023-10-04:
> Have you configured MQTT in Home assistant correctly?
> 
> 
> 
> Have you told Grott to publish to MQTT?
> 
> 
> 
> What do the Grott logs show?

Only thing is that I could not co figure anonymous access in mqtt. Is there a workaround for?

muppet3000, 2023-10-04:
> > Have you configured MQTT in Home assistant correctly?
> > Have you told Grott to publish to MQTT?
> > What do the Grott logs show?
> 
> Only thing is that I could not co figure anonymous access in mqtt. Is there a workaround for?

It depends how you're running MQTT, by default it's anonymous.

muppet3000, 2023-10-04:
> > @thoompje - Just as an option - If you install the https://github.com/muppet3000/homeassistant-grott integration and disable the grott_ha one you're using, you'll get a lot more of the data available in HA. Just an option.
> 
> Hi, I am using this Integration ( https://github.com/muppet3000/homeassistant-grott ) and the inverter is publishing to Grot and relaying to Shine, I am receiving MQTT traffic to HA, but not Battery% and Export to Grid , but I can see that the Shine App is getting this info . Does Grot only relay a subset of data to MQTT ? Thanks

It depends on what type of inverter you have, you need to choose the correct value for `invtype` in the grott.ini file. The options are: `default,sph,spf,max,tl3`

thoompje, 2023-10-05:
> > > Have you configured MQTT in Home assistant correctly?
> > > Have you told Grott to publish to MQTT?
> > > What do the Grott logs show?
> > 
> > 
> > Only thing is that I could not co figure anonymous access in mqtt. Is there a workaround for?
> 
> It depends how you're running MQTT, by default it's anonymous.

Its not here
[image]

And i cannot edit mosquitto to anonymous, thats something they deleted from it. Also with MQTT Explorer i cannot connect without credentials. If you know how to edit it to anonymous that would be great.

muppet3000, 2023-10-05:
> > > > Have you configured MQTT in Home assistant correctly?
> > > > Have you told Grott to publish to MQTT?
> > > > What do the Grott logs show?
> > > 
> > > 
> > > Only thing is that I could not co figure anonymous access in mqtt. Is there a workaround for?
> > 
> > 
> > It depends how you're running MQTT, by default it's anonymous.
> 
> Its not here [image]
> 
> And i cannot edit mosquitto to anonymous, thats something they deleted from it. Also with MQTT Explorer i cannot connect without credentials. If you know how to edit it to anonymous that would be great. Here they say also that it is deleted from Mosquitto [home-assistant/addons#2623](https://github.com/home-assistant/addons/issues/2623)

I run mosquitto outside of Home Assistant and have the following config:
```
allow_anonymous true
listener 1883
persistence true
persistence_location /mosquitto/data/
log_dest file /mosquitto/log/mosquitto.log
```
Given that it's siloed off from the rest of my network it's not really an issue for me.

RVis78, 2023-10-07:
Can somebody maybe help me to install Grott > MQTT step by step in Dutch - NEDERLANDS

I already installed integration:
MQTT broker (core-mosquitto)
Grott (empty configuration) - where to find my Device id, while the + will not work! No entities are added!

Add-on:
Zigbee2MQTT installed to use zigbee devices such as motion detection and light.

I am confused because at 

    muppet3000
    /
    homeassistant-grott

use docker, while i allready have grott installed (visible in HA > Integrations)

Please can anybody help me, if possible step by step in DUTCH

muppet3000, 2023-10-09:
@RVis78 - How have you installed Grott? I see from this message that you have installed the home-assistant integration, but have you installed the actual program Grott?
Are you running HomeAssistant OS? If so, you have access to add-ons and I can give you a simple link to help get it installed through that.

RVis78, 2023-10-09:
Hi Chris,Thanks for your reply! Yes I run HA OS on a RPi4, as add-on I installed Grott, but I don’t have any idea how to install the actual program Grott! I would like to receive the single link, because I don’t have enough experience with installing it as docker, or phyton program.Thanks a lot in advance!Met vriendelijke groet,Remco Visser

muppet3000, 2023-10-09:
You can run Grott (the program) as an add-on using this repository: https://github.com/egguy/grott-home-assistant-add-on

Note, when you install it you need to make the following choices:
1. Use the 'edge' version when you install it
2. Check `enable grott MQTT`
3. Disable `HA integration`

Then data will flow into MQTT and the home-assistant grott integration will pick up the values and display them for you.

RVis78, 2023-10-10:
I’ve already run/install the program as described! I also forwarded the information on the inverter side (by adding the right ip address) enabled Grott MQTT and disabled HA integration. But there are no extra entities available in my MQTT configuration. What else can I try?

muppet3000, 2023-10-10:
Can you please share the output from the grott service then? We should be able to see the output processing messages from your inverter and forwarding them on to MQTT in there.

RVis78, 2023-10-10:
Hi again, I reinstalled everything once again! Don’t have any clue how? But it works!Met vriendelijke groet,Remco Visser

muppet3000, 2023-10-10:
hahahaha, that's brilliant. I'd just take that as a 'win' and get on with life :D 

Glad you've got it working now.