johanmeijer/grott #430: Cannot reach datalogger anymore

kmroumen, 2023-10-03:
Due to the constant issues with the Growatt API I tried to setup Grott & MQTT for integration in Home assistant. I've installed Grott via Python on a Ubuntu machine, along with mosquito. Then I've altered the IP address for my Datalogger via server.growatt.com (successfully), but now it's disconnected from growatt.com and no MQTT messages seem to be coming in. I think my Ubuntu-machine IP changed (forgot to make it static earlier), but there's no way to alter it on growatt.com anymore. Resetting of the ShineLanBox has no result (cloud icons flickers on and off). Is there any other way to restore my previous configuration?

[image]
 

[image]

[image]

johanmeijer, 2023-10-03:
Updating the shinelanbox should work. 

You can either set an url (need to enable this I think) or a ip adress. 

Can you see in the grott log (systemlog with journalctl) if grott receives data?

kmroumen, 2023-10-03:
Thanks, got access to the datalogger again and tried to update it using the correct IP address. Had no success message, but logger is offline again in server.growatt.com. Output of journalctl: https://pastebin.com/yb0aWin0

Seems good, right? Doesn't show up in MQTT explorer though.

johanmeijer, 2023-10-03:
I do not see any communication yet through grott 

Do you see any data at the growatt website?

kmroumen, 2023-10-04:
I found that trying to alter the config using the ShineLanBox web interface was working way better than via server.growatt.com. The IP address of my Grott-server was not updated correctly. Was able to update it via the web interface and now data comes in nicely and is processed as it should. Thanks for your help and your great work!

johanmeijer, 2023-10-04:
Can we close this one?

closed 2023-10-04 by kmroumen