johanmeijer/grott #432: Invalid records in the communication with Growatt 6000TL3-S

altserg, 2023-10-04:
I have Growatt 6000TL3-S 3 phase inverter with serial JNE4A140B5. 

When I do not set the inverter type in grott.ini, I receive MQTT messages but there are errors in the Growatt data parsing, see below
- Inverter serial not recognized - using inverter type default.
- Data less then minimum record length, data not processed
- Grott data ack record or data record not defined no processing done

When I set the inverter type to TLX, I receive no MQTT data, and the look looks like 2nd log below.

{"device": "JNE4A140B5", "time": "2023-10-03T16:49:10", "buffered": "no", "values": {"datalogserial": "DELETED", "pvserial": "JNE4A140B5", "pvstatus": 1, "pvpowerin": 6040, "pv1voltage": 3401, "pv1current": 9, "pv1watt": 3060, "pv2voltage": 2980, "pv2current": 10, "pv2watt": 2980, "pvpowerout": 5630, "pvfrequentie": 4993, "pvgridvoltage": 4062, "pvgridcurrent": 10, "pvgridpower": 2340, "pvgridvoltage2": 4093, "pvgridcurrent2": 9, "pvgridpower2": 2122, "pvgridvoltage3": 4062, "pvgridcurrent3": 10, "pvgridpower3": 2340, "pvenergytoday": 62, "pvenergytotal": 183548, "totworktime": 93620845, "pvtemperature": 336, "pvipmtemperature": 368, "epv1today": 37, "epv1total": 109628, "epv2today": 34, "epv2total": 96168, "epvtotal": 205796}}

**Generic invertor type log:**
`- Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.140', 49626), raddr=('47.91.67.66', 5279)>
         - Data less then minimum record length, data not processed
         
         - Growatt packet received:
                  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.2.140', 49626), raddr=('47.91.67.66', 5279)>
         - Growatt original Data:
                 \x00\x0e\x00\x06\x01\x01\x01\x04\x0d\x22\x2c\x41\x20\x46\x44\x71\x31\x59\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x25\x39\x24\x40\x35\x76\x46\x5f\x35\x54\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7e\x70\x4f\x58\x7a\x75\x61
                 \x74\x74\x6b\x72\x6e\x77\x61\x78\x05\x4b\x05\x6f\x72\x61\x74\x72\x7c\x7e\x02
                 \x77\x64\x74\x74\x41\x44\x6f\x77\x6a\x54\x67\xce\x62\x74\x77\x69\x74\x74\x40
                 \x1e\x7f\x59\x61\x7c\x74\x47\x75\x1a\x67\x62\x74\x7c\x47\x72\x68\x16\x61\x74
                 \x74\x47\x72\x6d\xba\x67\x71\xe0\x92\x40\x6e\x6d\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x07\x7e\x4e\x7b\x2b\x74\x74\x47\x5f
                 \x6f\x2e\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73\xc3\x35\x61\x74\x74
                 \x47\x72\x6e\x00\xcf\x74\x77\x64\x82\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x70\x74\x47\x98\x0f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x32\x65
         - Grott automatic protocol detection
         - Grott data record length 265
         - layout   :  T060104
         - no matching record layout found, try generic
         - Record layout used :  T06NNNN
         - Grott Growatt data decrypted
         - Growatt plain data:
                 000e0006010101044a50433641323036433600000000000000000000000000000000000000004
                 a4e45344131343042350000000000000000000000000000000000000000170a04082a15020000
                 002c000100000c710c7700050000063b0c6d00050000063600000b201389101b00080000076c1
                 02e0008000007751003000800000761000000000002cd060594d532011a000000000000000000
                 0000000000000001400c210c4a0000002d00594e2000000000000000000001ac4200000000000
                 177ae000323f00000000000000000000000000000000000040000ea6000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000004504
         - Possible Inverter serial JNE4A140B5
         - Inverter serial not recognised - using inverter type default
         - Growatt new layout processing
                 - decrypt       :  True
                 - offset        :  6
                 - record layout :  T06NNNN

         - Grott data record date/time processing started
         - date-time:  2023-10-04T08:42:21
         - Grott values retrieved:
                 -  datalogserial        :  HIDDEN
                 -  pvserial             :  JNE4A140B5
                 -  pvstatus             :  1
                 -  pvpowerin            :  318.5
                 -  pv1voltage           :  319.1
                 -  pv1current           :  0.5
                 -  pv1watt              :  159.5
                 -  pv2voltage           :  318.1
                 -  pv2current           :  0.5
                 -  pv2watt              :  159.0
                 -  pvpowerout           :  284.8
                 -  pvfrequentie         :  50.0
                 -  pvgridvoltage        :  412.3
[... 271 lines cut ...]
                 \x77\x67\x74\x74\x40\xe2\x6f\x77\x6c\x75\x67\xc2\x62\x71\x77\x69\x74\x74\x40
                 \x1f\x7f\x47\x61\x7c\x74\x47\x75\x19\x67\x62\x74\x7c\x47\x72\x68\x16\x61\x74
                 \x74\x46\x72\x6d\xba\x66\x71\xe0\x90\x6b\x6e\x6b\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x05\x7e\x4c\x7b\x30\x74\x74\x47\x5f
                 \x6f\x2e\x2f\x54\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x73\xc3\x35\x61\x74\x74
                 \x47\x72\x6e\x00\xcf\x74\x77\x64\x82\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x70\x74\x47\x98\x0f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x9c\xfa
         - Grott automatic protocol detection
         - Grott data record length 265
         - layout   :  T060104TLX
         - no matching record layout found, try generic
         - no matching record layout found, standard processing performed
         - Record layout used :  none
         - Grott Growatt data decrypted
         - Growatt plain data:
                 000f0006010101044a50433641323036433600000000000000000000000000000000000000004
                 a4e45344131343042350000000000000000000000000000000000000000170a04082e1c020000
                 002c000100000e570d8e0005000006c70c9c00060000079000000d011385101e00080000076d1
                 0300008000007761003000800000761000000010002cd070594d719011c000000000000000000
                 0000000000000001420c230c510000002d00594e2000000000000000000001ac4200000000000
                 177ae000323f00000000000000000000000000000000000040000ea6000000000000000000000
                 0000000000000000000000000000000000000000000000000000000000000000eb9b
         - Grott data ack record or data record not defined no processing done`

johanmeijer, 2023-10-04:
It seems to be OK for me (using no invtype). The data looks reasonable.

Maybe I write to much information to the log. The messages you see are not error messages but informational (debug for me). Do not worry about that to much. If I have time I will sanitise the messages in a next release (is on the todo list).

altserg, 2023-10-04:
Better reporting using python logging mechanist and not print() would be very appreciated
Closing the issue as according to the code owner the verbose log looks ok. Let me know if you need more data from my inverter to parse unknown type records in the future

closed 2023-10-04 by altserg