johanmeijer/grott #435: Inverter type and data parsing

talisker77, 2023-10-05:
Serial number, build number and device
[image]
Have selected default for invtype in grott.ini and have serial number EMFHCL900B

To me it seems like data shown in shineapp and on growatt portal differ from the data shown in HA (Home assistant).
I've made a dump of grott log and will attach later.

Question: 
- Is 'default' correct invtype to use or should I select something else.

I appreciate this project and it seems to be working better than the HACS Growatt API after api portal login issue with server.growatt, and hope this data I will attach will help make the project better.

talisker77, 2023-10-05:
Attached log from grott
[grott.log](https://github.com/johanmeijer/grott/files/12819325/grott.log)
Hope this will help

johanmeijer, 2023-10-05:
If I look at your data I think you should specify invtype=tl3. 

What model inverter do you have?

talisker77, 2023-10-06:
My inverter is a mid 6ktl3-xl. I will try that during the day.
Thanks.

talisker77, 2023-10-06:
[image]
Looks better now.

closed 2023-10-06 by talisker77