johanmeijer/grott #439: Issues decoding SPH5000 messages

patcher-ms, 2023-10-07:
Hello, I have a SPH5000 hybrid inverter and I am having issues with decoding its messages. With the default I was having reading entirely off. I set `ginvtype=sph` and that partially helped me to get more realistic results. I haven't checked them all though.

I noticed that the parsed timestamp has definitely some issues. It is way off from the actual time and, more concerning, I am getting the same timestamp for different messages, which in turn causes very little data to be available in InfluxDB (even if I take account that the timestamp is wrong).

Example log entry:
```

08/10/2023 0:20:07
	 - Growatt packet received:
08/10/2023 0:20:07
		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.19.0.8', 34876), raddr=('47.91.67.66', 5279)>
08/10/2023 0:20:07
	 - Growatt original Data:
08/10/2023 0:20:07
		 \x01\x6b\x00\x06\x03\x3f\x01\x04\x0d\x22\x2c\x4f\x22\x30\x3c\x75\x2b\x5c\x77
08/10/2023 0:20:07
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
08/10/2023 0:20:07
		 \x20\x22\x22\x20\x37\x0b\x37\x5f\x35\x33\x74\x74\x47\x72\x6f\x77\x61\x74\x74
08/10/2023 0:20:07
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7e\x73\x54\x62\x69\x74\x61
08/10/2023 0:20:07
		 \x74\x74\x3b\x72\x69\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
08/10/2023 0:20:07
		 \x77\x61\x74\x74\x47\x72\x6f\x76\x23\xd4\x7d\x81\x75\xbc\x7f\xc6\x74\x8e\x47
08/10/2023 0:20:07
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
08/10/2023 0:20:07
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x74
08/10/2023 0:20:07
		 \x14\x67\xff\x4e\xcb\x6f\x7c\x61\x74\x77\x3c\x72\x6f\x77\x61\x74\x74\x47\x72
08/10/2023 0:20:07
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x2c\x74
08/10/2023 0:20:07
		 \x47\x3e\xc3\x76\x23\x3a\xf2\x47\x72\x6f\x4e\x61\x74\x45\xec\x72\x6f\x77\x49
08/10/2023 0:20:07
		 \x74\x74\x67\x33\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
08/10/2023 0:20:07
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
08/10/2023 0:20:07
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x25\x6f\x75
08/10/2023 0:20:07
		 \x47\x46\x58\x6e\x6a\x61\x74\x76\x4b\x62\xd6\x67\x85\x74\x74\x47\x72\x6f\x77
08/10/2023 0:20:07
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
08/10/2023 0:20:07
		 \x6f\x77\x60\x74\x74\x47\xb4\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x61\x74\x74
08/10/2023 0:20:07
		 \x47\x72\x6f\x77\x61\x74\x77\xaf\x76\x0b\x77\x67\x74\x74\x47\x72\x6f\x77\x61
08/10/2023 0:20:07
		 \x74\x74\x67\x72\x6f\x77\x61\x74\x74\x47\x72\x6a\x0f\x61\x74\x74\x47\x70\x62
08/10/2023 0:20:07
		 \x77\x5f\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
08/10/2023 0:20:07
		 \x72\x6f\xc4\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xc1\x6f\x77\x64\x60
08/10/2023 0:20:07
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6a\x63\x61\x74\x74\x50\x72\x69\x7c
08/10/2023 0:20:07
		 \xf6\x5c\xc0\x47\x72\x6f\x72\x61\x74\x75\x7e\x72\x6f\x77\x53\x74\x74\x73\x0a
08/10/2023 0:20:07
		 \x6f\x77\x61\x6b\x74\x47\x7f\xf0\x77\x61\x74\x58\x47\x72\x7c\xc3\x61\x74\x74
08/10/2023 0:20:07
		 \x61\x72\x6f\x6f\xdb\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
08/10/2023 0:20:07
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
08/10/2023 0:20:07
		 \x77\x62\x9c\x74\x47\x72\x0e\x77\x61\x74\x74\x47\x4c\x7b\xf5\x9f\xaa\x74\x50
08/10/2023 0:20:07
		 \x5a\xdb\x69\xdf\x45\x36\xc1\xf4\x6f\x74\x61\x3e\x74\x23\x64\x5f\x77\x61\x74
08/10/2023 0:20:07
		 \x74\x47\x72\xe9\xf1\x61\x74\x74\x47\x72\x6f\x36\x31\x35\x24\x47\x72\x63\xba
08/10/2023 0:20:07
		 \x6d\xbd\x74\x46\x7c\x7f\x7b\xaa\x78\xb8\x4b\xbe\x63\xbc\x6d\xbf\x78\x8b\x7e
08/10/2023 0:20:07
		 \xa3\x7b\xaa\x78\xb8\x4b\xbe\x63\xbc\x6d\xb8\x74\x47\x76\x0a\x73\x80\x74\x75
08/10/2023 0:20:07
		 \x47\x72\x6f\xb1\x61\x74\x74\x47\x76\x62\x77\x61\x74\x74\x47\x72\x6f\x77\x61
08/10/2023 0:20:07
		 \x74\x74\x47\x72\x6f\x77\x32\x74\x74\x0d\x2f\x6f\x77\x61\x50\x74\x47\x65\x7e
08/10/2023 0:20:07
		 \x77\x61\x77\xf0\x47\x72\x6c\xf3\x61\x74\x74\x27\x72\x6e\x77\x61\x7b\xef\x47
08/10/2023 0:20:07
		 \x73\x6f\x77\x73\xdf\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
08/10/2023 0:20:07
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
08/10/2023 0:20:07
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
08/10/2023 0:20:07
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
08/10/2023 0:20:07
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x7e\xa3\x7b\xa8
08/10/2023 0:20:07
		 \x74\x75\x47\x62\x6f\x71\x61\x75\x75\x4c\x72\xb8\x77\x62\x74\x75\x47\x72\x6f
08/10/2023 0:20:07
		 \x49\x61\x4a\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x51\x6f\xb7\x61\x74\x74\x47
08/10/2023 0:20:07
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
08/10/2023 0:20:07
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
08/10/2023 0:20:07
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
08/10/2023 0:20:07
		 \x6f\xb0\xad
08/10/2023 0:20:07
	 - Grott automatic protocol detection
08/10/2023 0:20:07
	 - Grott data record length 839
08/10/2023 0:20:07
	 - layout   :  T060104XSPH
08/10/2023 0:20:07
	 - no matching record layout found, try generic
08/10/2023 0:20:07
	 - Record layout used :  T06NNNNXSPH
08/10/2023 0:20:07
	 - Growatt data decrypted V2
08/10/2023 0:20:07
	 - Grott Growatt data decrypted
08/10/2023 0:20:07
	 - Growatt plain data:
08/10/2023 0:20:07
		 016b0006033f01044a50433843444832593300000000000000000000000000000000000000004
08/10/2023 0:20:07
		 f554354434c453042520000000000000000000000000000000000000000170a07131006030000
08/10/2023 0:20:07
		 007c00060000000000000000000000000000000000000000000142a009c607d308a700fa00000
08/10/2023 0:20:07
		 00000000000000000000000000000000000000000000000000000000000000000000000037513
08/10/2023 0:20:07
		 8b09b9000b0000037b000000000000000000000000000000000000000000000000005800004ca
08/10/2023 0:20:07
		 c01424e8600000039000031ab0000002800002041000000000000000000000000000000000000
08/10/2023 0:20:07
		 0000000000000000000000000000000000000000000000000000000000000000520e0133012a0
08/10/2023 0:20:07
		 11d0000020c10b910e40000000000000000000000000000000000000000000000000000000100
08/10/2023 0:20:07
		 0000c600000000000000010000000000000000000003e80464000600000000000000000020000
08/10/2023 0:20:07
		 0000000000000057800000000020d003e00000000000000000000000000000000000000b30000
08/10/2023 0:20:07
		 000000000000000000b3000005140000000000000000000005140000001700060b9728b400000
08/10/2023 0:20:07
		 0050000013900000032000034780000001f00000d9f0000002c000013b400000026000018ba00
08/10/2023 0:20:07
		 00000000000000000000000000000000000000000000000000000000000000000003e80000006
08/10/2023 0:20:07
		 100000000003e1482fede001728b41ebe314286860003004a0064163000000000000086860000
08/10/2023 0:20:07
		 000000004150415000000ccd0cc900010e100ccb0ccc0ccc0ccb0ccb0ccc0ccc0ccb0ccc0ccc0
08/10/2023 0:20:07
		 ccb0ccc0000046504e10001000000c600000000040d0000000000000000000000000000005300
08/10/2023 0:20:07
		 004a5d0000002400001711000003840000038400000060000100000f9b0001000012ab0000000
08/10/2023 0:20:07
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
08/10/2023 0:20:07
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
08/10/2023 0:20:07
		 000000000010ccc0cc90001001000060001010b00d7000300010000003e003e00000000000000
08/10/2023 0:20:07
		 00002300c00000000000000000000000000000000000000000000000000000000000000000000
08/10/2023 0:20:07
		 000000000000000000000000000000000000000000000000000000000c7cc
08/10/2023 0:20:07
	 - Growatt new layout processing
08/10/2023 0:20:07
		 - decrypt       :  True
08/10/2023 0:20:07
		 - offset        :  6
08/10/2023 0:20:07
		 - record layout :  T06NNNNXSPH
08/10/2023 0:20:07
08/10/2023 0:20:07
	 - Grott data record date/time processing started
08/10/2023 0:20:07
	 - date-time:  2023-10-07T19:16:06
```

If I search the logs for `date-time` I see that the same value has been outputted multiple times:
```
08/10/2023 0:16:07
	 - date-time:  2023-10-07T19:16:06
08/10/2023 0:17:07
	 - date-time:  2023-10-07T19:16:06
08/10/2023 0:18:07
	 - date-time:  2023-10-07T19:16:06
08/10/2023 0:19:07
	 - date-time:  2023-10-07T19:16:06
08/10/2023 0:20:07
	 - date-time:  2023-10-07T19:16:06
08/10/2023 0:21:07
	 - date-time:  2023-10-07T19:16:06
```
Other values (e.g. `totworktime`) are different so it looks like there is an issue decoding the timestamp.

I'm happy to investigate this further myself but I'd appreciate an hint on where to look

patcher-ms, 2023-10-08:
Looking closer into this and matching with the code, `170a07131006` should be parsed as `2023-10-07T19:16:06`.
Any idea why multiple messages sent by the datalogger could have the same timestamp?

patcher-ms, 2023-10-09:
Using the timezone from the server (via the env variable `gtime="server"`) allows me to see all the measurements in InfluxDB. I'll later go through all of them to double check that the mapping is correct for my inverter.

johanmeijer, 2023-10-10:
The only reason I can think of is that either the datalogger is re-sending the same data (but then the data should be also the same) or the datalogger can not keep its time (some dataloggers has problems with that). Then gtime="server" is indeed a solution. 

What type of datalogger do you have?

patcher-ms, 2023-10-13:
I have a ShineWIFI-S, running firmware 3.0.0.2 which as far as I understand is the latest. Using `gtime="server"` is an acceptable solution.

I also have a question. I see that I am getting two different kind of messages:
```
- Grott automatic protocol detection
- Grott data record length 585
- layout   :  T060136XSPH
- no matching record layout found, try generic
- Record layout used :  T060136XSPH
```
and
```
- Grott automatic protocol detection
- Grott data record length 839
- layout   :  T060104XSPH
- no matching record layout found, try generic
- Record layout used :  T06NNNNXSPH
```

The second message seems to work as intended. For the first message I also get a `- Grott data ack record or data record not defined no processing done` in the logs. Is that expected? The payload seems to contain some data but I yet have to decode it.
```
\x01\x35\x00\x06\x02\x41\x01\x36\x0d\x22\x2c\x4f\x22\x30\x3c\x75\x2b\x5c\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x20\x22\x22\x20\x37\x0b\x37\x5f\x35\x33\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7e\x79\x42\x62\x46\x75\x18
\x6c\x0d\xd3\x72\x6a\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
\x77\x61\x74\x74\x49\x25\x6f\x76\x68\xf6\x74\x46\x72\x6f\x77\x61\x74\x74\x47
\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x47\x72\x6f\x77\x61\x74\x74\x44\xf6\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x4c\x1a\x6f\x77
\x61\x74\x74\x47\x61\xec\x7e\xe5\x74\x74\x47\x72\x6f\x67\x61\x74\x74\x47\x72
\x6f\x69\xae\x75\x54\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x47\x72\x6f\x77\x61\x74\x0d\xd2\x08\x7e\x77\x61\x74\x74\x47\x72\x6f\x77\x61
\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x70\x47\x72\x7b
\x0d\x61\x74\x74\x4f\x72\x6f\x79\x5b\x74\x74\x97\xbc\x6f\x77\xb1\xba\x76\x46
\x8d\x90\x88\x94\x74\x7e\x47\x16\x6f\x77\x51\x88\x74\x47\x76\x83\x77\xb3\x74
\x85\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x2e\x4a
```

anoppe, 2023-10-13:
> I have a ShineWIFI-S, running firmware 3.0.0.2 which as far as I understand is the latest. Using gtime="server" is an acceptable solution.
I had the same issue, see #442 . I've solved my issue also by using the server date time instead of the one sent by the data logger. I'm still not sure if the issue was caused by the datalogger or that it was because of an interpretation issue, I didn't had the motivation to investigate that.,

anoppe, 2023-10-13:
> I also saw this in the logs after I change `invtype` parameter value to `min`.

For what it's worth, hereby my output:
```
- Growatt original Data:
		 \x00\x44\x00\x06\x02\x41\x01\x04\x1f\x35\x2b\x41\x22\x3c\x3c\x72\x21\x35\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3e\x2e\x2d\x46\x37\x0d\x43\x5f\x45\x35\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7e\x79\x4a\x6b\x7d\x75\x6a
		 \xcc\x78\x73\x72\x6e\x77\x61\x72\x41\x42\x0c\x6f\x72\x61\x74\x76\xa5\x77\x38
		 \x77\x67\x74\x74\x44\x21\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x42\xde\x7c\xf2\x68\x5d
		 \x74\x40\x72\x6f\x72\xa1\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x4e\x5b\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x72\xa0\xb2\x72\x6f\x77\x69\x74\x74\x47\xff\x6f\x77\x61\xe4\x74
		 \x47\x72\x6b\x77\x61\x74\x38\x47\x72\x6f\x73\x61\x74\x74\x03\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x48\xa9\x6e\x41\x61\x74\x74\x47\x72\xf1\x77\x61\x75
		 \x6c\x47\x72\x6f\x77\x61\x74\x75\x10\x7c\x60\x77\x61\x3a\x54\x47\x77\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x78\x72\x7e\xde\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\xa4\x74\x2b\x72\x6f\x72\xed\x7a\x55\x42\xfc\x6f\x77\x61\xe2\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xc7\x36
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060104XMIN
	 - no matching record layout found, try generic
	 - no matching record layout found, standard processing performed
	 - Record layout used :  none
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00440006024101045847443643484835535a00000000000000000000000000000000000000005
		 1594c32434a313032540000000000000000000000000000000000000000170a0d0d1912020bb8
		 0c34000100000635057e0005000002e2055700060000035300000000000000000000000000000
		 0000000000000000000000005ac138509290007000005c0000000000000000000000000000000
		 000929000000000000000000000000000000000006d4f5000000080000008d000000900000000
		 40000004c00000004000000440000000000000000000000000000000000000000000000000000
		 000000000000000000000000000000000000000000000fdb013600000000009e0000011800000
		 000000001570e0f00004e20000500000000000000000000000000000000000000000000000000
		 0000000000000000000000000000000000000000000c350cb1000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000d0006c0000058c0e21058e00000096000000000000000000000
		 00000000000b559
	 - Grott data ack record or data record not defined no processing done

```

alsocalledchris, 2023-10-17:
I'm also having issues with the same SPH inverter - for me the 'pvenergytoday' is not correct - it is currently showing 6.4 in Grott's log and then that is translated to 64 in the sent MQTT message - the Shine App is showing 12..1 (I have 2 strings). (running Docker image 2.8.2)

johanmeijer, 2023-10-19:
> I'm also having issues with the same SPH inverter - for me the 'pvenergytoday' is not correct - it is currently showing 6.4 in Grott's log and then that is translated to 64 in the sent MQTT message - the Shine App is showing 12..1 (I have 2 strings). (running Docker image 2.8.2)

I do not think that is a problem. Grott logs show a formated value of the pvenergytoday (specified in the layout definition). 
The MQTT messages (and Influxdb) contains the unformatted values as beeing sent by the inverter. The entry today is being sent in 0.1 Kwh units. This 64 * 0.1 watt is 6.4 Kwh

johanmeijer, 2023-10-19:
@patcher-ms 

_The second message seems to work as intended. For the first message I also get a - Grott data ack record or data record not defined no processing done in the logs. Is that expected? The payload seems to contain some data but I yet have to decode it._

That is correct the first record (the "03" data type) is a announcement record which is not decoded (and does not need to be decoded) by Grott.

alsocalledchris, 2023-10-19:
> I do not think that is a problem. Grott logs show a formated value of the pvenergytoday (specified in the layout definition).
>
> The MQTT messages (and Influxdb) contains the unformatted values as [...]

Thanks for your reply. You are correct that the MQTT was correct for the given value of 64 to 6.4 however my point was that it should have been 12.1Kwh or I guess 121 - the value is not correct from the template (sorry for not being clearer). Is there a way to debug this?