johanmeijer/grott #441: "Ghost" unit in homeassistant

parhedberg, 2023-10-09:
Hi there! 

I have a Growatt 10000 TL3-S.

I am having an issue since i first started using Grott about a year ago.. 
In Homeassistant (using the grott HACS plugin) i get two units. One with real values, and one with bananas values. 

I thought I was doing something wrong and started to investigate things. So now i have the latest 2.8.2 version of grott, and latest HACS grott version and still got this issue.. 

In the logs it seems that the data is sometimes corrupt? But that is just a wild guess. 

This is some logs from the docker container:

`Grott Growatt logging monitor : 2.8.2

Grott Command line parameters processed:
	verbose:     	 True
	config file: 	 grott.ini
	output file: 	 <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
	nomqtt:      	 False
	inverterid:  	 automatic
	pvoutput:    	 False
	blockcmd:    	 False
	noipf:       	 False

Grott process configuration file

Grott process environmental variables

	Pulled 'gblockcmd=False' from the environment

	Pulled 'gnomqtt=False' from the environment

	Pulled 'gmqttip=10.0.2.3' from the environment

	Pulled 'gmqttauth=True' from the environment

	Pulled 'gmqttuser=parhedberg' from the environment

	Pulled 'gmqttpassword=br3db4nD' from the environment

	Pulled 'gpvoutput=True' from the environment

	Pulled 'gpvapikey=9fab81ab644c4bb8661f924066557fe693ef5816' from the environment

	Pulled 'gpvsystemid1=92619' from the environment

	Pulled 'ginflux=False' from the environment

	Pulled 'ginflux2=False' from the environment

	Pulled 'gifdbname=grottdb' from the environment

Grott InfluxDB server IP address env invalid

	Pulled 'gifport=8086' from the environment

	Pulled 'gifuser=grott' from the environment

	Pulled 'giforg=grottorg' from the environment

	Pulled 'gifbucket=grottdb' from the environment

	Pulled 'giftoken=influx_token' from the environment

	Pulled 'gextension=false' from the environment

	Pulled 'gextname=grottext' from the environment

Grott override settings if set in commandline
[... 515 lines cut ...]
		 \x72\x6f\x77\x61\x74\x74\x47\x73\xe2\x7c\x83\x7f\xaf\x47\x72\x6f\x5a\x61\x2d
		 \x3a\x67\x72\x6f\x77\x61\x74\x74\x47\xdd\x6f\x75\xb6\xe0\x74\x47\x72\xd1\x77
		 \x63\x8b\x41\x47\x77\xb9\xbe\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x43\x72\x6e\xf1\xc1\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\xce\xef
	 - Grott automatic protocol detection
	 - Grott data record length 225
	 - layout   :  T055104
	 - no matching record layout found, try generic
	 - Record layout used :  T05NNNN
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00e7000500d951044e4143333931323637384e464238393135303536000000000000020000002
		 c000100000f5c12c200040000078013a80004000007dc00000db8138a0fec0005000004960fd0
		 00050000048e0fee0005000004960000015b00058e0a06b2610d014e000000000000000000000
		 00000000000018d0be20bdb0000002d00594e2000000000000000af0002d794000000be0002ff
		 350005d6c9000000000000000000000000000000000004000186a000000000000000000000000
		 0000000000000000000000000000000000000000000000000000000000000ba9b
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T05NNNN
`

anoppe, 2023-10-09:
@parhedberg your pvoutput key is visible for everybody…

johanmeijer, 2023-10-10:
Wow that is strange. It seems dat the datalogger "forgets" to encrypt the data. For your ghost record the original data is unencrypted and the decrypted data is encrypted. 

For me this seems to be a datalogger error. Grott is checking the CRC (the data  checksum) and that seems to be oke. Maybe an update of the firmware of the datalogger might help.