johanmeijer/grott #446: Getting negative value of previous day in the morning

saurjk, 2023-10-12:
Hi, 

I am getting everyday between 8 to 9, I get a surge of negative value from the total of the previous day's value, for example:

[image]

I see the following when looking through the sensor

[image]

which probably explains the reason for the negative values in the morning. How would I fix this? Do I need to change something in the ini file or am I using the wrong sensor? I am using :  Power Generation - Total (eactotal)

Thanks.

Neflardio, 2023-10-12:
You need to use "lifetime solar generation" sensor rather than the daily solar generation sensor when configuring the energy dashboard in HA.

Neflardio, 2023-10-12:
If those sensors represent the total solar generation of your system thus far, they will work. They both have the same value so either will work. The main thing is using a sensor that does not give a daily reading. HA compares the last read value with the new value to give work out solar generation. A sensor with a daily readout resets to 0, which is why HA will give a value equivalent to negative previous day generation in the morning if using a daily sensor.

closed 2023-10-12 by saurjk