johanmeijer/grott #447: MOD 10KTL3-XH not showing batteries charged/discharged today

sidequest-tomas, 2023-10-12:
Have tried to make battery values show, but with no luck. Any suggestions? 
I set inverter type to default or tl3 then i get state of charge, but no other battery info.
 
Logged as issue here: https://github.com/muppet3000/homeassistant-grott/issues/49#issuecomment-1758156236

rabbadab, 2023-10-17:
I have the same Inverter but when I set it with tl3 i do not get anything about my ARK battery. How can I contribute with data to be able to get all this information to MQTT? I use the lates Docker version if that helps. 
**My setup is like this:**
I have one inverter: MOD 10KTL3-XH
3 battery modules named: ARK 2.5H-A1 
Smart meter and a ShineLan-X

This is the data I got when I use TL3 but it's several strange values here like the frequency that is 13.7 Hz... 

`- Growatt original Data:
	 \x00\x98\x00\x06\x02\x41\x01\x04\x00\x22\x28\x47\x22\x3c\x26\x74\x41\x22\x77
	 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	 \x3e\x33\x2d\x45\x37\x0c\x39\x5e\x45\x26\x74\x74\x47\x72\x6f\x77\x61\x74\x74
	 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x75\x6a
	 \xcc\x78\x73\x72\x6e\x77\x61\x74\x74\x4b\x70\x6f\x77\x61\x74\x74\x47\x77\x79
	 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
	 \x72\x6f\x77\x61\x74\x76\x55\x72\x6f\x73\xe6\x74\x74\x45\x69\x7c\xfe\x69\x83
	 \x74\x41\x72\x6f\x72\x00\x7d\x7c\x47\x77\x6f\x77\x65\xf0\x7c\xb4\x72\x69\x77
	 \x61\x71\x2a\x48\x19\x60\x05\x6e\xe3\x74\x47\x54\x0e\x77\x61\x74\x74\x47\x72
	 \x47\x04\x60\x8a\x70\x5d\x72\x6f\x77\xe9\x74\x75\x4f\xe5\x6f\x76\x65\x78\x74
	 \x47\x72\x2d\x77\x61\xf1\x6d\x47\x72\x6f\x49\x61\x74\x0a\xb4\x72\x6f\x77\x61
	 \x74\x74\x47\x72\x6f\x77\x61\xaa\x74\x47\xb3\xfb\x77\x61\x74\x72\x47\x72\x5a
	 \xa4\x61\x74\x75\x13\x72\x6e\xf0\xab\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
	 \x72\xef\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6e\x60\x99\x74
	 \xcb\x46\x01\x6f\xc7\x61\x74\x74\xb3\x79\x99\x7b\x68\x3a\x54\x47\x72\x6f\x76
	 \xe7\xd4\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
	 \x6f\x77\x61\x74\x74\x47\xc6\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
	 \x45\x60\x6f\x77\x61\xf5\x78\x72\x7e\xde\x77\x61\x74\x34\x47\x72\x56\x8e\x61
	 \x74\x74\x03\x72\x6f\x36\x9d\x74\x74\x47\x71\x6f\x77\x65\xbb\x74\x46\x71\x77
	 \x77\x61\x74\x0e\x47\x72\xa2\x34\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
	 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
	 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x77\xe7\x61\x74\x74\x52\x70\x6e\x77
	 \x61\x76\xc9\x16\xb6\x6f\x75\x61\x7d\x63\xbf\x79\xc0\x77\x63\x74\x76\x47\xa8
	 \x6f\x77\x61\x74\x75\xad\x72\x6f\x77\x61\x74\x74\x7e\x8b\x6f\x77\x20\x88\x74
	 \x47\x72\x6e\x7b\x28\x74\x77\x47\x42\x6f\x77\x61\xdb\x74\xd4\x72\x6d\x77\x6c
	 \x74\x74\x47\x72\x6f\x76\x61\x74\x74\x69\x72\x41\x77\x61\x74\x74\x47\x72\x6f
	 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\x61\x74\x74\x47
	 \x72\x66\x26\x57\x8b\x96\x46\x4d\x66\xb3\x68\xb0\x74\x47\x72\x0b\x2f\xa1\x3d
	 \xb4\x47\x72\x6f\x77\x49\xad\x74\x47\x72\x6f\x7b\xa0\x78\xd6\x16\x6a\x6f\x77
	 \x61\x76\x74\x47\x73\x11\x02\x99\x74\x74\x44\x9a\x3a\x5f\x61\x74\x74\x47\x72
	 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xe7\x7c
 - Grott automatic protocol detection
 - Grott data record length 585
 - layout   :  T060104XTL3
 - no matching record layout found, try generic
 - Record layout used :  T06NNNNXTL3
 - Growatt data decrypted V2
 - Grott Growatt data decrypted
 - Growatt plain data:
	 00980006024101044750473043485233334d00000000000000000000000000000000000000005
	 1444c31434b4b3132470000000000000000000000000000000000000000000000000000020bb8
	 0c340001000000000c02000000000000051600000000000000000000000000000000000000000
	 00000000212000004870000021b138908f7000600000561090800050000048408f30006000005
	 5e0f6b0f720f9700002661000000000000287301fe041a00000088000108970001040c0000004
	 2000085190000003e00007ef30000000000000000000000de0000c19400000006000035d30000
	 0154000187ca00000000000000000000008000000000fffa000000000000000117f800bf01730
	 0b0000000f40bf60c094e200000000186a0000000000000000000000000000000000000000000
	 0000b40000000000010000000000000212000000810c350cb100000040000039f900000044000
	 041fc00000003000004cf000103180000007a0000cd4300000000000000000000000000000000
	 0000000000000000000000000000000000000000271000001890000000150201000002bd51c40
	 002000917f80baf0002000200da0000000001ea00000000000039f9000041fc000000010c4900
	 030030000000af00930002000d0000000000010000002e002e000000000000000000000000000
	 000000000000000040000000000095136ffe2013f09c409c40000006458c049c00000000028d9
	 000000000cc10ca25118000000020000017e75f8000003e855280000000000000000000000000
	 000000000009513
 - Growatt new layout processing
	 - decrypt       :  True
	 - offset        :  6
	 - record layout :  T06NNNNXTL3
 - Grott data record date/time processing started
 - no or no valid time/date found, grott server time will be used (buffer records not sent!)
[... 30 lines cut ...]
	 -  epv2today            :  0.0
	 -  epv2total            :  0.0
	 -  epvtotal             :  7167.2
	 -  pvtemperature        :  19.1
	 -  pvipmtemperature     :  37.1
	 -  pvboottemperature    :  17.6
	 -  uwBatVolt_DSP        :  24.4
	 -  pbusvolt             :  3062
	 -  nbusvolt             :  3081
 - MQTT jsonmsg: 
		 {"device": "QDL1CKK12G", "time": "2023-10-17T17:42:35", "buffered": "no",
		 "values": {"pvstatus": 1, "pvpowerin": 0, "pv1voltage": 3074, "pv1current":
		 0, "pv1watt": 0, "pv2voltage": 1302, "pv2current": 0, "pv2watt": 0,
		 "pvpowerout": 393216, "pvfrequentie": 1374, "pvgridvoltage": 3947,
		 "pvgridcurrent": 3954, "pvgridpower": 261554176, "pvgridvoltage2": 9825,
		 "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 10355,
		 "pvgridcurrent3": 510, "pvgridpower3": 68812800, "Vac_RS": 136, "Vac_ST": 1,
		 "Vac_TR": 2199, "pvenergytoday": 66572, "pvenergytotal": 66, "totworktime":
		 34073, "epv1today": 62, "epv1total": 32499, "epv2today": 0, "epv2total": 0,
		 "epvtotal": 71672, "pvtemperature": 191, "pvipmtemperature": 371,
		 "pvboottemperature": 176, "uwBatVolt_DSP": 244, "pbusvolt": 3062,
		 "nbusvolt": 3081}}
 - MQTT message message sent
 - Grott Send data to PVOutput disabled 
 - Grott Send data to Influx disabled `

johanmeijer, 2023-10-19:
Can you please specify invtype=min to see if this solves the problem? be aware you have to run the beta version of Grott for this!

rabbadab, 2023-10-19:
> Can you please specify invtype=min to see if this solves the problem? be aware you have to run the beta version of Grott for this!

Can I run the beta with docker image? I'm just calling for ledidobe/grott:latest when I build it...

rabbadab, 2023-10-19:
Okej fast lunch test. It looks much better I can see SOC and frequency is correct and so on. I have not have time to digg deeper now but will try to compare it with server.growatt data view. Thank you Johan!

MuadDib007, 2024-06-29:
Same issue here with the MOD 7000TL3-XH(BP) with the APX Battery.
No data / incorrect data for Phase 2 and 3.
Also battery charge and discharge data missing.
Tried with min, tl3, max, spa
Somehow with the beta of grott I seem to miss the verbose output to include here.
Any hint on how to get that going?

MuadDib007, 2024-06-30:
In my case running the ledidobe/grott with beta version 2.8.3 it seems to work with invtype=min and HA is picking up on the values including the battery totals
Inverter type is MOD 7000 TL3-XH BP with APX 15 kWh battery connected