johanmeijer/grott #453: Issue with mqttinverterintopic using v2.7.8

ORi0N, 2023-10-15:
```
[Generic]
invtype=default
invtypemap = {'GPxx': 'sph', 'TPxx': 'sph', '5Fxx': 'default'}
ip = x.x.x.x
port = 5279

[Growatt]

[MQTT]
mqttinverterintopic = True
ip = x.x.x.x
port = 1883
topic = energy/growatt
auth = True
user = xxx
password = xxx

[PVOutput]

[influx]

[extension]
```

But somehow it just keeps posting in the general topic without deviceid in it:

Oct 15 17:59:56 energy/growatt {“device”: “5FZxx”, “time”: “2023-10-15T16:3 9:52”, “buffered”: “yes”, “values”: {“datalogserial”: “4KZxx”, “pvserial”: “5FZxx”, “pvstatus”: 1, “pvpowerin”: 4686, “pv1voltage”: 2458, “pv1current” : 9, “pv1watt”: 2333, “pv2voltage”: 3150, “pv2current”: 7, “pv2watt”: 2353, “pvp owerout”: 4594, “pvfrequentie”: 4995, “pvgridvoltage”: 2401, “pvgridcurrent”: 19 , “pvgridpower”: 4609, “pvgridvoltage2”: 0, “pvgridcurrent2”: 0, “pvgridpower2”: 0, “pvgridvoltage3”: 0, “pvgridcurrent3”: 0, “pvgridpower3”: 0, “pvenergytoday” : 165, “pvenergytotal”: 342570, “totworktime”: 195029729, “pvtemperature”: 401, “pvipmtemperature”: 422, “epv1today”: 78, “epv1total”: 162578, “epv2today”: 88, “epv2total”: 182327, “epvtotal”: 344905}}

ORi0N, 2023-10-20:
Nobody is using the "mqttinverterintopic' setting?

johanmeijer, 2023-10-20:
Oh yes. Totally forgotten about this8 one. This a documentation (or coding?) error. 

You have to specify : inverterintopic = True.

Put it on the todolist.

jacco-hass, 2023-10-30:
Both options do not work in docker, and I don't see an environment variable for it in docker_compose.yaml. Is that missing?

closed 2023-10-30 by ORi0N