johanmeijer/grott #458: Missing 2nd & 3rd Phase Power SPH 6000 TL3 BH-UP

pimisen, 2023-10-16:
I am using the Homeassistant grott Integration of muppet3000 and noticed that it does not show the values for the power of 2nd & 3rd phase. First created an Issue their, but then i had a look at the mqtt message and saw, the values are not send by the grott Server:
Could you have a look please, why they are missing? You Work is much appreciated already :-)

So th issue seems to be on the grott-Server side:
`{
"device": "TLJ4CGA00R",
"time": "2023-10-16T08:58:31",
"buffered": "no",
"values": {
"pvstatus": 5,
"pvpowerin": 7700,
"pv1voltage": 2071,
"pv1current": 21,
"pv1watt": 4513,
"pv2voltage": 4814,
"pv2current": 6,
"pv2watt": 3258,
"pvpowerout": 898,
"pvfrequentie": 4997,
"pvgridvoltage": 3917,
"pvgridcurrent": 6,
**"pvgridpower": 4302,**
"pvgridvoltage2": 4123,
"pvgridcurrent2": 6,
**"pvgridpower2": 0,**
"pvgridvoltage3": 4035,
"pvgridcurrent3": 6,
**"pvgridpower3": 0,**
"totworktime": 21156656,
"eactoday": 21,
"pvenergytoday": 21,
"eactotal": 7897,
"epvtotal": 10930,
"epv1today": 1,
"epv1total": 4570,
"epv2today": 0,
"epv2total": 6360,
"pvtemperature": 359,
"pvipmtemperature": 359,
"pvboosttemp": 361,
"bat_dsp": 1586,
"eacharge_today": 14,
"eacharge_total": 891,
"batterytype": 1,
"uwsysworkmode": 5,
"systemfaultword0": 0,
"systemfaultword1": 0,
"systemfaultword2": 0,
"systemfaultword3": 0,
"systemfaultword4": 0,
"systemfaultword5": 0,
"systemfaultword6": 0,
"systemfaultword7": 2048,
"pdischarge1": 0,
"p1charge1": 5700,
"vbat": 1577,
"SOC": 12,
"pactouserr": 0,
"pactousertot": 0,
"pactogridr": 0,
"pactogridtot": 0,
"plocaloadr": 2100,
"plocaloadtot": 2100,
"spdspstatus": 5,
"spbusvolt": 7233,
"etouser_tod": 39,
"etouser_tot": 458,
"etogrid_tod": 0,
"etogrid_tot": 2424,
"edischarge1_tod": 9,
"edischarge1_tot": 5058,
"eharge1_tod": 1,
"eharge1_tot": 5290,
"elocalload_tod": 41,
"elocalload_tot": 9258
}
}

QoS: 0 - Retain: false `

pimisen, 2023-10-16:
Here is a log sample

`2023/10/16 15:29:51 | stdout | - Data less then minimum record length, data not processed
2023/10/16 15:29:51 | stdout | <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.6', 5279), raddr=('172.17.0.1', 45543)>
2023/10/16 15:29:51 | stdout | - Growatt packet received:
2023/10/16 15:29:51 | stdout |  
2023/10/16 15:29:51 | stdout | - Grott extension processing disabled
2023/10/16 15:29:51 | stdout | - Grott Send data to Influx disabled
2023/10/16 15:29:51 | stdout | - Grott Send data to PVOutput disabled
2023/10/16 15:29:51 | stdout | - MQTT message message sent
2023/10/16 15:29:51 | stdout | "elocalload_tot": 9292}}
2023/10/16 15:29:51 | stdout | "eharge1_tod": 68, "eharge1_tot": 5357, "elocalload_tod": 75,
2023/10/16 15:29:51 | stdout | "etogrid_tot": 2484, "edischarge1_tod": 12, "edischarge1_tot": 5061,
2023/10/16 15:29:51 | stdout | "spbusvolt": 7329, "etouser_tod": 40, "etouser_tot": 459, "etogrid_tod": 60,
2023/10/16 15:29:51 | stdout | "pactogridtot": 42300, "plocaloadr": 0, "plocaloadtot": 0, "spdspstatus": 5,
2023/10/16 15:29:51 | stdout | "SOC": 100, "pactouserr": 0, "pactousertot": 0, "pactogridr": 42300,
2023/10/16 15:29:51 | stdout | "systemfaultword7": 256, "pdischarge1": 0, "p1charge1": 0, "vbat": 1597,
2023/10/16 15:29:51 | stdout | "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0,
2023/10/16 15:29:51 | stdout | "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0,
2023/10/16 15:29:51 | stdout | 895, "batterytype": 1, "uwsysworkmode": 5, "systemfaultword0": 0,
2023/10/16 15:29:51 | stdout | "pvboosttemp": 414, "bat_dsp": 1607, "eacharge_today": 18, "eacharge_total":
2023/10/16 15:29:51 | stdout | 132, "epv2total": 6492, "pvtemperature": 469, "pvipmtemperature": 438,
2023/10/16 15:29:51 | stdout | 7980, "epvtotal": 11088, "epv1today": 27, "epv1total": 4596, "epv2today":
2023/10/16 15:29:51 | stdout | "totworktime": 21203493, "eactoday": 104, "pvenergytoday": 104, "eactotal":
2023/10/16 15:29:51 | stdout | "pvgridvoltage3": 4063, "pvgridcurrent3": 52, "pvgridpower3": 0,
2023/10/16 15:29:51 | stdout | "pvgridvoltage2": 4184, "pvgridcurrent2": 52, "pvgridpower2": 0,
2023/10/16 15:29:51 | stdout | "pvgridvoltage": 3968, "pvgridcurrent": 52, "pvgridpower": 37427,
2023/10/16 15:29:51 | stdout | "pv2watt": 39353, "pvpowerout": 37278, "pvfrequentie": 5000,
2023/10/16 15:29:51 | stdout | "pv1current": 4, "pv1watt": 984, "pv2voltage": 4738, "pv2current": 83,
2023/10/16 15:29:51 | stdout | "values": {"pvstatus": 5, "pvpowerin": 40300, "pv1voltage": 2081,
2023/10/16 15:29:51 | stdout | {"device": "<x-d by me>", "time": "2023-10-16T15:16:19", "buffered": "no",
2023/10/16 15:29:51 | stdout | - MQTT jsonmsg:
2023/10/16 15:29:51 | stdout | -  elocalload_tot       :  929.2
2023/10/16 15:29:51 | stdout | -  elocalload_tod       :  7.5
2023/10/16 15:29:51 | stdout | -  eharge1_tot          :  535.7
2023/10/16 15:29:51 | stdout | -  eharge1_tod          :  6.8
2023/10/16 15:29:51 | stdout | -  edischarge1_tot      :  506.1
2023/10/16 15:29:51 | stdout | -  edischarge1_tod      :  1.2
2023/10/16 15:29:51 | stdout | -  etogrid_tot          :  248.4
2023/10/16 15:29:51 | stdout | -  etogrid_tod          :  6.0
2023/10/16 15:29:51 | stdout | -  etouser_tot          :  45.9
2023/10/16 15:29:51 | stdout | -  etouser_tod          :  4.0
2023/10/16 15:29:51 | stdout | -  spbusvolt            :  732.9
2023/10/16 15:29:51 | stdout | -  spdspstatus          :  0.5
2023/10/16 15:29:51 | stdout | -  plocaloadtot         :  0.0
2023/10/16 15:29:51 | stdout | -  plocaloadr           :  0.0
2023/10/16 15:29:51 | stdout | -  pactogridtot         :  4230.0
2023/10/16 15:29:51 | stdout | -  pactogridr           :  4230.0
2023/10/16 15:29:51 | stdout | -  pactousertot         :  0.0
2023/10/16 15:29:51 | stdout | -  pactouserr           :  0.0
2023/10/16 15:29:51 | stdout | -  SOC                  :  1.0
2023/10/16 15:29:51 | stdout | -  vbat                 :  159.7
2023/10/16 15:29:51 | stdout | -  p1charge1            :  0.0
2023/10/16 15:29:51 | stdout | -  pdischarge1          :  0.0
2023/10/16 15:29:51 | stdout | -  systemfaultword7     :  256
2023/10/16 15:29:51 | stdout | -  systemfaultword6     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword5     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword4     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword3     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword2     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword1     :  0
2023/10/16 15:29:51 | stdout | -  systemfaultword0     :  0
2023/10/16 15:29:51 | stdout | -  uwsysworkmode        :  5
2023/10/16 15:29:51 | stdout | -  batterytype          :  1
2023/10/16 15:29:51 | stdout | -  eacharge_total       :  89.5
2023/10/16 15:29:51 | stdout | -  eacharge_today       :  1.8
2023/10/16 15:29:51 | stdout | -  bat_dsp              :  160.7
2023/10/16 15:29:51 | stdout | -  pvboosttemp          :  41.4
2023/10/16 15:29:51 | stdout | -  pvipmtemperature     :  43.8
2023/10/16 15:29:51 | stdout | -  pvtemperature        :  46.9
[... 95 lines cut ...]
2023/10/16 15:29:51 | stdout | \x6f\x77\x73\x74\x74\x44\x0d\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x61\x74\x74
2023/10/16 15:29:51 | stdout | \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2023/10/16 15:29:51 | stdout | \xa1\x46\xc4\x6e\xe9\x61\x74\x72\x00\x6e\xcf\x77\x61\x74\x74\x47\x72\x6f\x77
[... cut ...]
2023/10/16 15:29:51 | stdout | \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
2023/10/16 15:29:51 | stdout | \x01\x86\x00\x06\x03\x3f\x01\x04\x1f\x35\x2b\x41\x22\x3f\x36\x73\x33\x29\x77
2023/10/16 15:29:51 | stdout | - Growatt original Data:
2023/10/16 15:29:51 | stdout | <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.6', 57590), raddr=('47.91.67.66', 5279)>
2023/10/16 15:29:51 | stdout | - Growatt packet received:
2023/10/16 15:29:51 | stdout

<br class="Apple-interchange-newline">
</body>
</html>`

johanmeijer, 2023-10-19:
Strange I going to look at this.

pimisen, 2023-10-19:
@johanmeijer the "xxxxx-d by me" is not Part of the message. I just removed my Inverter SN from the log :-)

pimisen, 2023-10-24:
@johanmeijer Supported the investigation & your efforts with a small paypal donation .... Not enough, not much, but hope its good for a couple of Spezi :-)

johanmeijer, 2023-10-29:
I looked into your data and did the mapping again, but sorry this is really what is shown in the data (and with the documented register mappings). 

It is the first data from a 3 phase SPH I have seen, so I can not compare it. Maybe somebody with a 3 phase SH can also share his data so we can compare. 

In the meanwhile is it possible for you to update the firmware of the datalogger? And do you see different values in the phone app or online?

paralichko, 2025-02-23:
Same here, with SPH10000TL3 BH-UP 

[image]