johanmeijer/grott #461: Support MQTT Discovery for HomeAssistant

splitbrain, 2023-10-20:
There are currently multiple efforts to integrate Grott easier into HomeAssistant. There is the grott_ha.py in the examples directory which is also included in the grott-addon by @egguy and then there is [homeassistant-grott](https://github.com/muppet3000/homeassistant-grott) by @muppet3000 which seems to be similar.

However, since grott already publishes to MQTT, the easiest integration would be the MQTT broker integration using [MQTT Discovery](https://www.home-assistant.io/integrations/mqtt#mqtt-discovery). From what I understand, grott would need to publish a discovery topic that explains the available topics and they would become available as HomeAssistant entities automatically. No additional extensions or configurations needed.

muppet3000, 2023-10-20:
The @egguy add-on is Grott just bundled in a friendly way to run as part of home assistant.

The grott-intergration takes to content published to MQTT and formats it nicely.

Does the method you've described for MQTT processing in home assistant include the specification of units of measurement? Whether or not it is a historic metric? Which icon to use for the values? Human readable names for the values?

If it does provide all of those things, surely that means that info has to be published onto MQTT which means there's a load of excess information on MQTT for people that aren't planning on using MQTT.

I think myself @johanmeijer and @egguy all agree that we could do some better collaboration and documentation, it's just a case of finding the time to actually do it.

splitbrain, 2023-10-20:
> Does the method you've described for MQTT processing in home assistant include the specification of units of measurement?

yes, see the example here: https://www.home-assistant.io/integrations/mqtt#sensors

egguy, 2023-10-21:
Hi, 

This is what `grott ha` extension does. When receiving a measure message from the inverter, the plugin generates all the MQTT discovery information (e.g. measure name, total_increasing type. Icon) by using the internal mapping (with some custom overrides...). This is the same system used by zigbee2mqtt

I created the ha extension to be the simplest possible and have the least number of decencies, so it will be easier to install. In consequence, there is less post-processing (e.g. summing `epv` output).

The `addon` allows you ton install and configure it in one click, but you can fully disable the extension and MQTT output, will just have a basic `grott` install to just send to pvoutput for example.

Many people disable the integrated HA add-on to use your extension.

But yeah, I agree we should coordinate the documentation effort. I started to improve the doc of my extension with step how to integrate with yours if they didn't want the duplicate sensors: https://github.com/egguy/addon-grott-beta/blob/main/grott-beta/DOCS.md