johanmeijer/grott #462: Missing values

simon2022, 2023-10-20:
Hello,
first of all: really great software, thanks for that.

My problem:
I am currently trying to read two values which are displayed in the ShinePhone app:

[image]

Unfortunately, I haven't been able to figure out what names these have in grott/mqtt. 
I would really appreciate if someone could help me out.

Thanks a lot

simon2022, 2023-10-26:
More information:
- I use the Shine Lan-X Stick.
- I use plain grott with mqtt

I always get two different data intercepted in the Grott log:

 Growatt plain data 1:
                 00c300060241010447504730434b4d32483700000000000000000000000000000000000000005
                 34c4c3043484730365a0000000000000000000000000000000000000000000000000000020bb8
                 0c34000100000009014c000000000000012d00000000000900000000000000000000000000000
                 000000001e9000000000000018d1382092c0002000001a3000000000000000000000000000000
                 00092c00000000000009490000000000000b3200adf09f000000290000225d0000256a0000001
                 30000108000000016000014ea00000000000000000000002200000c6f00000000000004d30000
                 004800002d1900000000000000000000002900000000fffa0000000000000000000000f600f60
                 000000001240ea800004e200000000000000000000000000000000000300030002e0000000000
                 00003c00000000000100000000000001e90000002e0c350cb100000011000006f400000014000
                 006e10000000800000077000025f10000002d0000219600000000000000000000000000000000
                 0000000000000000000000000000000000000000271000000f4e000000150201000002bd3ca90
                 002000a0ea307a20002000200e30000000001ae00000000000006f4000006e100000001070100
                 190001000000bb00ae0001000a000000000001000000310031000000000000000000000000000
                 0000000000000000400000000000a3c6effe2015b09c409c40000006442903750000000001e3d
                 000000000ca80c7f3c6e000000010001000000000000000000000021001e01781068000100090
                 007000200000f1d

MQTT jsonmsg 1:
                         {"device": "SLL0CHG06Z", "time": "2023-10-26T20:05:09", "buffered": "no",
                         "values": {"datalogserial": "GPG0CKM2H7", "pvserial": "SLL0CHG06Z",
                         "pvstatus": 1, "pvpowerin": 9, "pv1voltage": 332, "pv1current": 0,
                         "pv1watt": 0, "pv2voltage": 301, "pv2current": 0, "pv2watt": 9,
                         "pvpowerout": 397, "pvfrequentie": 4994, "pvgridvoltage": 2348,
                         "pvgridcurrent": 2, "pvgridpower": 419, "pvgridvoltage2": 0,
                         "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0,
                         "pvgridcurrent3": 0, "pvgridpower3": 0, "totworktime": 11399327,
                         "pvenergytoday": 41, "pvenergytotal": 8797, "epvtotal": 9578, "epv1today":
                         19, "epv1total": 4224, "epv2today": 22, "epv2total": 5354, "pvtemperature":
                         246, "pvipmtemperature": 292, "battery1soc": 10}}

Growatt plain data 2:
                 00c4000600e6012047504730434b4d32483700000000000000000000000000000000000000005
                 34c4c3043484730365a000000000000000000000000000000000000000000000000000000a000
                 0000860000093a000009380000093b000000050000000500000006000001c7000003bd000003a
                 f0000049e0000049d000005adfffffbbdfffffd4dfffffbb1000001800000032b0000028a0000
                 0933000012e1fffff4ba0000027a000001f200000ff700000ff900000ffa000012c6000009c80
                 00000d000000ad000001ed400001c8e00000ba000000000000000000000000000000000000000

MQTT jsonmsg 2:
                         {"device": "GPG0CKM2H7", "time": "2023-10-26T20:05:14", "buffered": "no",
                         "values": {"datalogserial": "GPG0CKM2H7", "pvserial": "SLL0CHG06Z",
                         "voltage_l1": 2362, "voltage_l2": 2360, "voltage_l3": 2363, "Current_l1": 5,
                         "Current_l2": 5, "Current_l3": 6, "act_power_l1": 455, "act_power_l2": 957,
                         "act_power_l3": 943, "app_power_l1": 1182, "app_power_l2": 1181,
                         "app_power_l3": 1453, "react_power_l1": -1091, "react_power_l2": -691,
                         "react_power_l3": -1103, "powerfactor_l1": 384, "powerfactor_l2": 811,
                         "powerfactor_l3": 650, "pos_rev_act_power": 2355, "pos_act_power": 2355,
                         "rev_act_power": 2355, "app_power": 4833, "react_power": -2886,
                         "powerfactor": 634, "frequency": 498, "L1-2_voltage": 4087, "L2-3_voltage":
                         4089, "L3-1_voltage": 4090, "pos_act_energy": 4806, "rev_act_energy": 2504}}

These two data sets always alternate every 2 minutes (my update interval).

Please let me know if I should provide any more info. I would really love to get this done.

SebastianGode, 2023-11-06:
He @simon2022 what inverter do you have?
See this issue:
https://github.com/johanmeijer/grott/discussions/477

If it's an SPH inverter you will need to set the inverter type correctly.

simon2022, 2023-11-10:
@SebastianGode thank you for your suggestion.
Unfortunately, I get incorrect values using the SPH entry. I also do not have an SPH inverter.

So far I have been able to find out that the T060104X layout is used. Unfortunately, the values I want to read out (for example pactouser) are not defined in this layout-json in Grott.
And I have no idea how to get the values to define them.

simon2022, 2023-11-10:
I'm using: Growatt Hybrid Inverter MIN 4200TL-XH

simon2022, 2023-11-10:
Sometimes this Message occurs:

         - Record layout used :  T060103X
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 007c00060241010347504730434b4d32483700000000000000000000000000000000000000005
                 34c4c3043484730365a0000000000000000000000000000000000000000000000000000020000
                 007c0001019100000064000027100000a41003e8414c312e30005a414241000a0002000003e80
                 03c003c006400640000534c4c3043484730365a00000000000100640000000020202050562049
                 6e7665727465722020000013ec020107e7000b000a001200180039000507300a55128e141e064
                 a0a55128e141e064a0a55128e141e07a309e2137e139200640032003200320032003200320032
                 00320032000a000a09e20000414c42413034303130313130013100020000139c003209b409e20
                 844081600140005096f08fc2710271027102710271027100096000001e400ff4e2000ff4e2000
                 ff4e2000ff4e203e0900000f01002a000103e800000bb80c340000534c4c3043484730365a000
                 0000000000000000000000000000000000000000001f40000019000010000157c000002580000
                 000000000000000016a800000000000000000000004b000a051e0a008a010e008e01150000000
                 51d00000064006400000000051d00000000000000000000000000000000000000000000000000
                 00000000000000000000000000000000000000000000000000000000000000000000000000000
                 000000000000051514b3330303030323331313030435300005a4542410000564341410004080f
                 00040002000100000000001004000000003000000103000a00000000000000000000000000000
                 00000000000d843
         - Grott data ack record or data record not defined no processing done

And no processing is made.

johanmeijer, 2023-11-27:
The T060103X does not need any processing while thi sis not a dat record (it is an "announcement" record). This record is ignored by Grott.

simon2022, 2023-12-06:
@johanmeijer thanks for pointing this out!

But unfortunately this does not solve the actual problem.
I need access to the two values highlighted in the first post.

If there is anything else I can do to help by providing more data, please let me know.

Thanks a lot

simon2022, 2024-05-05:
@johanmeijer Can I help you in any way by sending you more data? I would really like to solve the problem after half a year. Thanks a lot!

PedroKTFC, 2024-05-05:
Do you use the Grott integration ([here](https://github.com/muppet3000/homeassistant-grott))? That might help with getting the values you want. I then get values that look like the image below. Perhaps you can work out which of them are the ones you want?

[image]

johanmeijer, 2024-05-17:
Can you try it with invertertype=min? According the record information that should fit better.