johanmeijer/grott #463: Values are wrong by factor of 10

PietFourie, 2023-10-22:
Hi.
I have a SPF5000 ES inverter installed.
I entered the Inverter type to SPF in the grott.ini file. It now provides the correct variables.
In the command window (logging) the list of values, "**- Grott values retrieved:**" are correct and correspond to the output in the Growatt server and app.
However, the **"MQTT jsonmsg:"** has all the values multiplied by 10. I am not sure why but this provides the wrong values to the MQTT server.
Maybe this can be considered a bug?
[Bug.txt](https://github.com/johanmeijer/grott/files/13063766/Bug.txt)

johanmeijer, 2023-10-22:
No this is not a bug. In the MQTT values the original values comming from the inverter are being used. Most often this means actual power is display in 0.1 W, energy in 0.1Kwh, Volt in 0.1 V. (as documented in the Growatt Modbus protocol documentation)

In the grott log the values are formatted to a more "readable" value. But for the MQTT message processing client (e.g HA, Nodered) it means it has to take care of the formatting/calculation.

PietFourie, 2023-10-22:
Thank you for the clarification. I thought it might be something like that. That's why the question mark .... -:)

ORi0N, 2023-11-05:
In your MQTT sensors use the value_template to correct *10 or /10

```
- state_topic: energy/growatt/TPXXXXX
  value_template: "{{ value_json['values']['pvpowerin'] | int / 10 }}"
  unique_id: growatt_actual_input_power_TPXXXXX
  device_class: power
  unit_of_measurement: "W"
  name: "Growatt TPXXXXXX - Input watt (Actual)"
```

johanmeijer, 2023-12-04:
Can we close this issue (@PietFourie)