johanmeijer/grott #464: time stamps are off in data?

kafeenstra, 2023-10-22:
I've created a custom extension for grott, which puts the data into a mysql database. It seems to run fine, except that timestamps are often off. There are three weird things happening:
- about once a day, around midnight, an entry appears with a timestamp in 2066
- at seemingly random moments, one entry has its timestamp skip forward by an hour. I can still see the daily yield gradually incrementing, so I know they are in the right order in the database.
- some time in the afternoon (somewhere between 14.00 and 16:00 CET) timestamps skip back to 08.00's or 09:00's -- minutes still seem to increment as expected, so it seems to offset is whole hours. Sometimes, the offset disappears again.
Graphs and other statistics on the growatt website are all looking fine. My local graph that is built out of the database, obviously, isn't entirely correct. I put the timestamps received by grott into the mysql database 'as is' only using strptime, with no timezone conversion.
Am I the only one experiencing this?
Any ideas how to get to the bottom of this?

kafeenstra, 2023-10-22:
I show a few examples here, first of the 2066 entry:
```
+---------------------+--------+---------+----------+------------+--------------+------------+-----------+-------------+
| timestamp           | status | pgrid_w | eday_kwh | etotal_kwh | grid_voltage | emonth_kwh | eyear_kwh | portal_type |
+---------------------+--------+---------+----------+------------+--------------+------------+-----------+-------------+
| 2023-10-21 11:27:12 | NULL   |     7.9 |      0.6 |        633 |        232.4 |       NULL |      NULL | GroWatt     |
| 2023-10-21 11:02:58 | NULL   |     7.1 |      0.6 |        633 |          231 |       NULL |      NULL | GroWatt     |
| 2023-10-21 11:07:09 | NULL   |     8.4 |      0.6 |        633 |        231.2 |       NULL |      NULL | GroWatt     |
| 2066-06-10 23:06:55 | NULL   |       0 |        0 |        633 |        231.7 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:51:13 | NULL   |       0 |        0 |        633 |        233.3 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:55:29 | NULL   |       0 |        0 |        633 |        231.5 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:59:45 | NULL   |       0 |        0 |        633 |        232.3 |       NULL |      NULL | GroWatt     |
```
(so, the normal dates before the 2066 one seem to be from the morning, but I think they're shifted here as well. The ones starting from 08:51 are in the morning and seem to be correct. 
Then, the one hour off single entry:
```
| 2023-10-22 13:35:15 | NULL   |    34.1 |      0.3 |      633.3 |        238.8 |       NULL |      NULL | GroWatt     |
| 2023-10-22 13:39:26 | NULL   |    34.4 |      0.3 |      633.3 |        238.8 |       NULL |      NULL | GroWatt     |
| 2023-10-22 14:43:59 | NULL   |    35.2 |      0.3 |      633.3 |        238.5 |       NULL |      NULL | GroWatt     |
| 2023-10-22 13:47:48 | NULL   |    38.4 |      0.3 |      633.3 |        238.5 |       NULL |      NULL | GroWatt     |
| 2023-10-22 13:51:59 | NULL   |    73.3 |      0.3 |      633.3 |        238.1 |       NULL |      NULL | GroWatt     |
```
This was a slow day, so no visible increments in daily yield (`eday_kwh`), but they're simply all the same (so certainly not horribly out of order).

```
| 2023-10-22 14:29:44 | NULL   |    58.4 |      0.3 |      633.3 |        235.7 |       NULL |      NULL | GroWatt     |
| 2023-10-22 14:33:57 | NULL   |    73.6 |      0.3 |      633.3 |        235.6 |       NULL |      NULL | GroWatt     |
| 2023-10-22 14:38:09 | NULL   |    65.4 |      0.4 |      633.4 |        236.2 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:44:27 | NULL   |    40.5 |      0.4 |      633.4 |        236.2 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:48:39 | NULL   |    47.8 |      0.4 |      633.4 |        236.1 |       NULL |      NULL | GroWatt     |
| 2023-10-22 08:52:51 | NULL   |    42.5 |      0.4 |      633.4 |        232.6 |       NULL |      NULL | GroWatt     |
```
and then later that same day
```
| 2023-10-22 10:04:10 | NULL   |    61.8 |      0.4 |      633.4 |        236.6 |       NULL |      NULL | GroWatt     |
| 2023-10-22 10:08:22 | NULL   |    30.5 |      0.4 |      633.4 |        235.6 |       NULL |      NULL | GroWatt     |
| 2023-10-22 10:30:55 | NULL   |    22.7 |      0.4 |      633.4 |          235 |       NULL |      NULL | GroWatt     |
| 2023-10-22 17:44:05 | NULL   |    21.5 |      0.4 |      633.4 |        232.8 |       NULL |      NULL | GroWatt     |
| 2023-10-22 17:49:03 | NULL   |    23.1 |      0.4 |      633.4 |        233.1 |       NULL |      NULL | GroWatt     |
| 2023-10-22 17:54:01 | NULL   |      27 |      0.4 |      633.4 |        234.9 |       NULL |      NULL | GroWatt     |
```
Again, note  how the daily yield keeps incrementing steadily (same columns as earlier example)

johanmeijer, 2023-10-24:
Some dataloggers show strange data/time behavior. I am not sure which models/versions but you can search in the issues and see that some people has problems. 

What can you do? 
- Update the firmware of the datalogger (maybe it helps)
- Set time=server in the grott.ini. This will ignore the inverter /date time in the record and use the server time. 
- Override date/time in your extension with whatever you want.