johanmeijer/grott #466: Influxdb timestamps not properly reflecting timezone offset from UTC

tabascoz, 2023-10-25:
Hi There, 

I moved to Portugal and i am setting up another solar installation, based on the same hardware i own in Brazil. I noticed that grafana graphs were delayed by one hour. Started to investigate the issue, so my conclusion was that Influxdb time was using local tine rather than UTC. 
I enabled the debug and got it the following in syslog:
```

...
out 25 18:43:48 note grott[13781]:         timezone:                     local
...
out 25 18:39:47 note grott[13693]:          - Grott original time :  2023-10-25T18:39:42 adjusted UTC time for influx :  2023-10-25T18:39:42`
```

My system is Gentoo - locale properly set as shown below:
```
note ~ # timedatectl
               Local time: qua 2023-10-25 18:41:03 WEST
           Universal time: qua 2023-10-25 17:41:03 UTC
                 RTC time: qua 2023-10-25 17:41:03
                Time zone: Europe/Lisbon (WEST, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```
If I change system timezone to Etc/GMT-1 it fixes the issue, like below:
```
note ~ # timedatectl 
               Local time: qua 2023-10-25 18:43:01 +01
           Universal time: qua 2023-10-25 17:43:01 UTC
                 RTC time: qua 2023-10-25 17:43:01
                Time zone: Etc/GMT-1 (+01, +0100)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

And influxdb data got a proper offset to record UTC. 

`out 25 18:43:54 note grott[13781]:          - Grott original time :  2023-10-25T18:43:49 adjusted UTC time for influx :  2023-10-25T17:43:49`

Do you think there's a problem with python pytz to process Europe/lisbon timezone?

I am using 2.8 branch checked out yesterday. 

Thanks, Fernando

johanmeijer, 2023-10-29:
Strange for Europe/Amsterdam it is working fine.

djgrubson, 2024-04-16:
had similar issues, looks like adding to docker-compose
- gtimezone=europe/warsaw fix this 

`Grott original time :  2024-04-16T16:13:57 adjusted UTC time for influx :  2024-04-16T14:13:57`