johanmeijer/grott #469: MID 15 KTL3-XH

flopp999, 2023-10-27:
I have tried using "invtype = sph" but the data is very strange, Grid Frequency is between 15-65 Hz :)

Do I have to change the offset or maybe this inverter is not added?

flopp999, 2023-10-27:
[image]

johanmeijer, 2023-10-29:
Can you try to run it with invtype=tl3, this might fit better.

flopp999, 2023-10-29:
Hi I tried tl3 but still strange frequency and SOC for battery is missing

johanmeijer, 2023-10-29:
Can you sent m ethe original growatt data from the grott  log?

flopp999, 2023-10-29:
Growatt original Data:
                 \x00\x32\x00\x06\x03\x3f\x01\x04\x00\x22\x28\x47\x22\x33\x43\x76\x28\x28\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x24\x3b\x2f\x44\x30\x70\x36\x5f\x45\x2d\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x74\x6a
                 \xcc\x78\x73\x72\x6e\x77\x61\x74\x74\x44\x35\x6f\x77\x61\x74\x74\x47\x73\x93
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7e\xc5\x8b\x8b\xba\xd8\x7c\xf0\x69\x91
                 \x74\x40\x72\x6f\x71\x58\x7d\x73\x47\x75\x6f\x77\x67\x25\x7c\xb8\x72\x69\x77
                 \x61\x71\x11\x48\x0d\x60\xf8\x6e\xe7\x74\x47\x5e\x12\x77\x61\x74\x74\x47\x72
                 \x47\xf6\x61\x6c\x66\x82\x72\x6f\x77\x43\x74\x74\x44\x69\x6f\x77\x63\x47\x74
                 \x47\x72\x69\x77\x61\x74\xa0\x47\x72\x6f\x64\x61\x74\x75\x18\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x60\x29\x74\x47\x62\xbd\x77\x61\x74\x74\x47\x72\x6f
                 \xed\x61\x74\x75\x42\x72\x6f\x67\x55\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x76\x77\x61\x74\x74\xb8\x88\x6f\x77\x61\x74\x74\x47\x72\x6f\x6b\x0d\x75
                 \x3c\x46\xc1\x6e\x3f\x61\x74\x75\x09\x7c\x57\x79\x5c\x3a\x54\x47\x72\x6f\x75
                 \x28\x84\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x4e\x6f\x77\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x6f\x78\x72\x7e\xde\x77\x61\x74\x79\x47\x72\x6e\x54\x61
                 \x74\x74\x39\x72\x6f\x74\x3b\x74\x74\x47\x14\x6f\x77\x63\x42\x74\x47\x70\xa6
                 \x77\x61\x74\x6e\x47\x72\x6d\x5a\x61\x74\x74\x46\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x53\x64\x47\x72\x73\x62\x61\x74\x74\x4c\x73\x6e\x77
                 \x61\x76\xcb\xe5\x54\x6f\x75\x61\x10\x68\xc7\x7c\x26\x77\x60\x74\x74\x46\x40
                 \x6e\xeb\x61\x74\x74\x47\x72\x6f\x74\x9d\x74\x74\x46\x51\x6f\x77\x62\x2e\x74
                 \x47\x72\x6d\x79\x57\x74\x45\x47\x5a\x6f\xad\x60\x77\x74\xf1\x72\x6a\x77\x50
                 \x74\x10\x47\x16\x6f\x73\x61\x74\x75\xcb\x73\xff\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x73\x61\x74\x74\x47
                 \x72\x0b\xd5\x47\x74\x60\x47\x72\x7c\xff\x34\x84\x74\x47\x72\x0b\x61\x51\x66
                 \x54\x47\x72\x6f\x77\x65\xc4\x74\x47\x72\x6e\x7a\x6e\x79\x73\x53\x94\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x02\x99\x74\x74\x42\xae\x3e\x37\x3c\x63\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x7e\xdd\x7a\x4f\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\xba\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x44\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x2c\x2c\x1f\x2a\x6f\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a
                 \x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x2c\x1f\x2a\x37\x2f\x39\x2c\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x4f\xa7
         - Grott automatic protocol detection
         - Grott data record length 839
         - layout   :  T060104XTL3
         - no matching record layout found, try generic
         - Record layout used :  T06NNNNXTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted

flopp999, 2023-10-29:
I have smart meter SDM630MCT.
Firmware in Inverter is ZBDB-0006, if that is a matter of problem.
Latest firmware seems to be 0014 but I needed 0006 to be able to use the system as a Helsingborg system to balance the net

johanmeijer, 2023-10-31:
Ok your data record looks more like T:_X (Min-type record) can you please run invtype=min? 

the SOC value is then shown at the bdc1_soc keyword (first BCD Power module). 

If the SDM630 is connected to the inverter then a separate datarecord is sent and a separate MQTT message is created. Can you look at the Grott log if you can find one?

flopp999, 2023-10-31:
With "min" it finds the data logger(I think) GPGxxxxxx.
I am pretty new with Growatt so I have not learned how it works, yet.
It doesn't find any data for the battery. Before I had the KLNxxxx but that one is not created. I deleted and added Grott integration in HA.
Tomorrow and can try to look through the MQTT messages.
The battery is presented as CXMxxxxx in the Shine app.
[image]

flopp999, 2023-11-01:
So this is what I get when I use -v and changes "minrecl" from 100 to 1
I saw message coming from those layouts:
T060116MIN,T060118MIN, T060119MIN, T060137MIN(could see my serial number for the battery in here), T060120, T060150XMIN, T060150MIN, T060103XMIN, T060104MIN, T060138XMIN
Anyone of these that can help you?

flopp999, 2023-11-01:
T060137XMIN I can see my serial number for each battery module I have

flopp999, 2023-11-01:
If I comment-out Invtype it choose T060104X

>  Grott data record length 839
>          - layout   :  T060104X
>          - no matching record layout found, try generic
>          - Record layout used :  T06NNNNX
>          - Growatt data decrypted V2
>          - Grott Growatt data decrypted
> 
and it shows SOC for battery1 in the debug but I do not get it in HA
[image]

Simenglert, 2024-02-23:
I have the same inverter. Did you find a way to get the battery data in HA? In my case theres no battery data at all no matter what invertype I am using. For me the best way is using invertype "mod". Which one are you using?