johanmeijer/grott #474: Grottserver on Docker stops silently

Elexy, 2023-11-02:
I am running Grottserver on an [ARM device](http://nanopi.io/nanopi-Fire3.html) and after startup it works for a while but then the log messages just stop and nothing can connect to it anymore. The run time decreases if I start a second container on different ports. The second one, stops within minutes, while the "1st" one stays alive for a few hours. Restarting makes them work again, but the stop silently in a similar manner.

I see these messages in the syslog, also when the service work normally. Not sure if they have anything to do with it. 
```
[Thu Nov  2 13:46:55 2023] cp15barrier_handler: 3162 callbacks suppressed
[Thu Nov  2 13:46:55 2023] "python" (17033) uses deprecated CP15 Barrier instruction at 0xf7c732e0
[Thu Nov  2 13:46:55 2023] "python" (17033) uses deprecated CP15 Barrier instruction at 0xf7c732e8
[Thu Nov  2 13:46:55 2023] "python" (17033) uses deprecated CP15 Barrier instruction at 0xf7c732e0
````
My workaround it to run it on a linux machine with x86 arch. There I seem to have no problem.
I'd love to run this on mi SBC, so let me know what I can do to debug and / or provide more info.

Elexy, 2023-11-03:
Some additional information: 
running on x86 doe not help much as the service stops working silently after a while as well.
The difference is that the last log messages are still there. 

Adding a healthcheck now so I can automatically restart it if that fails.
```
 healthcheck:
      test: ["CMD", "curl", "-f", "http://localhost:5782"]
      interval: 1m30s
      timeout: 10s
      retries: 2
      start_period: 40s
```

Elexy, 2023-11-03:
Supecting similar issues as https://github.com/johanmeijer/grott/issues/418

FreezyExp, 2023-11-05:
Hmm perhaps connecting to the command line of the docker and checking what the status is of the proccesses running.
I kind of gave up checking the ports as I could not find a fix, my workaround for #418 was to stop the docker an hour after sundown and starting it an hour before sundown. With the added bonus of less processing while the inverter is idle.

Elexy, 2023-11-05:
Thanks @FreezyExp , we're off the grid, so the Inverters are working 24/7. 
Untill I get to the bottom of this I'll resort to periodical restarts. It seems to be something with my setup, otherwise more people would be complaining, I suspect.

Elexy, 2023-11-07:
FYI, my temporary restart workaround is running this on a 5 minute cron
`[ $(docker logs --since=1m grottserver1 | wc -m) -eq 0 ] && /usr/local/bin/docker-compose -f /path/to/docker-compose.yml restart grott1`

johanmeijer, 2023-12-04:
I am not sure what happens. I not running the grottserver container in production. I will start doing it for a while and see if it also happens at my site.

martreumer, 2024-02-01:
I am running ledidobe/grott container and am experiencing similar problems. Restarting the container seems to fix the issue so I have implemented a daily restart of the container. In the middle of the night when there's no sun to avoid dataloss. But am curious to follow this issue.

rany2, 2024-02-01:
@martreumer do you face the same issue if you use the ghcr.io/rany2/grott:master image instead?

martreumer, 2024-02-02:
I use docker compose, can I just stop the old container, replace the image in my compose file with rany2/grott:master and start the container? Or is the configuration file any different?

rany2, 2024-02-02:
@martreumer the configuration isn't any different, just replace the image in the compose file

martreumer, 2024-02-02:
@rany2 will try and keep you updated

marcel-hi, 2024-03-07:
Is there any update/solution. Since last week I run this docker-container on my raspberry pi and it also stops every night. Restarting is solving the issue, so it seams the same?

anoppe, 2024-03-07:
What might be more robust as opposed to configuring a corn job is to make the docker runtime responsible to keep your container running at all times, is to configure restart behavior: https://docs.docker.com/config/containers/start-containers-automatically/
For all my critical containers, I have configured: unless-stopped (otherwise you can't stop the container to do maintenance for example ;-) )

marcel-hi, 2024-03-07:
@anoppe ik heb inderdaad ook "restart: unless-stopped" opgenomen. Het lijkt erop dat de container overigens zelf niet stopt, maar de communicatie naar MQTT en naar de server van growatt stopt wel.

marcel-hi, 2024-03-07:
yes, it is strange, I am considering adding a cron job to restart it every night like @martreumer . However I was hoping @martreumer might have an other solution or an update on the issue.

anoppe, 2024-03-07:
Nothing is logged at the time it occurs?

marcel-hi, 2024-03-07:
I don't think very interessting log messages. This are messages when I read the  logs of the docker with 'docker-compose logs grott'
There are no timestamps by the way, I don't know why.

Just after de last correct message:
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 5279), raddr=('172.18.2.126', 12407)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 	 - Grott connection error
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 5279), raddr=('172.18.2.126', 12407)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 	 - peer has disconnected
[36mgrott              |[0m 	 - ('172.18.2.126', 18566) has connected
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 5279), raddr=('172.18.2.126', 18566)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 5279), raddr=('172.18.2.126', 18566)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Data less then minimum record length, data not processed
[36mgrott              |[0m 
[36mgrott              |[0m 	 - Growatt packet received:
[36mgrott              |[0m 		  <socket.socket fd=4, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('192.168.1.10', 57332), raddr=('47.91.67.66', 5279)>
[36mgrott              |[0m 	 - Growatt original Data:
[36mgrott              |[0m 		 \x00\x02\x00\x05\x00\xd9\x01\x03\x0d\x22\x2c\x43\x58\x40\x40\x71\x31\x2d\x34
[36mgrott              |[0m 		 \x32\x31\x46\x7e\x46\x59\x47\x59\x37\x74\x47\x72\x6f\x77\x61\x76\x74\x47\x72
[36mgrott              |[0m 		 \x43\x76\x60\x74\x75\x47\x72\x6f\x13\x61\x10\x53\x57\x72\x6f\x16\xc9\x7a\x64
[36mgrott              |[0m 		 \x06\x3a\x5e\x59\x51\x74\x74\x47\x72\x6f\x77\x61\x74\x75\x47\x76\x6c\x9f\x61
[36mgrott              |[0m 		 \x6a\x73\x03\x78\x29\x65\xf0\x60\x6a\x04\x21\x2a\x45\x58\x40\x42\x77\x4a\x2c
[36mgrott              |[0m 		 \x77\x71\x85\x65\x47\x73\x6f\x77\x61\x74\x74\x47\x72\x6f\x70\x54\x7e\x32\x55
[36mgrott              |[0m 		 \xe3\x7c\xee\x66\xea\x7d\xa5\x60\xfc\x64\xec\x74\xa6\x45\x73\x6f\x5a\x61\x2d
[36mgrott              |[0m 		 \x73\xaf\x72\x6c\x77\x67\x74\x73\x47\x7b\x6f\x5d\x62\x9c\x77\xaf\x71\x87\x74
[36mgrott              |[0m 		 \x89\x77\x9c\x44\x9a\x6c\x9f\x62\x9c\x33\x35\x1d\x18\x16\x15\x00\x54\x0e\x1c
[36mgrott              |[0m 		 \x19\x12\x13\x00\x11\x35\x13\x07\x15\x00\x45\x4c\x76\x4a\x6f\x77\x61\x77\x75
[36mgrott              |[0m 		 \x76\x72\x6f\x77\x60\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x46\x61\xf3\x77\xa9
[36mgrott              |[0m 		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x9f\xde
[36mgrott              |[0m 	 - Grott automatic protocol detection
[36mgrott              |[0m 	 - Grott data record length 225
[36mgrott              |[0m 	 - layout   :  T050103
[... cut ...]
[36mgrott              |[0m 		 07440a461291141e435345323934363038430010f1110001000000000000000007350a4612911
[36mgrott              |[0m 		 399079e09e21293138d00d20201002d005907e8000300070007000d000503e803e803e803e803
[36mgrott              |[0m 		 e803e803e803e847726f7761747420496e7665727465726168626131383138000000040131000
[36mgrott              |[0m 		 0000100000000000000000001139c00c80000000000000000000000000000c9aa
[36mgrott              |[0m 	 - Grott data ack record or data record not defined no processing done

anoppe, 2024-03-07:
hmm, not very useful... :(

johanmeijer, 2024-03-11:
Is this also happening with the latest multiple architecture docker containers (ledidobe/grott:beta)?

marcel-hi, 2024-03-11:
I was using version 2.8.2. Now I have installed the beta version. According to the logs it is version 2.8.3. I will see if that resolves the problem.

marcel-hi, 2024-03-13:
@johanmeijer  I've been running the beta version for 2 days now. Yesterday and today I had to restart the Docker before it worked after a night when the Growatt inverter was down. So it seems like the beta version isn't the solution.

johanmeijer, 2024-03-13:
@marcel-hi. Ok. This is a difficult problem. 

Lets try determineren what can be the problem. First how do you config look like.

What server do you have (Linux, windows, nas,rpi).
Do you run grott or grottserver?
What is your inverter type and datalogger (shinelan, wifi or shinlink).
Does your inverter goes offline at night or is always on.

Can you sent the last log entries. It is a pity we do not have time stamps can you see if it stop in the evening or when it restarts in the morning (or somewhere in the middle)?

johanmeijer, 2024-03-13:
I think docker logs --timestamps will show timestamps with the log lines.

marcel-hi, 2024-03-15:
@johanmeijer thanks, I will try to answer the questions and with the tip of --timestamps I have also timestamps
complete logs: [grott_log20240315 - anoniem.txt](https://github.com/johanmeijer/grott/files/14614742/grott_log20240315.-.anoniem.txt)

**What server do you have (Linux, windows, nas,rpi).**
Server: Raspberry pi 4, with Raspbian GNU/Linux 11 (bullseye)

**Do you run grott or grottserver?**
I am not sure. I think grott, because I use this with a docker-compose file, with these settings :
    image: ledidobe/grott:beta
    container_name: grott
    restart: unless-stopped
    ports:
      - "5279:5279"
    #volumes:
     #- /opt/grott/grottstub.ini:/app/grott.ini
     #- /opt/grott/grottstub.py:/app/grottext.py
    environment:
      - gnomqtt=False
      - gmqttip=192.168.1.10
      - gmqttauth=True
      - gmqttuser=mqtt
      - gmqttpassword=*secret*
      - gpvoutput=False
    networks:
      - network

 **What is your inverter type and datalogger (shinelan, wifi or shinlink).**
inverter: Growatt2500-MTL-S
datalogger: ShineWIFI-S

**Does your inverter goes offline at night or is always on.**
The inverter goes offline at night (or in case there is no sunshine).

**logs**
In the attachment, I have added the logs of the last days. I have changed some potential personal info (password, serial number). And it is now with the timestamps, thanks for the tip.

For example on 2024-03-12T17:33:46.236891975Z there is a log message with information that goes to correctly to mqtt

After 18 minutes, there is a connection error, likely due to the inverter is shutting down. I believe this is normal behaviour:
[36mgrott              |[0m 2024-03-12T17:51:26.405251656Z 	 - Grott connection error
[36mgrott              |[0m 2024-03-12T17:51:26.405389525Z 	 - peer has disconnected

the next day there is a new connection, so that looks okay:
[36mgrott              |[0m 2024-03-13T06:08:23.160212035Z 	 - ('172.28.5.254', 35042) has connected

However, until I restarted grott no usefull information was send to my MQTT server. 

when I restart the grot docker:

[36mgrott              |[0m 2024-03-13T12:22:09.059408025Z Grott Growatt logging monitor : 2.8.3
...

It seems like the standard messages that appear when I (re) start the docker. Just a few seconds later, there is a response containing useful information that is also send to my mqtt server: 

[36mgrott              |[0m 2024-03-13T12:22:32.752087828Z 	 - Grott values retrieved:

It seems that something occurs when the inverter shutsdown and restarts in the morning, causing no usefull information to be processed. However when I restart de grott docker this issue is for that day resolved. there is the same pattern on other days.

I hope you can find something in the logs that can help determine the problem?

johanmeijer, 2024-04-04:
Sorry for the late reaction..... I am still there. I am working on a new grott/grottserver version and I found some network issue's but I am not sure if this will also helps you. 

I am seeing some interesting things with my shinelink where the datalogger connection is always active (and the inverter is started in the morning). This does not apply to my shinewifi-s where the datalogger (and inverter) is powered off during the ninight and activated in the morning again.  

Is the 2500 MTL-S always on or does it also power down totally in the evening?

anoppe, 2024-04-05:
For what it's worth: 
My MIN 2500TL-X with the ShineWifi-X dongle powers down every day and comes back alive in the morning again when there is enough power generated by the panels. I don't face the issues @marcel-hi does, by the way. I'm also running Grott using docker.