johanmeijer/grott #475: Grott not sending messages to MQTT while Growatt module is connected?

Rolf-Smit, 2023-11-03:
I have a docker-compose setup with a combination of services:

- Home Assistant
- MQTT (Mosquitto)
- Grott

Home Assistant runs fine, as well as MQTT (Home Assistant can connect to it), however I'm not able to get any messages send from Grott to MQTT.

Notice how in the logs my Shinelink/-X module is connecting to Grott. In fact on the server.growatt.com website I see data being reported, this must mean the proxy is working?

However I see nothing connecting to my MQTT instance. Any guidance on how to debug Grott to what is happening exactly. I don't see Grott attempting to decode any messages (I do see the Shinelink disconnect sometimes), if I enter the Grott IP address in my browser I do see activity.

[image]

```
grott            | Grott settings:
grott            | 
grott            | _Generic:
grott            | 	version:             	 2.8.2
grott            | 	verbose:             	 True
grott            | 	trace:               	 False
grott            | 	config file:         	 grott.ini
grott            | 	minrecl:             	 100
grott            | 	decrypt:             	 True
grott            | 	compat:              	 False
grott            | 	invtype:             	 default
grott            | 	invtypemap:          	 {}
grott            | 	include_all:         	 False
grott            | 	blockcmd:            	 False
grott            | 	noipf:               	 False
grott            | 	time:                	 auto
grott            | 	sendbuf:             	 True
grott            | 	timezone:            	 local
grott            | 	valueoffset:         	 6
grott            | 	offset:              	 6
grott            | 	inverterid:          	 automatic
grott            | 	mode:                	 proxy
grott            | 	grottip              	 default
grott            | 	grottport            	 5279
grott            | _MQTT:
grott            | 	nomqtt               	 False
grott            | 	mqttip:              	 192.168.0.103
grott            | 	mqttport:            	 1883
grott            | 	mqtttopic:           	 energy/growatt
grott            | 	mqttmtopic:          	 False
grott            | 	mqttmtopicname:      	 energy/meter
grott            | 	mqttinverterintopic: 	 False
grott            | 	mqtttretain:         	 False
grott            | 	mqtttauth:           	 True
grott            | 	mqttuser:            	 homeassistant
grott            | 	mqttpsw:             	 **secret**
grott            | _Growatt server:
grott            | 	growattip:           	 47.91.67.66
grott            | 	growattport:         	 5279
grott            | _PVOutput:
grott            | 	pvoutput:            	 False
grott            | 	pvdisv1:             	 False
grott            | 	pvtemp:              	 False
grott            | 	pvurl:               	 https://pvoutput.org/service/r2/addstatus.jsp
grott            | 	pvapikey:            	 yourapikey
grott            | 	pvinverters:         	 1
grott            | 	pvsystemid:          	 systemid1
grott            | _Influxdb:
grott            | 	influx:             	 False
grott            | 	influx2:            	 False
grott            | 	database:           	 grottdb
grott            | 	ip:                 	 localhost
grott            | 	port:               	 8086
grott            | 	user:               	 grott
grott            | 	password:           	 **secret**
grott            | 	organization:       	 grottorg
grott            | 	bucket:             	 grottdb
grott            | 	token:              	 **secret**
grott            | _Extension:
grott            | 	extension:          	 False
grott            | 	extname:            	 grottext
grott            | 	extvar:             	 {'none': 'none'}
grott            | 
grott            | 
grott            | Grott proxy mode started
grott            | Hostname : f5bf64f158f3
grott            | IP :  172.18.0.3 , port :  5279 
grott            | 
grott            | 	 - ('192.168.0.197', 1995) has connected
grott            | 	 - Grott connection error
grott            | 	 - peer has disconnected
grott            | 	 - ('192.168.0.197', 1999) has connected
grott            | 	 - ('192.168.0.134', 59979) has connected
grott            | 
grott            | 	 - Growatt packet received:
grott            | 		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.18.0.3', 37256), raddr=('47.91.67.66', 5279)>
grott            | 	 - Grott - grottproxy - Invalid data record received, processing stopped for this record
```

Rolf-Smit, 2023-11-03:
Maybe this line can be changed to show a bit more information about what exception exactly occurred? The current "Grott connection error" is not really helpful and throws away all meaningful information.

https://github.com/johanmeijer/grott/blob/eafd8f11da54119b89159c5f0fb7a533f068fc6a/grottproxy.py#L137C21-L137C21

johanmeijer, 2023-12-04:
Hmm there is not more to see about this error I am afraid. 

Invalid data record means that the record is corrupted. The Grott integrity check of the record detected that the CRC code does not match the content of the record. 
The only thing that might be added is the content of the record.