johanmeijer/grott #478: Grott service fails on startup. Status=217/USER

thompy93, 2023-11-03:
Hi,

Not an experienced Linux user, so bear with me. 
Managed to set up GROTT on a Pi and it is working correctly as proxy. 

Following WIKI to set GROTT up as a linux service. 

There I encounter this error after starting GROTT. 

`
thompy93@raspberrypi:/etc/systemd/system $ sudo systemctl status grott
× grott.service - Grott Growatt monitor
     Loaded: loaded (/etc/systemd/system/grott.service; enabled; preset: enabled)
     Active: failed (Result: start-limit-hit) since Fri 2023-11-03 19:59:36 CET; 12min ago
   Duration: 1ms
   Main PID: 3694 (code=exited, status=217/USER)
        CPU: 0

Nov 03 19:59:36 raspberrypi systemd[1]: grott.service: Scheduled restart job, restart counter is at 5.
Nov 03 19:59:36 raspberrypi systemd[1]: Stopped grott.service - Grott Growatt monitor.
Nov 03 19:59:36 raspberrypi systemd[1]: grott.service: Start request repeated too quickly.
Nov 03 19:59:36 raspberrypi systemd[1]: grott.service: Failed with result 'start-limit-hit'.
Nov 03 19:59:36 raspberrypi systemd[1]: Failed to start grott.service - Grott Growatt monitor.
`

From what I understand this is related to user somehow. 

Default in grott.service is `User=Pi`

I dont have a Pi user, so I tried to change that to my rpi user without success. 
Also tried adding

`User=root
Groupt=root
`

No success. 

Owner of file is root. 
Rights on file are rwx rwx rwx.

flopp999, 2023-11-03:
Try user thompy93

thompy93, 2023-11-03:
That is my rpi user. Tried that. 

I also see in the grott.service that it is pointing to path 

```
WorkingDirectory=/home/pi/grott/
ExecStart=-/usr/bin/python3 -u /home/pi/grott/grott.py -v
#StandardInput=socket
StandardOutput=journal
```

WorkingDirectory and ExecStart should reflect user I installed Pi with?

flopp999, 2023-11-03:
Correct, change pi to the user you created during install of raspbian

thompy93, 2023-11-03:
Ok.

All changed now. 

Seems like a deamon reload was not enough. 
sudo service grott restart

And I get this

```

● grott.service - Grott Growatt monitor
     Loaded: loaded (/etc/systemd/system/grott.service; enabled; preset: enabled)
     Active: active (running) since Fri 2023-11-03 20:27:48 CET; 2s ago
```

Thank you for pointers to resolve this.

flopp999, 2023-11-03:
Great
Fun that you got it working

closed 2023-11-03 by thompy93