johanmeijer/grott #486: 3 different layouts from one pvserial

DB501, 2023-11-15:
Hi,
I just started using Grott, so it is possible that I have some mistake in my configuration. But I probably receive battery status in different packet than other data.

I was able to find battery charge in one of them. Following config seems to work, but strangely the time in the date is wrong:
```
        self.recorddict13 = {"T060136X": {
            "decrypt"           : {"value" :"True"},
            "datalogserial"     : {"value" :16, "length" : 10, "type" : "text"},
            "pvserial"          : {"value" :76, "length" : 10, "type" : "text"},
            # "date"              : {"value" :136, "divide" : 10}, 
            "batterySoc"           : {"value":1034 , "length" : 2, "type" : "num", "divide" : 1}
            }}
```
Logs:
 ```
        - Growatt original Data:
                 \x00\x2e\x00\x06\x03\x3f\x01\x04\x1f\x35\x2b\x42\x23\x3c\x20\x77\x3c\x25\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x3b\x27\x2b\x40\x37\x00\x3f\x5f\x47\x56\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x7f\x7b\x46\x70\x47\x74\x61
                 \x74\x74\x3b\x72\x69\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x76\x63\x51\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x67\xf3\x57\x31\x6f\x71\x61\x74\x66\x88\x62\xff\x77\x67\x74\x74\x47\x72
                 \x60\xdf\x61\x72\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
                 \x47\x2c\xed\x76\x26\x9d\x39\x47\x72\x6f\x77\x61\x74\x1b\x71\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x18\x57\x75
                 \x58\x46\x6f\x6e\x6c\x61\x74\x7c\x40\x6e\x91\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x60\x74\x74\x45\xb9\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x61\x74\x74
                 \x47\x72\x6f\x77\x61\x74\x77\xaf\x76\x0b\x77\x67\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x67\x72\x6f\x77\x61\x7c\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x7a\x66
                 \x77\x6e\x74\x74\x67\xa2\x6d\x9c\x61\x33\x74\x5a\x72\x6f\x77\x61\x54\xa4\x47
                 \x72\x6f\x77\x63\xc8\x74\x54\x8d\x9d\x77\x61\x74\x74\x47\x72\x6f\x77\x41\xa4
                 \x74\x47\x8d\xbe\x88\xad\x8b\xa1\x47\x72\x4f\xa7\x61\x74\x74\x97\x72\x69\x6b
                 \x98\x74\x8e\x47\x72\x6f\x71\x61\x74\x7c\xdc\x72\x6f\x77\x61\x74\x74\x0b\x2e
                 \x6f\x77\x61\x74\x74\x47\x61\x99\x77\x61\x74\x74\x47\x72\x78\x66\x61\x74\x74
                 \x41\x72\x6f\x5f\x1d\x74\x74\x47\x72\x6f\x77\x61\x74\x7d\x68\x72\x6f\x77\x61
                 \x74\x74\x4e\x49\x6f\x77\x61\x74\x74\x47\x7b\x50\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\xeb\x77\x61\x74\x74\x47\x7d\x67\x7e\x61\x74\x75\x60
                 \x72\x95\x64\xf0\x64\x94\x47\x72\x28\x23\x61\xae\x74\x26\x7a\x8f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6d\x77\x61\x74\x74\x47\x72\x67\x97
                 \x66\x14\x78\xd2\x7e\xe4\x73\x5b\x75\x72\x47\xa2\x6f\xc9\x60\x76\x70\x49\x72
                 \x2f\x77\x65\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x76\x0a\x73\x80\x74\x75
                 \x47\x72\x6d\xbc\x61\x74\x74\x47\x71\x87\x77\x61\x74\x74\x47\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x2c\xbb\x6f\x77\x61\x74\x74\x47\x6d\x90
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6e\x77\x62\x2f\x6a\x47
                 \x73\x6f\x74\xc1\xfe\x65\xb8\x63\x90\x66\x9e\x65\x8b\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x36\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x1d\x90\xba\x9e\xba\x8b
                 \x92\x8d\xba\x77\x7d\x73\xa4\x09\x52\x93\x6f\xb4\x7c\x74\x46\x72\x6f\x77\x61
                 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
                 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
                 \x72\x6f\x77\x61\x74\x74\x52\x6a\x6f\xf7\x61\x74\x74\x47\x72\x6f\x77\x61\x74
                 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
                 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
                 \x6f\x63\x9e
         - Grott automatic protocol detection
         - Grott data record length 839
         - layout   :  T060104XTL3
         - Record layout used :  T060104XTL3
         - Growatt data decrypted V2
         - Grott Growatt data decrypted
         - Growatt plain data:
                 002e0006033f010458474435424854304e4a00000000000000000000000000000000000000005
                 4504a3443474d3030370000000000000000000000000000000000000000170b0f010228030000
[... 223 lines cut ...]
```
If this behaviour is correct, I will have to create issue with muppet3000/homeassistant-grott (It can't handle update of only one entity of the whole device for one MQTT message).

Docker version 2.8.2
Environment:
```
            - gblockcmd=False
            - ginvtype=tl3
            # - gcompat=True

            - gnomqtt=False
              ...
            - gpvoutput=False
            ...
            - ginflux=False
            ...
            - gextension=false
            - gextname=grottext
            - 'gextvar = {"ip": "192.168.0.47", "port": "8000"}'
            - TZ=Europe/Prague         
```
Additional configs T060104Xtl3.json

Can you tell me if I have some mistake in config?
I will try to test it further and I would love to help in any meaningful way.

DB501, 2023-11-16:
Ok, I found the battery status in T060104. The following part of json helped:
```json
    "bat_Volt"          : {"value" :714, "length" : 4, "type" : "num", "divide" : 10},
    "edischrtoday"      : {"value" :874, "length" : 4, "type" : "num", "divide" : 10},
    "edischrtotal"      : {"value" :882, "length" : 4, "type" : "num", "divide" : 10},
    "echrtoday"         : {"value" :890, "length" : 4, "type" : "num", "divide" : 10},
    "echrtotal"         : {"value" :898, "length" : 4, "type" : "num", "divide" : 10},
    "elocalload_tod"    : {"value" :906, "length" : 4, "type" : "num", "divide" : 10},
    "elocalload_tot"    : {"value" :914, "length" : 4, "type" : "num", "divide" : 10},
    "batterySOC"        : {"value" :1010, "length" : 2, "type" : "num", "divide" : 1}
```
I have no idea where you take the names for the values.

johanmeijer, 2023-11-27:
The names are (are little bit reformatted from the Growatt Inverter Modbus RTU Protocol - II V1 - 24-English document.