johanmeijer/grott #490: Add support for Growatt MOD series

aupilot, 2023-11-23:
Hi Johan,

It would be really great if you can add support for MOD series [1]. This inverter is partially compatible with the "default" mode, but some important data is missing (e.g. Output power). Here is an example of MQTT output:

{"device": "SFHDCxxxxx", "time": "2023-11-23T13:10:40", "buffered": "no", "values": {"datalogserial": "KWK2Dxxxxx", "pvserial": "SFHDCKF04X", "pvstatus": 1, "pvpowerin": 7915, "pv1voltage": 2800, "pv1current": 17, "pv1watt": 4760, "pv2voltage": 1661, "pv2current": 19, "pv2watt": 3155, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 466031499, "totworktime": 917504, "pvenergytoday": 216666199, "pvenergytotal": 276238388, "epvtotal": 170, "epv1today": 1685, "epv1total": 739772, "epv2today": 103, "epv2total": 1050, "pvtemperature": 690, "pvipmtemperature": 0, "battery1soc": 0}}

Cheers,
Kir

[1] https://growatt.tech/product/growatt-mod-5000-tl3-xh-3-phase-energy-storage-inverter/

ruedigerO, 2023-11-26:
Hi Johan,

first: A great project! :-)
But I have the same requirement like Kir:
Inverter: MOD 7KTL3-XH
4 battery modules named: ARK 2.5H-A1
Smart meter and a ShineLan-X
I tried to the setting invtype=tl3 but without success.
There are a lot of values missing (all regarding battery) and others are wrong (e.g.  pvfrequentie:  20.4)

How can I help to fix this?

johanmeijer, 2023-11-27:
Can you try it with  invtype=min that should work also for mod type inverters

ruedigerO, 2023-11-27:
Unfortunately not. I'd tried already. Also invtype=tl3 does not work. 
With invtype=default I get some data but e.g. all battery data is missing.

	 - Grott automatic protocol detection
	 - Grott data record length 278
	 - layout   :  T060120
	 - Record layout used :  T060120
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 00b10006010e01205847443643483632344400000000000000000000000000000000000000004
		 d454b304346463043540000000000000000000000000000000000000000170b1b15212800c800
		 000086000008f7000008f8000008fb0000000d0000001d0000000d00000823000018870000081
		 000000bf100001bd100000c27fffff801fffff85ffffff745000002a200000394000002900000
		 28ba000033ebffffe84500000317000001f400000f8800000f8c00000f8b000010bd000001880
		 000003700000f05000017cb0000124600000f3c00000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000074ad
	 - Growatt new layout processing
		 - decrypt       :  True
		 - offset        :  6
		 - record layout :  T060120

	 - Grott data record date/time processing started
	 -......

-> the values for datalogserial and pvserial correct (voltage_l? and Current_l?). 
From my point of view the data record seems a bit small?!

ruedigerO, 2023-11-29:
Hi Johan,
does the output help? Or do you need more information?

johanmeijer, 2023-12-04:
Can you send me the encrypted data record (with the /x data). That makes it more easy for me to test.

ruedigerO, 2023-12-05:
After updating of the grott executables (to v Beta-2.8.x) in my grottserver container and setting invtype=min it looks much better.
I'll check tomorrow if all data is available

aupilot, 2023-12-06:
I tried the Beta-2.8.x. The "mini" mode does not seem to produce any data and the "default" mode works the same as the release 2.7.8

Rido, 2023-12-29:
I don't want to hijack this issue, but I am also having problems with my MOD inverter, so thought will use this issue instead of creating a new one. I'm not getting all the data with my MOD 8000TL3-X inverter.

After reading this issue I added the `invtype=min` to my Docker environment. The logging says it's using the layout `T065029MIN`. 

But the value `pvpowerout` for example stays empty in my MQTT output.

Encrypted data:
```\x5a\x50\x46\x59\x75\x4b\x4f\x47\x58\x4e\x40\x75\x48\x5f\x47\x58\xbe
\x75\x44\x47\x6c\x6f\x72\x26\x39\x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5c
\x77\x5c\x59\x77\x77\x74\x70\x11\x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61
\x7b\x77\x69\x33\x20\x14\x25\x5f\x47\x51\x44\x74\x52\x72\x68\x40\x4f\x40\x5a
\x61\x75\x44\x47\x78\x6f\x76\x51\x74\x7f\x47\x72\x6f\x7a\x61\x76\x45\x7f\x72
\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7e
\x41\x5d\x77\x69\x74\x6a\x0c\x24\x24\x47\x22\x3f\x30\x76\x33\x27\x77\x61\x74
\x77\x61\x74\x67\xcf\x72\x6b\x77\x60\x41\x74\x42\x72\x6e\x46\x61\x72\x74\x45
\x47\x45\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x33\x3c\x3c\x02\x31\x22\x4e\x51
\x24\x21\x2a\x44\x37\x0c\x36\x5e\x36\x29\x74\x74\x47\x72\x6f\x77\x61\x74\x74
\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
\x00\xcb\x00\x06\x00\xed\x50\x29\x0c\x25\x24\x46\x22\x3f\x30\x76\x33\x27\x77
```

MQTT output:
```
{"device": "xxxxxx", "time": "2023-12-29T09:42:01", "buffered": "no", "values": {"pvserial": "xxxxx", "pvstatus": 1, "pvpowerin": 703, "pv1voltage": 1415, "pv1current": 1, "pv1watt": 141, "pv2voltage": 1407, "pv2current": 1, "pv2watt": 140, "pv3voltage": 0, "pv3current": 0, "pv3watt": 0, "pv4voltage": 0, "pv4current": 0, "pv4watt": 0, "pvpowerout": 0, "pvfrequentie": 0, "pvgridvoltage": 0, "pvgridcurrent": 0, "pvgridpower": 0, "pvgridvoltage2": 0, "pvgridcurrent2": 0, "pvgridpower2": 0, "pvgridvoltage3": 0, "pvgridcurrent3": 0, "pvgridpower3": 45421450, "vacrs": 2281, "vacst": 4, "vactr": 0, "ptousertotal": 59771108, "ptogridtotal": 262144, "ptoloadtotal": 59640044, "totworktime": 327680, "pvenergytoday": 74846072, "pvenergytotal": 258543461, "epvtotal ": 0, "epv1today ": 2197, "epv1total": 2362294, "epv2today": 0, "epv2total": 496, "epv3today": 0, "epv3total": 1387, "etousertoday": 0, "etousertotal": 0, "etogridtoday": 0, "etogridtotal": 0, "eloadtoday": 0, "eloadtotal": 0, "deratingmode": 0, "iso": 0, "dcir": 0, "dcis": 0, "dcit": 0, "gfci": 1883, "pvtemperature": 516, "pvipmtemperature": 233, "temp3": 223, "temp4": 0, "temp5": 0, "pbusvoltage": 3049, "nbusvoltage": 3052, "ipf": 20000, "realoppercent": 0, "opfullwatt": 80000, "standbyflag": 0, "faultcode": 0, "warningcode": 0, "systemfaultword0": 0, "systemfaultword1": 0, "systemfaultword2": 0, "systemfaultword3": 0, "systemfaultword4": 0, "systemfaultword5": 0, "systemfaultword6": 0, "systemfaultword7": 60, "invstartdelaytime": 0, "bdconoffstate": 0, "drycontactstate": 0, "edischrtoday": 0, "edischrtotal": 0, "echrtoday": 0, "echrtotal": 0, "eacchrtoday": 0, "eacchrtotal": 0, "priority": 0, "epsfac": 0, "epsvac1": 0, "epsiac1": 0, "epspac1": 0, "epsvac2": 0, "epsiac2": 0, "epspac2": 0, "epsvac3": 0, "epsiac3": 0, "epspac3": 0, "epspac": 0, "loadpercent": 0, "pf": 0, "dcv": 0, "bdc1_sysstatemode": 0, "bdc1_faultcode": 0, "bdc1_warncode": 0, "bdc1_vbat": 0, "bdc1_ibat": 0, "bdc1_soc": 0, "bdc1_vbus1": 0, "bdc1_vbus2": 0, "bdc1_ibb": 0, "bdc1_illc": 0, "bdc1_tempa": 0, "bdc1_tempb": 0, "bdc1_pdischr": 0, "bdc1_pchr": 0, "bdc1_edischrtotal": 0, "bdc1_echrtotal": 0, "bdc1_flag": 800, "bdc2_sysstatemode": 0, "bdc2_faultcode": 0, "bdc2_warncode": 23, "bdc2_vbat": 12, "bdc2_ibat": 28, "bdc2_soc": 23, "bdc2_vbus1": 12, "bdc2_vbus2": 29, "bdc2_ibb": 0, "bdc2_illc": 0, "bdc2_tempa": 0, "bdc2_tempb": 25565, "bdc2_pdischr": 0, "bdc2_pchr": 0, "bdc2_edischrtotal": 65536, "bdc2_echrtotal": 0, "bdc2_flag": 0, "bms_status": 0, "bms_error": 0, "bms_warninfo": 0, "bms_soc": 0, "bms_batteryvolt": 0, "bms_batterycurr": 0, "bms_batterytemp": 0, "bms_maxcurr": 0, "bms_deltavolt": 0, "bms_cyclecnt": 0, "bms_soh": 0, "bms_constantvolt": 0, "bms_bms_info": 0, "bms_packinfo": 0, "bms_usingcap": 0, "bms_fw": 0, "bms_mcuversion": 0, "bms_commtype": 0}}
```

johanmeijer, 2023-12-29:
The record you included is not a data record. So I can not detect what the problem is. 

Have you tried it with invtype=min? That layout also seems to fit the MOD types inverters the best at this moment.

Rido, 2023-12-31:
> The record you included is not a data record. So I can not detect what the problem is.
> 
> Have you tried it with invtype=min? That layout also seems to fit the MOD types inverters the best at this moment.

That's strange, I copied all data from the Synology Docker output.  I have now tried to export a full output of a data snapshot. Synology gives me a HTML file. You can view the output in your browser.

And I made a overview in Home Assistant with all the parsed data from the attached data snapshot. Some data entries are missing like the actual output and the generated energy seems a bit weird. When I check the data on server.growatt.com it shows a total generation of `232.1 kWh`.

[image]

```html
<!DOCTYPE html>
<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
<style>body{font-family:"Lucida Console",Consola,Monaco,Monospace,"Courier New",Courier,Serif;}h2{text-align:center;}table{width:95%;border-collapse:collapse;}table,th,td{border:1px solid black;}td{white-space:nowrap;text-align:left;}</style>
</head>
<body>
<h2>grott</h2>
<table>
<tr><th>date</th><th>stream</th><th>content</th></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Grott&nbsp;data&nbsp;ack&nbsp;record&nbsp;or&nbsp;data&nbsp;record&nbsp;not&nbsp;defined&nbsp;no&nbsp;processing&nbsp;done
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;2d33312031333a32323a35309762
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;60016000456312e30001b000136001c000130001e0005474d542b38001f0013323032332d3132
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;30000a000130000b0000000d0002313800140008475453573030303000150007372e342e302e3
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;20008001e4b564b30434b44314148000000000000000000000000000000000000000000090001
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;31000000000000000000000000000000000000000013880004000135000500013100060002333
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;b564b30434b44314148000000000000000000000000000000000000000052484845434d393033
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;0009000600ed50294b574b31434b4431414800000000000000000000000000000000000000004
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Growatt&nbsp;plain&nbsp;data:
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Grott&nbsp;Growatt&nbsp;data&nbsp;decrypted
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Growatt&nbsp;data&nbsp;decrypted&nbsp;V2
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Record&nbsp;layout&nbsp;used&nbsp;:&nbsp;&nbsp;T065029MIN
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;no&nbsp;matching&nbsp;record&nbsp;layout&nbsp;found,&nbsp;try&nbsp;generic
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;layout&nbsp;&nbsp;&nbsp;:&nbsp;&nbsp;T065029MIN
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Grott&nbsp;data&nbsp;record&nbsp;length&nbsp;245
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>	&nbsp;-&nbsp;Grott&nbsp;automatic&nbsp;protocol&nbsp;detection
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x5a\x50\x46\x59\x74\x43\x4f\x46\x52\x4e\x46\x75\x48\x5a\x47\xf6\x16
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x75\x44\x47\x6c\x6f\x72\x26\x39\x20\x6c\x4a\x6f\x68\x61\x67\x46\x77\x40\x5c
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x77\x5c\x59\x77\x77\x74\x70\x11\x43\x41\x47\x61\x6f\x74\x46\x44\x6f\x6b\x61
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x7b\x77\x69\x33\x20\x14\x25\x5f\x47\x51\x44\x74\x52\x72\x68\x40\x4f\x40\x5a
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x61\x75\x44\x47\x78\x6f\x76\x51\x74\x7f\x47\x72\x6f\x7a\x61\x76\x45\x7f\x72
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x7e
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x41\x5d\x77\x69\x74\x6a\x0c\x24\x24\x47\x22\x3f\x30\x76\x33\x27\x77\x61\x74
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x77\x61\x74\x67\xcf\x72\x6b\x77\x60\x41\x74\x42\x72\x6e\x46\x61\x72\x74\x45
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:52</td><td>stdout</td><td>		&nbsp;\x47\x45\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
</td></tr>
[... 855 lines cut ...]
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>	&nbsp;-&nbsp;Data&nbsp;less&nbsp;then&nbsp;minimum&nbsp;record&nbsp;length,&nbsp;data&nbsp;not&nbsp;processed
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>		&nbsp;&nbsp;&lt;socket.socket&nbsp;fd=5,&nbsp;family=AddressFamily.AF_INET,&nbsp;type=SocketKind.SOCK_STREAM,&nbsp;proto=0,&nbsp;laddr=('172.17.0.6',&nbsp;5279),&nbsp;raddr=('172.17.0.1',&nbsp;60794)&gt;
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>	&nbsp;-&nbsp;Growatt&nbsp;packet&nbsp;received:
</td></tr>
[... cut ...] p;5279)&gt;
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>	&nbsp;-&nbsp;Growatt&nbsp;packet&nbsp;received:
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:37</td><td>stdout</td><td>
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:36</td><td>stdout</td><td>	&nbsp;-&nbsp;('172.17.0.1',&nbsp;60794)&nbsp;has&nbsp;connected
</td></tr>
<tr><td>2023/12/31&nbsp;13:22:36</td><td>stdout</td><td>	&nbsp;-&nbsp;('172.17.0.1',&nbsp;60474)&nbsp;has&nbsp;disconnected
</td></tr>
</table></body></html>
```

techmantel, 2024-03-29:
@Rido Check my comment here https://github.com/johanmeijer/grott/issues/81#issuecomment-2027467918
I have the same inverter and `ginvtype=tl3` works!