johanmeijer/grott #492: How to confirm MQTT, Influx and PVOutput is working?

paytah232, 2023-12-02:
Hi @johanmeijer,

I've got this running in docker on a Synology NAS, using home assistant as well. I'm using grott.ini for the docker, and as far as I can tell, grott is still relaying the data to the internet (Growatt Server), as I can access it from the ShinePower app, but I can't seem to see any information coming through on MQTT, Influx or PVOutput.

I'm wondering how I can test to confirm, as I expected the data should be easily available but I can't seem to get any response.

The setup:
Synology NAS, running grott in docker with grott.ini
Synology NAS is available locally on 192.168.0.2
ShineLAN Box is available locally on 129.168.0.169, which has sever IP 192.168.0.2.
Grott Docker uses 192.168.0.2:5279 to docker:5279

ShineLan Box:
[image]

PVOutput:
[image]

MQTT:
[image]

INFLUX:
[image]

I get the following from grott logs, suggesting my config is at least working:
[grott.csv](https://github.com/johanmeijer/grott/files/13533410/grott.csv)

Any help would be appreciated, thank you.

johanmeijer, 2023-12-04:
Are you sure your data is being routed via Grott? Have you changed the adres in the inverter / datalogger. 

From the log you sent I can not see grott is working ok. This seems not to be a real grott log to me.

paytah232, 2023-12-12:
I can see that the reported server IP matches the NAS I am running off, and no longer matches growatt.server.com. However, I am not sure if that is the correct method of setup. Can you confirm full steps to ensure datalogger is pointing the right server (i.e the local server)?

johanmeijer, 2023-12-17:
Is the information the only information you see in the Grott log on your NAS? 

Grott runs in docker at your NAS? How do you start Grott? You need to map the NAS port to the Grott Container.

paytah232, 2023-12-18:
I recently did a server wipe, but yes, it was in a container, with all the ports set up as far as I know. I'll get it set back up again shortly and show you.

paytah232, 2023-12-23:
@johanmeijer
I've got it set up back again. I can confirm that I was getting no data via the shine app when the container was down, and now that it's up, it has come back, so Grott is working as expected.

What I want to test now is how I can confirm the MQTT Broker is working, as I still get nothing on the topic channel.
[image]

According to the docker logs, as below, MQTT seems to be active - yet I still receive nothing, with no errors to trace:
[grott.csv](https://github.com/johanmeijer/grott/files/13758829/grott.csv)

johanmeijer, 2023-12-24:
Hmm in your logfile I still see no data being processed by grott. I can see there is some communication flow (probably pings). It can be a timing thing (just at the moment of capturing no data). 

I expect some messages that there is data being written to MQTT in the grott log.

paytah232, 2023-12-27:
@johanmeijer Happy to try some things you suggest to try and get the MQTT working. What are your suggestions?

paytah232, 2024-02-27:
@johanmeijer I have looked into this again, but still unsure what could be the cause. Are you able to help me find the cause of the issue at all please?

johanmeijer, 2024-02-28:
I need more log information.  Is there information in the log which show the values? 
Do see messages that data is (or not)send to MQTT or PVOutput?

paytah232, 2024-02-28:
I can get that for you. Is there a way to increase the log verbosity so we get more information?