johanmeijer/grott #495: Intermittent dropouts in influx recordings

raadjek, 2023-12-12:
Hi,
I'm using Grott for few months now in proxy mode. The values are correctly reported to server.growatt.com. I'm using ShinePhone app and growatt web interface for checking graphs and values.

I have enabled influx + grafana to create my own graphs (similar to growatt official ones), but I see occasional drop outs in data. Portion(s) of graph is missing - see screenshots below. The data is missing from the influx DB as well, so it's not Grafana issue. 
I'm not sure where should I look - influx or MQTT or Grott. Since energy dashboard shows the data, I guess MQTT works fine? Can you please point me in right direction what to check to see where the data is lost/rejected? I didn't find any rejections or errors in Influx log (settings>add-ons>influx>log).

I'm not sure if that's related, but Grott shows similar code very often:

	 - Growatt packet received:
		  <socket.socket fd=5, family=2, type=1, proto=0, laddr=('172.30.33.8', 54026), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x03\x8d\x00\x06\x02\x41\x01\x36\x1f\x35\x2b\x42\x23\x3c\x47\x73\x33\x25\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3b\x39\x2b\x46\x36\x0b\x3a\x5f\x47\x34\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x76\x78\x7f\x55\x5c\x77\x75\x18
		 \x6c\x0d\xd3\x72\x6a\x77\x63\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x53\x88\x6f\x59\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x27\x92\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x5d\x32\x90\x88
		 \x9e\xf1\x74\x47\x61\xe6\x67\x0e\x64\x0a\x57\x16\x6f\x7a\x61\x7d\x74\x4b\x8d
		 \x90\x8b\x79\x75\xd9\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x0d\xd2\x08\x7e\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x4d\x21\x77\x61\x74\x77\x47\x72\xe5
		 \x8e\x61\x74\x74\x47\x72\x6f\xf6\x43\x74\x74\xb8\x34\x6f\x77\x9e\x32\x7e\x71
		 \x72\x6f\x77\x5f\x74\x78\x47\x12\x6f\x77\x61\x5f\x74\x47\x72\x5e\x76\x08\x75
		 \x6c\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x27\x6d
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060136XSPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060136XSPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 038d0006024101365847443542483334414a00000000000000000000000000000000000000005
		 44e4a32424c483030550000000000000000000000000000000000000000170c0b122e18027918
		 7994000500020000000000000000000000000000000014fa002e0000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000060e00000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000000000000000001a40ffffff8500001389106f107e1064000d0009000cfffffc1801
		 ad000000000000000000000000000000000000000079957a11000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000003f4e0000000300008af900000000000
		 081220000ff460000ff460a360000003e000c00600000002b0000003101690118000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 000000000005502
	 - Grott data ack record or data record not defined no processing done

I'm using HAOS on dedicated RP. Grott (beta 2.8) in HACS, Mosquito 6.4.0, influx 4.8.0. I tried beta as well as stable version of Grott, no difference.
Grott config: 

	ata_path: /config/grott
	gmode: proxy
	grott_mqtt: true
	ginvtype: sph
	ginvtypemap: ""
	retain: true
	verbose: true
	ha_plugin: true
	mqtt: {}
	gifdbname: grottdb
	gifip: 192.168.1.24
	gifpassword: ****
	gifport: 8086
	gifuser: grottdbuser
	ginflux: true
	ginverterid: TNJ2BLH00U
	gmqttip: 192.168.1.24
	gmqttpassword: *****
	gmqttuser: mqttuser
	gmqttauth: true

Can you please point me in right direction?
I'm attaching few screenshots.
Grafana shows no data in last 12 hours:
[image]
Growatt web shows the data OK:
[image]
Energy dashboard also shows data:
[image]
Grafana in last 24 hours shows the data stopped around 21:00 (~13 hours ago):
[image]
Short about 1 hour dropout:
[image]

Thank you

johanmeijer, 2023-12-17:
The record you included (T060136XSPH) is not a normal data record and Grott will not proces this record. 
The data record should be T060104XSPH records in your case. Do you see this type of records in the Grott log in the period of the missing values?  

Do you see other Grott messages (e.g. errors with writing to MQTT or Influxdb)  at that time?

raadjek, 2023-12-18:
It seems the Grott specific log (Settings>System>Log>Grott) has short time span (not sure if somewhere on filesystem I could find full log), but anyway - no, I don't see T060104XSPH record during the period of missing values.

Mosquito gets its data every 5 mins, so that seems to be OK (TNJ2BLH00U is serial number of my inverter):

	2023-12-18 09:47:46: New connection from 172.30.33.7:33771 on port 1883.
	2023-12-18 09:47:46: New client connected from 172.30.33.7:33771 as TNJ2BLH00U (p2, c1, k60, u'addons').
	2023-12-18 09:47:46: Client TNJ2BLH00U disconnected.
	2023-12-18 09:48:54: New connection from 172.30.32.2:54758 on port 1883.
	2023-12-18 09:48:54: Client <unknown> closed its connection.
	2023-12-18 09:50:54: New connection from 172.30.32.2:41294 on port 1883.
	2023-12-18 09:50:54: Client <unknown> closed its connection.
	2023-12-18 09:52:46: New connection from 172.30.33.7:39409 on port 1883.
	2023-12-18 09:52:46: New client connected from 172.30.33.7:39409 as TNJ2BLH00U (p2, c1, k60, u'addons').
	2023-12-18 09:52:47: Client TNJ2BLH00U disconnected.

Influx log shows every 10sec or so, the same message:

	time="2023-12-18T09:52:31+01:00" level=info msg="Response: OK" component=server method=GET remote_addr="127.0.0.1:48288" response_time="116.72µs" status=200

johanmeijer, 2023-12-24:
The mosquitto dat log looks ok indeed. 

Influx I do not understand. Do you have other things logging to influx as well? It is really hard to determine for me what is happening if there is no Grott log available.

raadjek, 2024-01-02:
How can I switch on (more extensive) Grott logging - e.g. to a file? Or where can I find grott log on filesystem? Maybe it is there and I didn't find it.... I was only able to find the Grott log in UI, not on the filesystem.

johanmeijer, 2024-01-06:
I think you already enabled to most extensive logging (-v parm during start up). You can log to a file by specifing filename ar grott start. EG: 

python3 grott.py -v -c grott.ini -o log.txt

BaumSchorle, 2024-03-26:
@johanmeijer  I noticed something similar in my setup. I have grott running without issues for a longer period, but added influx the last couple of days. Yesterday evening I had the first incident where no data was written to influxdb but MQTT was still fine, today I had the second incident. After restarting grott container everything is back to normal.

I dug a bit in the leg and found the following:
- My last entry in influxdb was from today 13:36 (which can be found in the log)
- The next record grott processed was apparently suddenly from today 06:31

I can't explain where this data was coming from and why mqtt is still receiving correct data, while influxdb apparently is not.

Maybe you can find out why this could be happening. If I can be of any assistance, let me please know.

`

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.2', 5279), raddr=('172.17.0.1', 37984)>
	 - Data less then minimum record length, data not processed

	 - Growatt packet received:
		  <socket.socket fd=5, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.2', 51106), raddr=('47.91.67.66', 5279)>
	 - Growatt original Data:
		 \x03\x4c\x00\x06\x02\x41\x01\x36\x1f\x35\x2b\x41\x22\x3f\x41\x74\x34\x5c\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x3b\x27\x2b\x41\x37\x0c\x22\x5f\x3c\x38\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x79\x77\x6e\x4a\x51\x43\x75\x18
		 \x6c\x0d\xd3\x72\x6a\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x49\x61\x6f\x24\x70\x0c\x74\x14\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x75\x40\x3a\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\xba\xc6\x90\x88
		 \x9b\x65\x74\x47\x61\xe7\x78\xdc\x7b\xa5\x48\xca\x6f\x2c\x61\x2f\x74\x1c\x8d
		 \x90\x7c\xbd\x76\x7d\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x0d\xd2\x08\x7e\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74
		 \x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61
		 \x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f
		 \x5d\x61\x74\x74\x47\x72\x6f\x77\x68\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x1c
		 \x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74
		 \x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77
		 \x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\x72
		 \x6f\x77\x61\x74\x74\x47\x72\x6f\x77\x61\x74\x74\x47\xeb\xf1
	 - Grott automatic protocol detection
	 - Grott data record length 585
	 - layout   :  T060136XSPH
	 - no matching record layout found, try generic
	 - Record layout used :  T060136XSPH
	 - Growatt data decrypted V2
	 - Grott Growatt data decrypted
	 - Growatt plain data:
		 034c00060241013658474436434b3533463300000000000000000000000000000000000000005
		 4504a35434b50304b59000000000000000000000000000000000000000018031a0d232c027918
		 799400050000000000000000000000000000000000000e1300531178005300000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000107480000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000fdb4fffffa11000013880fbd0fd10fb8005b005b005bffff0bdc02
		 09000000000000000000000000000000000000000079957a11000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
		 00000000000000000000000000000000000000000000000000000000000000000000000000000
[... 898 lines cut ...]
			 "rev_act_energy": 11353}}
	 - Grott MQTT topic used : energy/growatt
	 - MQTT message message sent
	 - Grott Send data to PVOutput disabled 
	 - Grott InfluxDB publihing started
	 - Grott original time :  2024-03-26T06:31:34 adjusted UTC time for influx :  2024-03-26T05:31:34
	 - Grott influxdb jsonmsg: 
			 [{'measurement': 'XGD6CK53F3', 'time': '2024-03-26T05:31:34', 'fields':
			 {'datalogserial': 'XGD6CK53F3', 'pvserial': 'TPJ5CKP0KY', 'voltage_l1':
[... cut ...]
			 -40038, 'rev_act_power': -40038, 'app_power': 40783, 'react_power': -4516,
			 'powerfactor': -993, 'frequency': 499, 'L1-2_voltage': 4033, 'L2-3_voltage':
			 4041, 'L3-1_voltage': 4031, 'pos_act_energy': 39371, 'rev_act_energy':
			 11353}}]
	 - Grott write to influxdb v2
	 - Grott extension processing disabled 

	 - Growatt packet received:
		  <socket.socket fd=6, family=AddressFamily.AF_INET, type=SocketKind.SOCK_STREAM, proto=0, laddr=('172.17.0.2', 5279), raddr=('172.17.0.1', 37984)>
	 - Data less then minimum record length, data not processed`

BaumSchorle, 2024-03-27:
Same thing happened yesterday as I was writing my comment, everything was fine until 21:19 and then all of sudden the next record shows the time 14:15. I attached a bit of a longer log exctract.
[grott.log](https://github.com/johanmeijer/grott/files/14769064/grott.log)

a restart of grott helps to fix this situation for a couple of hours again. I have no idea where this sudden change in time comes from, but have not looked at the code in detail.

raadjek, 2024-03-27:
I have also noticed that the time stamps do not match. In the beginning it was off slightly, but since the grott log itself does not have timestamp. I thought it's a delay in the log.
Today morning (09:25am) when I refresh the grott log, last influx db message shows 17:36 of yesterday:
```
 - Growatt data decrypted V2
	 - Grott MQTT topic used : energy/growatt
	 - Grott influxdb jsonmsg: 
			 [{'measurement': 'TNJ2BLH00U', 'time': '2024-03-26T17:36:16', 'fields':
			 {'datalogserial': 'XGD5BH34AJ', 'pvserial': 'TNJ2BLH00U', 'pvstatus': 5,
			 'pvpowerin': 22500, 'pv1voltage': 6231, 'pv1current': 36, 'pv1watt': 22567,
			 'pv2voltage': 0, 'pv2current': 0, 'pv2watt': 0, 'pvpowerout': 4578,
			 'pvfrequentie': 4999, 'pvgridvoltage': 4150, 'pvgridcurrent': 8,
			 'pvgridpower': 6991, 'pvgridvoltage2': 4178, 'pvgridcurrent2': 9,
			 'pvgridpower2': 0, 'pvgridvoltage3': 4173, 'pvgridcurrent3': 9,
			 'pvgridpower3': 0, 'totworktime': 108628570, 'eactoday': 38,
			 'pvenergytoday': 38, 'eactotal': 95673, 'epvtotal': 98466, 'epv1today': 50,
			 'epv1total': 98466, 'epv2today': 0, 'epv2total': 0, 'pvtemperature': 464,
			 'pvipmtemperature': 483, 'pvboosttemp': 478, 'bat_dsp': 2666,
			 'eacharge_today': 6, 'eacharge_total': 14805, 'batterytype': 1,
			 'uwsysworkmode': 5, 'systemfaultword0': 0, 'systemfaultword1': 0,
			 'systemfaultword2': 0, 'systemfaultword3': 0, 'systemfaultword4': 0,
			 'systemfaultword5': 0, 'systemfaultword6': 0, 'systemfaultword7': 0,
			 'pdischarge1': 0, 'p1charge1': 16400, 'vbat': 2666, 'SOC': 40, 'pactouserr':
			 0, 'pactousertot': 0, 'pactogridr': 0, 'pactogridtot': 0, 'plocaloadr':
			 6100, 'plocaloadtot': 6100, 'spdspstatus': 5, 'spbusvolt': 7324,
			 'etouser_tod': 13, 'etouser_tot': 47982, 'etogrid_tod': 0, 'etogrid_tot':
			 31478, 'edischarge1_tod': 28, 'edischarge1_tot': 39205, 'eharge1_tod': 35,
			 'eharge1_tot': 42241, 'elocalload_tod': 55, 'elocalload_tot': 111219}}]
```
I'm running HAOS, so it's difficult (at don't know how) to run grott from command line with some additional parameters.

anoppe, 2024-03-27:
For what it's worth: I have had a bad experience with the date/time coming from the inverter and don't trust that anymore. I had the same behavior when the inverter suddenly published results for dates/times earlier that day.
My solution was to instruct Grott to use server date/time instead.

BaumSchorle, 2024-03-27:
@anoppe thx. for the hint, found the gtime configuration variable in the code and have set it to server now in my docker container. This should hopefully fix this issue!

raadjek, 2024-03-27:
Is there any GUI option to use server time (HAOS) instead? Or in /config/grott/grott.ini (which is empty for me)?

BaumSchorle, 2024-03-27:
I am not sure how HAOS implementation works, but it is a simple config variable (similar to the other ones in grott)

BaumSchorle, 2024-04-03:
Just wanted to confirm that using gtime environment variable in my docker environment solved the problems I had and the system is running stable for days now.

anoppe, 2024-04-03:
I still think that Grott should use server time as default to prevent all these kind of issues.

raadjek, 2024-04-03:
What would be the value in the grott.ini? Something like?
`gtime=true`
Or should something similar go to configuration.yaml to set it as environmental variable?
I'm bit lost here. Thanks for directing me to the rigt place.

BaumSchorle, 2024-04-03:
`gtime=server` will do the trick.

BaumSchorle, 2024-04-03:
If you add it to the .ini you might just need to use time=server! Not sure how it works with the ini, since I am using the docker environment variables.

raadjek, 2024-04-04:
Thanks! really the gtime=server caused error during initialization. Perhaps time=server is correct as you say.
I have found 'gtime' in Grott GUI config. Set it to "server" and it seems it works just fine!
Grott log shows Grott influxdb jsonmsg timestamp exactly 1 hour behind (I expect it is GMT? Perhaps need to set gtimezone variable too).

Anyway since yesterday afternoon when I updated the setup I see no dropout! Fingers crossed!

johanmeijer, 2024-04-04:
Yeah sorry for the confusion, in grott,ini it is time=server. I know internally grott it is called gtime. gtime = "server" is also the setting for (the docker container) environmental variable. 

Grott should change the time from  the datarecord to the GMT time and use that for influxdb (as standard is being used in time series databases). 
Your reporting application (like grafana) should again correct it for your local time then, 

If things are not going well (e.g the server / container time / time zone settings are not setted right) you can also specify
     timezone=Europe/Amsterdam (example) 
in the grott.ini (gtimezone= for evironmental)

If using a container it is recommended to start the container with a time zone (e.g. TZ=Europe/Amsterdam).

johanmeijer, 2024-04-04:
To add to the time problems, indeed it seems that same dataloggers are not very good in keeping the time actual. 

If the datalogger starts (e.g. in the morning) then Growatt is sending a time configuration record and the datalogger time is 
set. If the time is running of during the day you can indeed update the time by stop/starting Grott (During initialisation Growatt will send a update for the time again). 

With time=server. Grott ignores the time in the datarecord (from the datalogger) and uses the time of the server it is running. Using this has one disadvantage: Grott has to ignore the buffered/historical  records from the datalogger that might has been saved while having no connection with the growatt server while these are not of any value any more if using the actual time.