johanmeijer/grott #496: How can I achieve fall back in case of Grott goes down

LightMoon, 2023-12-13:
I wonder if there is a solution out there that can provide some sort of fallback mechanisation in case of HA or ( Grott ) goes down?
I still would like to be able to send the inverter data to the Growatt cloud in case of the HA outage.

rany2, 2023-12-13:
You could use the sniffer mode though admittedly it's not the most elegant method and your router might not be capable. Beyond that you're out of luck I'm afraid.

LightMoon, 2023-12-13:
Even the custom fw like [this](https://github.com/malosaa/Growatt_Shine-S_HASS_Mod) doesn't provide this feature?

rany2, 2023-12-13:
> Even the custom fw like [this](https://github.com/malosaa/Growatt_Shine-S_HASS_Mod) doesn't provide this feature?

That wouldn't even let you connect to Growatt's servers anymore, it would be entirely selfhosted by you.

LightMoon, 2023-12-13:
Can you point me to the right documentation so I figure out how I can set it up in Sniff mode?
I just installed the [grott-home-assistant-add-on](https://github.com/egguy/grott-home-assistant-add-on/tree/main) by @egguy.

rany2, 2023-12-14:
Have a look at https://github.com/johanmeijer/grott/wiki/Rerouting-Growatt-Wifi-TCPIP-data-via-your-Grott-Server#for-sniffer-mode-linux-only 

I wouldn't recommend it though, I think it's much of a hassle to get working. I would instead recommend running grott standalone but I'm not sure if it's possible with that extension you linked to

LightMoon, 2023-12-14:
Great thanks! I digested that in the Sniff mode, the role of packet forwarding is handed over to the Linux kernel instead of Grott handling it. My edge router is DD-WRT which is basically a Linux with capability of i[pv4 packet forwarding and access to iptable](https://forum.dd-wrt.com/phpBB2/viewtopic.php?t=312749&sid=79a8baa4842ee2a98f17fa0bd25f6780).
@johanmeijer 
I wonder if possible to get the dd-wrt router to forward Dataloger (Wifi Dongle) traffic to Grott server ( running on sniff mode)?

johanmeijer, 2023-12-17:
I have not tested the sniffer mode for a long time. It is not that simple to configure it and to have it running in parallel next to my other setup. 

To be honest I wanted to drop support for it in the 3.0 version of grott and focus more on proxy and grottserver (and integrate those). If you think it is still needed I and can leave the option in, but be-aware it works as long as it works. 

Grott in sniffer mode should run at a system where the IP traffic passes. In basic these means that Grott should run at the system where the IPTables rerouting has been configured (in your case the DD-WRT router). I do not think it is possible to run a python application like grott to run on a DD-WRT router.

bardahlm, 2024-01-07:
One way I suggested in #480 is to use as mirror-port on the switch. Another might be to set up some kind of load balancer (HAProxy) that sends traffic to Grott as long at it is up and running but sends to the Growatt servers if Grott is down. You might be able to run HAProxy on your firewall.

johanmeijer, 2024-01-07:
The best fall-back is provided by the growatt inverter / datalogger itself. In the rare occasion Grott stops (I am running it for your without any problems). The data is stored (buffered) in the datalogger and sent out later. 

I do not think that a complex other solution will add any value (but that is my personal opionion, being a little bit biased).