johanmeijer/grott #505: T06NNNNXTL3 not strict enough

ThomasTNO, 2024-01-08:
Currently, if no matching version is discovered Grott attempts to find a matching version by replacing four digits. In my case this results in incorrect behaviour. Values are inserted in wrong locations.

Layout `T065104XTL3` is falsely matched to `T06NNNNXTL3`. 

As I am not sure what the reasoning is for wildcard matching four digits, I am creating an issue here.

Using the attached json file works properly (based on https://github.com/johanmeijer/grott/blob/eafd8f11da54119b89159c5f0fb7a533f068fc6a/examples/Record%20Layout/t065104x.json)

[t065104xtl3.json](https://github.com/johanmeijer/grott/files/13866420/t065104xtl3.json)

johanmeijer, 2024-01-10:
The wildcard matching is not the problem. That is there so we have not to specify a layout record for every datalogger (you use a shinelink and that gives a dive number with 5x, I use a wifishine that always gives a device number of one. 

I agree with you that there is a mismatch between the specific and generic record. 
Is it possible that your Inverter is a min-type or mod type inverter? Then specifying ginvtype=min would help. It seems that there is a differince between a mod-tl3 and a tl3-x type inverter. I might have mixed up both. 

Can you please give the original growatt data from the Grott log so I can see what the best option would be. 

Be aware I am working on a grott version that automatically select the right layout for most of the inverters.

ThomasTNO, 2024-01-10:
This is the MIN 3000TL-XE
```
                 xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx000000000000000000000000000000000000000xx                                                                                                                              
                 xxxxxxxxxxxxxxxxxxx0000000000000000000000000000000000000000000000000000020bb8                                                                                                                              
                 0c340000000000000267000000000000026900000000000000000000000000000000000000000                                                                                                                              
                 000000000000000000000000000138808fc000000000000000000000000000000000000000000                                                                                                                              
                 0008fc00000000000000000000000000000000042fa1210000000d0000fafd000103b70000000                                                                                                                              
                 3000086bf0000000a00007cf80000000000000000000000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000000000fffa0000000000000000000000d200d20                                                                                                                              
                 000000000e8024300004e20000000000000000000000000000000000000000000000000000000                                                                                                                              
                 00003c0000000000000000000000000000000000000c350cb1000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000002710000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000000000000000000000000000000000000000000000000000000                                                                                                                              
                 00000000000000000000000000010001000000000000000000000000000001680bb8271000090                                                                                                                              
                 000000000000671
```

Stripped some stuff for privacy/security (serials) (if I interpret it correctly)

johanmeijer, 2024-01-10:
Oke. This is not a data record. What is I do not know while you also stripped the header whit the record information ;)

If you have a min inverter and run Grott beta (2.8.2 docker version) please use invtype=min.

ThomasTNO, 2024-01-11:
Thanks, I will try to have a look later today.

ThomasTNO, 2024-01-11:
Seems to be working, thanks!