johanmeijer/grott #506: Zero values after Inverter goes offline

alfredocdmiranda, 2024-01-10:
I have Growatt 1500TL-X and it goes off when the sun goes down everyday and also the Datalogger. I am wondering if there is a way in Home Assistant to detect when the Datalogger is offline and put the values to zero.

patrick131184, 2024-01-12:
i just wanted to know that also. maybe this can be added to grott. when iverter goes offline to send 0 watts

alfredocdmiranda, 2024-01-12:
> i just wanted to know that also. maybe this can be added to grott. when inverter goes offline to send 0 watts.

I thought the same but I thought that I could be missing something or didn't read the docs very well.

> homeassistant workaround could be with this trigger if not updated for 15 minutes

Thanks for the tip :)

PedroKTFC, 2024-01-12:
Doesn't it also report your battery charge level, import from the grid and home consumption even when the solar generation is zero? I'm interested in these values and, one day when it's more easy to update my inverter from Home Assistant, I'll look to have an automation that updates how much to charge my battery level overnight from the grid based on the weather forecast and perhaps my forecast usage.

alfredocdmiranda, 2024-01-12:
I am not sure how it works when you are using batteries and other models. For example, a friend of mine has another model (I am not sure which one right now) and he said to me that his inverter doesn't shutdown during the night. I would read the manual before installing it.

muhkuh666, 2024-01-17:
I have the same issue !

johanmeijer, 2024-01-18:
Indeed the solar panel only models will go offline (at least my 1500-s and 750-s as well). 

At this moment Grott can not sent anything because it is only a serving hatch (proxy). It doesn't know anything about the status of  the inverter. It will only sent data if the inverter sends data (and the inverter will not do that if it is offline). 

It is also possible that during the day (with bad weather or early in the morning / late in the evening) the inverter goes multiple times off-/ online. I have seen this multiple times. It sometimes even reset the day values to zero again. 

Most of the time (at my side) the last value that is sent is 0. The status of the inverter (pvstatus) is then also 0.  Maybe you can do something in HA to retain the last value, till the inverter has started it again (and pvstatus changed to 1)?

muhkuh666, 2024-01-22:
Siehe hier:

[https://github.com/johanmeijer/grott/issues/506#issuecomment-1898826268](url)

Sorentino1964, 2024-01-31:
I have the same issue !
Growatt 2500TL

TrueMB, 2024-02-06:
> Most of the time (at my side) the last value that is sent is 0. The status of the inverter (pvstatus) is then also 0. Maybe you can do something in HA to retain the last value, till the inverter has started it again (and pvstatus changed to 1)?

Sadly this isn't the case for me.
I got the same case as the Issuer. Homeassistant should receive the data and do the history graph. But this one is not correct, since there is always at least 10W. Even in the night, which then results in wrong calculations.

I hoped that there is maybe a simple configuration for that, but that answers my question...

My next idea is to create in HA an automation, which triggers after not receiving updates for 10 minutes on the "PV Last Update" Sensor. After that it should publish the 0 Value over mqtt.
The Problem in that case: It wants the whole json to be published. Which either needs to be dynamically or breaks other values..... There would also be a loop.
So far as I could see it, it is not possible to change the sensor value, which got generated from grott.

Which then results in an extra helper, which has the same value as the grott sensor, but can be changed by the automation.

theoretically....

onkytonk, 2025-04-19:
Did anyone ever come up with a solution for this?
My last reading stay active for hours unfortunately. 🙁

johanmeijer, 2025-04-20:
I finally have a HA implementation active and can test with that also. But....  since a week I also replaced my 1000 and 1500 -s Inverters with a 3600 TL-XH with ARK batteries and now my inverters does not go offline any more (the battery provide also power at night). 

Hopefully there is somebody out the with HA skills and the right inverter.