johanmeijer/grott #508: PV3 missing on TL3 configuration

LightMoon, 2024-01-14:
PV3 data is missing and consequently values from some of the sensors are not correct.

Initially, I utilised Grott Add-ons. However, I eventually moved to the Grott native Docker version to eliminate potential contributing factors and having better access to Grott logs.

https://pastebin.com/F6ehUfvT

`root@Docker:~# docker images`
`REPOSITORY       TAG       IMAGE ID       CREATED         SIZE`
`hello-world      latest    d2c94e258dcb   8 months ago    13.3kB`
`ledidobe/grott   2.7.8     db656031005c   11 months ago   214MB
`

`root@Docker:~# docker ps -a`
`CONTAINER ID   IMAGE                  COMMAND                  CREATED        STATUS                      PORTS                                       NAMES`
`27ea93b42f21   ledidobe/grott:2.7.8   "python -u grott.py …"   10 hours ago   Up 54 minutes               0.0.0.0:5279->5279/tcp, :::5279->5279/tcp   admiring_goodall`

#https://github.com/muppet3000/homeassistant-grott/issues/6#issuecomment-1858571029
#https://github.com/egguy/grott-home-assistant-add-on/issues/74#issue-2045875351

LightMoon, 2024-01-16:
Update - If I change the invtype = min, I won't get any data for the `SPM2DBE044` growatt smart meter in HA.

Surb109, 2024-05-27:
I have a Growatt MIN 8000TL-X2 and  I'm having the same issue with  **PV3 Energy - Today** and **PV3 Energy - Total** not showing any data.

johanmeijer, 2024-06-18:
@LightMoon @Surb109 Does this not showing up in HA or also in Grott itself?

Surb109, 2024-06-18:
@LightMoon  that's correct it doesn't show up  Grott and I'm using Grott v2.8.3 and a value of MIN type inverter. As a result it's not available in HA as it's using MQTT to retrieve the data. 

I've included the log file from Grott if if helps.
[_grott_logs (1).txt](https://github.com/user-attachments/files/15893290/_grott_logs.1.txt)