johanmeijer/grott #509: Debug grott container

martreumer, 2024-01-14:
How do I access logs for Grott container, or is there another way to debug?

After starting the Grott container I still get updates in the Growatt Dashboard, but it is not connecting to MQTT. I have HA and Mosquitto both running in separate containers on the same machine and I can see HA connecting to MQTT just fine.

It would be nice to get some info about what Grott is doing.

martreumer, 2024-01-15:
It suddenly started working so no issues there anymore, but my question still stands. In case something goes wrong in the future it would be nice to know how to get info from grott

johanmeijer, 2024-01-18:
You can look in a log of a container with the docker logs commands.  

If you use a tool like portainer then it is also possible to use this to view the logs via the browser.

closed 2024-01-19 by martreumer